Koenkk/zigbee2mqtt #15528: innr RS 227 T refuses to work after restart

matzematthaey, 2022-12-13:
### What happened?

The innr RS 227 T bulb seems to be refusing to work at all after you restart the application.

happy to provide more feedback if you wish so.

All light bulbs are listed as follows:

Availability: Offline
Device type: Router
Zigbee Model: RS 227 T
Zigbee Manufacturer: innr
Description: GU10 spot 420 lm, dimmable, white spectrum
Support status: Supported

IEEE Address: 0xc498860000064eda
Network address: 0xAF06
Firmware build date: 20211013
Firmware version: 2.03.00
Manufacturer: [Innr](https://www.zigbee2mqtt.io/supported-devices/#v=Innr)
Model: [RS 227 T](https://www.zigbee2mqtt.io/devices/RS_227_T.html#innr-rs%2520227%2520t)
Power: 
Interview completed: True

### What did you expect to happen?

The bulbs work every time.

### How to reproduce it (minimal and precise)

- Pair a RS 227 T bulb
- restart z2m

### Zigbee2MQTT version

1.28.4 

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Debug log

Failed to ping 'Light: Büro Leselampe 1' (attempt 1/1, Read 0xc49886000006c63e/0 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received))

matzematthaey, 2022-12-13:
Just flashed coordinator firmware 0x26580700. Does not make any difference.

closed 2023-01-21 by github-actions[bot]

FWeinb, 2023-09-12:
I just ordered a few of these bulbs and can confirm this issue is still present with version 1.33.0. After paring everything seems to work fine but as soon as zigbee2mqtt is restarted these bulbs would not respond to any command.