Koenkk/zigbee2mqtt #15535: [New device support]:  _TZE200_nvups4nh

colseven, 2022-12-13:
### Link

https://de.aliexpress.com/item/1005004878609097.html?srcSns=sns_WhatsApp&spreadType=socialShare&bizType=ProductDetail&social_params=20745041741&aff_fcid=93226e6d25ce4ff79457c36aa8f7e35d-1670958859601-08691-_mrBMp90&tt=MG&aff_fsk=_mrBMp90&aff_platform=default&sk=_mrBMp90&aff_trace_key=93226e6d25ce4ff79457c36aa8f7e35d-1670958859601-08691-_mrBMp90&shareId=20745041741&businessType=ProductDetail&platform=AE&terminal_id=59021fe2b92a43d3b172a1e46d4a10b3&afSmartRedirect=y

### Database entry

0xe0798dfffe25418c

### Comments

not working
[image]
[image]

Blakout007, 2022-12-24:
Yeah, have same issue.

But is about trying to add it myself with help from https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html

Hope this will work 😂

Koenkk, 2022-12-25:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

colseven, 2022-12-25:
@Koenkk did Not work
[image]

colseven, 2022-12-25:
[image]

Koenkk, 2022-12-26:
Are you sure the external converter is added to your `configuration.yaml` and you restarted z2m after that?

colseven, 2022-12-26:
Yes i‘am sure, it is a door Sensor with temperature

colseven, 2022-12-26:
[image]
[image]

Koenkk, 2022-12-27:
Can you provide your `configuration.yaml` here? I don't understand why it's not recognized.

colseven, 2022-12-27:
Hope this will help @Koenkk 

[image]
[image]

Koenkk, 2022-12-28:
Can you provide the debug log when restarting z2m?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

colseven, 2022-12-29:
@Koenkk 
2022-12-29 08:38:37Received Zigbee message from '0xe0798dfffe25418c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,214],"type":"Buffer"},"datatype":2,"dp":7}],"seq":33536}' from endpoint 1 with groupID 0
2022-12-29 08:38:37Received Zigbee message from '0xe0798dfffe25418c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,49],"type":"Buffer"},"datatype":2,"dp":8}],"seq":33536}' from endpoint 1 with groupID 0
2022-12-29 08:38:37Received Zigbee message from '0xe0798dfffe25418c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,75],"type":"Buffer"},"datatype":2,"dp":2}],"seq":33536}' from endpoint 1 with groupID 0

colseven, 2022-12-29:
MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{"routes":false,"type":"raw","value":{"links":[{"depth":1,"linkquality":134,"lqi":134,"relationship":1,"routes":[],"source":{"ieeeAddr":"0xa4c13860f56bd1ca","networkAddress":5293},"sourceIeeeAddr":"0xa4c13860f56bd1ca","sourceNwkAddr":5293,"target":{"ieeeAddr":"0x00124b0026b8785d","networkAddress":0},"targetIeeeAddr":"0x00124b0026b8785d"},{"depth":255,"linkquality":60,"lqi":60,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"sourceIeeeAddr":"0xa4c1383be1a1c956","sourceNwkAddr":37626,"target":{"ieeeAddr":"0x00124b0026b8785d","networkAddress":0},"targetIeeeAddr":"0x00124b0026b8785d"},{"depth":255,"linkquality":33,"lqi":33,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c13810df7dfe85","networkAddress":8801},"sourceIeeeAddr":"0xa4c13810df7dfe85","sourceNwkAddr":8801,"target":{"ieeeAddr":"0x00124b0026b8785d","networkAddress":0},"targetIeeeAddr":"0x00124b0026b8785d"},{"depth":255,"linkquality":48,"lqi":48,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c138a4f5dd990c","networkAddress":56748},"sourceIeeeAddr":"0xa4c138a4f5dd990c","sourceNwkAddr":56748,"target":{"ieeeAddr":"0x00124b0026b8785d","networkAddress":0},"targetIeeeAddr":"0x00124b0026b8785d"},{"depth":255,"linkquality":21,"lqi":21,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xf0d1b800001da7d3","networkAddress":11310},"sourceIeeeAddr":"0xf0d1b800001da7d3","sourceNwkAddr":11310,"target":{"ieeeAddr":"0x00124b0026b8785d","networkAddress":0},"targetIeeeAddr":"0x00124b0026b8785d"},{"depth":255,"linkquality":89,"lqi":89,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c13809cb99265b","networkAddress":29829},"sourceIeeeAddr":"0xa4c13809cb99265b","sourceNwkAddr":29829,"target":{"ieeeAddr":"0x00124b0026b8785d","networkAddress":0},"targetIeeeAddr":"0x00124b0026b8785d"},{"depth":255,"linkquality":28,"lqi":28,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c138ae7e5a7fc1","networkAddress":7948},"sourceIeeeAddr":"0xa4c138ae7e5a7fc1","sourceNwkAddr":7948,"target":{"ieeeAddr":"0x00124b0026b8785d","networkAddress":0},"targetIeeeAddr":"0x00124b0026b8785d"},{"depth":0,"linkquality":57,"lqi":57,"relationship":0,"routes":[],"source":{"ieeeAddr":"0x00124b0026b8785d","networkAddress":0},"sourceIeeeAddr":"0x00124b0026b8785d","sourceNwkAddr":0,"target":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"targetIeeeAddr":"0xa4c1383be1a1c956"},{"depth":4,"linkquality":24,"lqi":24,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c13808d7d34a71","networkAddress":52270},"sourceIeeeAddr":"0xa4c13808d7d34a71","sourceNwkAddr":52270,"target":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"targetIeeeAddr":"0xa4c1383be1a1c956"},{"depth":1,"linkquality":51,"lqi":51,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c13860f56bd1ca","networkAddress":5293},"sourceIeeeAddr":"0xa4c13860f56bd1ca","sourceNwkAddr":5293,"target":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"targetIeeeAddr":"0xa4c1383be1a1c956"},{"depth":2,"linkquality":0,"lqi":0,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c138a4f5dd990c","networkAddress":56748},"sourceIeeeAddr":"0xa4c138a4f5dd990c","sourceNwkAddr":56748,"target":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"targetIeeeAddr":"0xa4c1383be1a1c956"},{"depth":5,"linkquality":69,"lqi":69,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c1387f51c259ba","networkAddress":16809},"sourceIeeeAddr":"0xa4c1387f51c259ba","sourceNwkAddr":16809,"target":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"targetIeeeAddr":"0xa4c1383be1a1c956"},{"depth":1,"linkquality":24,"lqi":24,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xf0d1b800001da7d3","networkAddress":11310},"sourceIeeeAddr":"0xf0d1b800001da7d3","sourceNwkAddr":11310,"target":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"targetIeeeAddr":"0xa4c1383be1a1c956"},{"depth":4,"linkquality":18,"lqi":18,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c13831455f18bd","networkAddress":31957},"sourceIeeeAddr":"0xa4c13831455f18bd","sourceNwkAddr":31957,"target":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"targetIeeeAddr":"0xa4c1383be1a1c956"},{"depth":1,"linkquality":9,"lqi":9,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xa4c13809cb99265b","networkAddress":29829},"sourceIeeeAddr":"0xa4c13809cb99265b","sourceNwkAddr":29829,"target":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"targetIeeeAddr":"0xa4c1383be1a1c956"},{"depth":2,"linkquality":81,"lqi":81,"relationship":1,"routes":[],"source":{"ieeeAddr":"0xe0798dfffe25418c","networkAddress":15599},"sourceIeeeAddr":"0xe0798dfffe25418c","sourceNwkAddr":15599,"target":{"ieeeAddr":"0xa4c1383be1a1c956","networkAddress":37626},"targetIeeeAddr":"0xa4c1383be1a1c956"},{"depth":0,"linkquality":136,"lqi":136,"relationship":0,"routes":[],"source":{"ieeeAddr":"0x00124b0026b8785d","networkAddress":0},"sourceIeeeAddr":"0x00124b0026b8785d","sourceNwkAddr":0,"target":{"ieeeAddr":"0xa4c13860f56bd1ca","networkAddress":5293},"targetIeeeAddr":"0xa4c13860f56bd1ca"},{"depth":2,"linkquality":60,"lqi":60,"relationship":1,"routes":[],"source":{"ieeeAddr":"0xa4c138a4f5dd990c","networkAddress":56748},"sourceIeeeAddr":"0xa4c138a4f5dd990c","sourceNwkAddr":56748,"target":{"ieeeAddr":"0xa4c13860f56bd1ca","networkAddress":5293},"targetIeeeAddr":"0xa4c13860f56bd1ca"},{"depth":2,"linkquality":45,"lqi":45,"relationship":1,"routes":[],"sou [... cut ...]
Debug 2022-12-29 08:52:07Received Zigbee message from '0xe0798dfffe25418c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,49],"type":"Buffer"},"datatype":2,"dp":8}],"seq":35328}' from endpoint 1 with groupID 0
Debug 2022-12-29 08:52:07Received Zigbee message from '0xe0798dfffe25418c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2}],"seq":35328}' from endpoint 1 with groupID 0

Koenkk, 2022-12-29:
It looks to be detected as support now, does it also appear as such in the frontend?

colseven, 2022-12-29:
@Koenkk 
[image]
[image]
[image]

Koenkk, 2022-12-29:
Strange, can you provide the debug log file? You can find it in the data/log directory.

colseven, 2022-12-29:
@Koenkk
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10318694/log.txt)

Koenkk, 2022-12-29:
I updated https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541 (changed the `fingerprint` part). Can you try again?

colseven, 2022-12-29:
Code changed, but I can only integrate the sensor in a few days because we are on vacation at the moment, let me know as soon as we are back home.

colseven, 2023-01-03:
@Koenkk 

He now recognizes the sensor as a temperature and humidity sensor, but it is a window contact. The values are not displayed either.

[image]
[image]
[image]
[image]

Koenkk, 2023-01-04:
I updated https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541 , can you check again/

colseven, 2023-01-08:
@Koenkk 
only humidity no Temperature and no Door Sensor ... but it is an different Image.

[image]
[image]
[image]

Koenkk, 2023-01-09:
Can you provide the debug log when blowing into the sensor? (this will trigger the sensor to send a temperature update)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

closed 2023-02-17 by github-actions[bot]