Koenkk/zigbee2mqtt #15539: TS0201 reports genPowerCfg but battery state null

Vebryn, 2022-12-13:
### What happened?

Hi,

First, Zigbee2MQTT is an awesome project, so grateful !

I got some TS0201 devices, all data are reported properly except battery on one device.

I tried to pair many times, issue still here.

[image]

### What did you expect to happen?

I expect to see battery state and voltage as they are reported from device.

### Zigbee2MQTT version

1.28.4 commit: 52e545f

### Adapter firmware version

 20220219

### Adapter

Slaesh's CC2652RB stick with an usb cable

### Debug log

info  2022-12-13 21:25:53: Device '0x5c0272fffecbe7d7' joined
info  2022-12-13 21:25:53: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffecbe7d7","ieee_address":"0x5c0272fffecbe7d7"},"type":"device_joined"}'
info  2022-12-13 21:25:53: Starting interview of '0x5c0272fffecbe7d7'
info  2022-12-13 21:25:53: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffecbe7d7","ieee_address":"0x5c0272fffecbe7d7","status":"started"},"type":"device_interview"}'
debug 2022-12-13 21:25:53: Device '0x5c0272fffecbe7d7' announced itself
info  2022-12-13 21:25:53: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffecbe7d7","ieee_address":"0x5c0272fffecbe7d7"},"type":"device_announce"}'
debug 2022-12-13 21:25:54: Received Zigbee message from '0x5c0272fffecbe7d7', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ2000_a476raq2","modelId":"TS0201"}' from endpoint 1 with groupID 0
debug 2022-12-13 21:25:57: Received Zigbee message from '0x5c0272fffecbe7d7', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":28}' from endpoint 1 with groupID 0
debug 2022-12-13 21:25:57: Received Zigbee message from '0x5c0272fffecbe7d7', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2130}' from endpoint 1 with groupID 0
debug 2022-12-13 21:25:58: Received Zigbee message from '0x5c0272fffecbe7d7', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2130}' from endpoint 1 with groupID 0
debug 2022-12-13 21:25:58: Received Zigbee message from '0x5c0272fffecbe7d7', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2022-12-13 21:25:59: Received Zigbee message from '0x5c0272fffecbe7d7', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2022-12-13 21:25:59: Received Zigbee message from '0x5c0272fffecbe7d7', type 'readResponse', cluster 'genBasic', data '{"appVersion":67}' from endpoint 1 with groupID 0
debug 2022-12-13 21:26:00: Received Zigbee message from '0x5c0272fffecbe7d7', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5420}' from endpoint 1 with groupID 0
debug 2022-12-13 21:26:00: Received Zigbee message from '0x5c0272fffecbe7d7', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
debug 2022-12-13 21:26:01: Received Zigbee message from '0x5c0272fffecbe7d7', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2022-12-13 21:26:01: Received Zigbee message from '0x5c0272fffecbe7d7', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2022-12-13 21:26:02: Received Zigbee message from '0x5c0272fffecbe7d7', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":28}' from endpoint 1 with groupID 0
debug 2022-12-13 21:26:02: Received Zigbee message from '0x5c0272fffecbe7d7', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":28}' from endpoint 1 with groupID 0
debug 2022-12-13 21:26:02: Received Zigbee message from '0x5c0272fffecbe7d7', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
info  2022-12-13 21:26:02: Successfully interviewed '0x5c0272fffecbe7d7', device has successfully been paired
info  2022-12-13 21:26:03: Device '0x5c0272fffecbe7d7' is supported, identified as: TuYa Temperature & humidity sensor with display (TS0201)
info  2022-12-13 21:26:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor with display","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Voltage of the battery in millivolts","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0201","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for humidity, takes into effect on next report of device.","name":"humidity_precision","property":"humidity_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the humidity value (absolute offset), takes into effect on next report of device.","name":"humidity_calibration","property":"humidity_calibration","type":"numeric"}],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0x5c0272fffecbe7d7","ieee_address":"0x5c0272fffecbe7d7","status":"successful","supported":true},"type":"device_interview"}'
info  2022-12-13 21:26:03: Configuring '0x5c0272fffecbe7d7'
info  2022-12-13 21:26:03: MQTT publish: topic 'homeassistant/sensor/0x5c0272fffecbe7d7/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x5c0272fffecbe7d7"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0x5c0272fffecbe7d7"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x5c0272fffecbe7d7 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x5c0272fffecbe7d7","unique_id":"0x5c0272fffecbe7d7_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2022-12-13 21:26:03: MQTT publish: topic 'homeassistant/sensor/0x5c0272fffecbe7d7/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x5c0272fffecbe7d7"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0x5c0272fffecbe7d7"},"device_class":"temperature","enabled_by_default":true,"name":"0x5c0272fffecbe7d7 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x5c0272fffecbe7d7","unique_id":"0x5c0272fffecbe7d7_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
info  2022-12-13 21:26:03: MQTT publish: topic 'homeassistant/sensor/0x5c0272fffecbe7d7/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x5c0272fffecbe7d7"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0x5c0272fffecbe7d7"},"device_class":"humidity","enabled_by_default":true,"name":"0x5c0272fffecbe7d7 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0x5c0272fffecbe7d7","unique_id":"0x5c0272fffecbe7d7_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
[... cut ...]
debug 2022-12-13 21:27:33: Received Zigbee message from '0x5c0272fffecbe7d7', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5320}' from endpoint 1 with groupID 0
info  2022-12-13 21:27:33: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffecbe7d7', payload '{"battery":null,"humidity":53.2,"linkquality":40,"temperature":21,"voltage":null}'
debug 2022-12-13 21:27:33: Received Zigbee message from '0x5c0272fffecbe7d7', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":28}' from endpoint 1 with groupID 0

Vebryn, 2022-12-13:
https://github.com/Koenkk/zigbee2mqtt/issues/11722

lux73, 2022-12-14:
How long have you wait? Sometimes it needs some hours before refrsh battery state

Vebryn, 2022-12-14:
Still no battery info after one day on

closed 2023-02-24 by github-actions[bot]

djashjones, 2023-06-10:
Same issue here. I'm thinking of migrating from Hubitat and never had an issue with it. Battery reporting is usually every 4 hours.