Koenkk/zigbee2mqtt #15541: No Start after Update to newest HOAS Base System

sebjosb, 2022-12-13:
### What happened?

After Updating to the newest Home Assistant Base System 9.4, the Add On will not start 

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

2022.11.2

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_

### Debug log

[23:25:06] INFO: Preparing to start...
[23:25:06] INFO: Socat not enabled
[23:25:08] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2022-12-13 23:25:13: Logging to console and directory: '/config/zigbee2mqtt/log/2022-12-13.23-25-13' filename: log.txt
Zigbee2MQTT:info  2022-12-13 23:25:13: Starting Zigbee2MQTT version 1.28.4 (commit #unknown)
Zigbee2MQTT:info  2022-12-13 23:25:13: Starting zigbee-herdsman (0.14.76)
Zigbee2MQTT:error 2022-12-13 23:26:57: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-12-13 23:26:57: Failed to start zigbee
Zigbee2MQTT:error 2022-12-13 23:26:57: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-12-13 23:26:57: Exiting...
Zigbee2MQTT:error 2022-12-13 23:26:57: Error: Failure send setConfigurationValue:{"type":"Buffer","data":[3,0,1,83,0,18,48,117]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:516:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

AliakseiT, 2022-12-14:
The problem is observed with both Stable and Edge add-ons.

vitaly-i, 2022-12-22:
Hello!
I have the same problem.
The port specified is correct /dev/ttyACM0

[image]
[image]
[image]

closed 2023-01-29 by github-actions[bot]