Koenkk/zigbee2mqtt #15542: Failed to configure messages for Smartwings roller shades

stu247, 2022-12-13:
### What happened?

On startup, with the manufacturer Smartwings model WM25/L-Z I get the frontend popup message

```Configuring ‘shade_w_bedroom_se’```

three times for each shade.  For each one, after a certain amount of time, I get the following message in a frontend popup and in the Logs:

```error 2022-12-13 17:16:48 Failed to configure 'shade_w_bedroom_se', attempt 3 (Error: ConfigureReporting 0xdc8e95fffe8a9b8c/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 45527 - 1 - 37 - 1 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:564:17) at processTimers (node:internal/timers:507:7))```

I am not able to send commands to the shades while these are running. 

### What did you expect to happen?

No error messages in the Logs.

### How to reproduce it (minimal and precise)

It does this every time at startup.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10222932/log.txt)

satter, 2022-12-15:
I have similar errors on configuring attempt of different devices
```
Dec 15 10:13:04 homie1 npm[194495]: Zigbee2MQTT:error 2022-12-15 10:13:04: Failed to configure 'efecta_co2', attempt 2 (Error: ConfigureReporting 0x00124b002871ec09/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 21825 - 1 - 13 - 1026 - 7 after 10000ms)
Dec 15 10:13:04 homie1 npm[194495]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
Dec 15 10:13:04 homie1 npm[194495]:     at listOnTimeout (node:internal/timers:559:17)
Dec 15 10:13:04 homie1 npm[194495]:     at processTimers (node:internal/timers:502:7))

Dec 15 10:21:31 homie1 npm[194495]: Zigbee2MQTT:error 2022-12-15 10:21:31: Failed to configure 'aqara_tvoc1', attempt 2 (Error: Bind 0x54ef441000106dca/1 msTemperatureMeasurement from '0x00124b00258d3c29/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Dec 15 10:21:31 homie1 npm[194495]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
Dec 15 10:21:31 homie1 npm[194495]:     at listOnTimeout (node:internal/timers:559:17)
Dec 15 10:21:31 homie1 npm[194495]:     at processTimers (node:internal/timers:502:7))
```
I think it is related to v1.28.4 upgrade

mtnbiker717, 2022-12-22:
Same errors, all my blinds keep trying to configure giving me numerous errors:

2022-12-22 08:53:33Failed to configure 'Shades LR Left', attempt 3 (Error: ConfigureReporting 0xe0798dfffedebf36/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 6853 - 1 - 35 - 1 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))
2022-12-22 08:54:04Failed to configure 'Shades Master Small', attempt 3 (Error: ConfigureReporting 0xe0798dfffecf885a/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30595 - 1 - 37 - 1 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

cymone-smart, 2023-01-14:
Same errors and issues.

BradleyFord, 2023-01-30:
same here as well

buddy31130, 2023-02-04:
Hi, same error, but with PIR sensor SNZB-03 :|

stu247, 2023-03-07:
I am still seeing this issue.  Is any other information needed?

Koenkk, 2023-03-07:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['WM25/L-Z'],
    model: 'WM25L-Z',
    vendor: 'Smartwings',
    description: 'Roller shade',
    fromZigbee: [fz.cover_position_tilt, fz.battery],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    meta: {battery: {dontDividePercentage: true}, coverInverted: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
        //await reporting.batteryPercentageRemaining(endpoint);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
    exposes: [e.cover_position(), e.battery()],
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed
- is the battery % still reported after this?

mtnbiker717, 2023-03-07:
Forgive me as I'm not well versed with yaml, I added the `external_converters:
  - ext_converter.js` to my configuration.yaml and when I check my configuration in HA, I'm getting this error: Integration error: external_converters - Integration 'external_converters' not found.

highincolorado, 2023-03-07:
> Forgive me as I'm not well versed with yaml, I added the
> 
> ```
> external_converters:
>   - ext_converter.js
> ```
> 
> to my configuration.yaml and when I check my configuration in HA, I'm getting this error: Integration error: external_converters - Integration 'external_converters' not found.

Did you add the file to the Zigbee2mqtt directory and add the text to the configuration.yaml in that directory?

mtnbiker717, 2023-03-07:
As you sent that I realized I was adding it to HA's configuration.yaml... I just got it in the Z2M folder and config yaml. I'll try it and let you know. Thanks!

highincolorado, 2023-03-07:
Yeah that was not obvious to me the first time I tried either

stu247, 2023-03-07:
@Koenkk I tried the ext_converter.js you posted.  It still generated the error messages.  The battery % is reported.

Koenkk, 2023-03-08:
@stu247 the error should be different now, can you provide me the log?

stu247, 2023-03-08:
Here is the new error message @Koenkk 

```
error 2023-03-08 13:24:49: Failed to configure 'shade_w_bedroom_sw', attempt 2 (Error: ConfigureReporting 0x70ac08fffe341ee0/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":62000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42903 - 1 - 30 - 258 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:564:17)
    at processTimers (node:internal/timers:507:7))
```

BradleyFord, 2023-03-09:
I just tested it on mine with this message

`Error 2023-03-09 14:11:35Failed to configure 'Smartwings', attempt 1 (Error: ConfigureReporting 0xe0798dfffeb93d07/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":62000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51115 - 1 - 138 - 258 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))`

BradleyFord, 2023-03-09:
Below is the second configure attempt

```
Info 2023-03-09 14:14:08MQTT publish: topic 'zigbee2mqtt/Smartwings', payload '{"battery":5,"last_seen":"2023-03-09T07:14:07+00:00","linkquality":109,"position":0,"state":"CLOSE"}'
Info 2023-03-09 14:14:08Configuring 'Smartwings'
Info 2023-03-09 14:14:14MQTT publish: topic 'zigbee2mqtt/Smartwings', payload '{"battery":5,"last_seen":"2023-03-09T07:14:13+00:00","linkquality":116,"position":0,"state":"CLOSE"}'
Info 2023-03-09 14:14:28MQTT publish: topic 'zigbee2mqtt/Smartwings', payload '{"battery":5,"last_seen":"2023-03-09T07:14:27+00:00","linkquality":109,"position":0,"state":"CLOSE"}'
Error 2023-03-09 14:14:38Failed to configure 'Smartwings', attempt 2 (Error: ConfigureReporting 0xe0798dfffeb93d07/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":62000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51115 - 1 - 196 - 258 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))
```

Koenkk, 2023-03-09:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/15542#issuecomment-1458685572 , also make sure now when manually closing the curtain it reports its position correctly.

BradleyFord, 2023-03-09:
My logs after updating the converter

```Info 2023-03-10 00:19:07Configuring 'Smartwings'
Info 2023-03-10 00:19:07MQTT publish: topic 'zigbee2mqtt/Smartwings', payload '{"battery":5,"last_seen":"2023-03-09T17:19:06+00:00","linkquality":153,"position":29,"state":"OPEN"}'
Info 2023-03-10 00:19:13MQTT publish: topic 'zigbee2mqtt/Smartwings', payload '{"battery":5,"last_seen":"2023-03-09T17:19:12+00:00","linkquality":156,"position":29,"state":"OPEN"}'
Info 2023-03-10 00:19:15MQTT publish: topic 'zigbee2mqtt/Smartwings', payload '{"battery":5,"last_seen":"2023-03-09T17:19:14+00:00","linkquality":123,"position":0,"state":"CLOSE"}'
Info 2023-03-10 00:19:27MQTT publish: topic 'zigbee2mqtt/Smartwings', payload '{"battery":5,"last_seen":"2023-03-09T17:19:26+00:00","linkquality":87,"position":0,"state":"CLOSE"}'
Error 2023-03-10 00:19:37Failed to configure 'Smartwings', attempt 1 (Error: ConfigureReporting 0xe0798dfffeb93d07/1 closuresWindowCovering([{"attribute":"currentPositionLiftPercentage","minimumReportInterval":1,"maximumReportInterval":62000,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51115 - 1 - 211 - 258 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))```

Koenkk, 2023-03-11:
@BradleyFord I don't think you are using the latest version, in https://github.com/Koenkk/zigbee2mqtt/issues/15542#issuecomment-1458685572 `currentPositionLiftPercentage` has been commented out while in your logs it generates the error.

mtnbiker717, 2023-03-12:
For the first time I got this:

`Successfully configured 'Blind Kitchen'`

I'll keep an eye on it and see what happens.

BradleyFord, 2023-03-12:
> @BradleyFord I don't think you are using the latest version, in [#15542 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/15542#issuecomment-1458685572) `currentPositionLiftPercentage` has been commented out while in your logs it generates the error.

Yep, your correct. dont know how I made that little mistake!! oh well.

But on a positive note, that change also fixes the issue for me as well. Thanks so much.

Is anything lost after that change? It seems to still be reporting battery and position correctly.

stu247, 2023-03-12:
I now have 13 roller shades and they all have the #13878 where the state is backwards as well as this issue.  I have not modified any settings on the roller shades via the remote.  I have worked around issue 13878 by using the code provided by @nathankopp.  Today I've merged @Koenkk code above and @nathankopp code.  I added the backwards_cover_state and set coverInverted to false.  When I run it I get the new error closuresWindowCovering message I list above.  The position and the up and down commands work correctly.  I can also see the HA battery percentage changing.

It is strange that the fix is working for @mtnbiker717 and @BradleyFord, but not me.  Is it possible we are using different roller shade firmware versions?  If I turn on debug in Zigbee2MQTT logging, I see messages like this for each shade every 5 minutes:

```
Received Zigbee message from 'shade_dining_e', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":2,"imageType":0,"manufacturerCode":4098}' from endpoint 1 with groupID 0
```
I am guessing the firmware version is fileVersion = 2.  Is that the version you are seeing @mtnbiker717 and @BradleyFord?

Here is my merged ext_converter.js:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

backwards_cover_state = {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {'open': 'downClose', 'close': 'upOpen', 'stop': 'stop', 'on': 'downClose', 'off': 'upOpen'};
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.command('closuresWindowCovering', lookup[value], {}, utils.getOptions(meta.mapped, entity));
        },
    };

const definition = {
    zigbeeModel: ['WM25/L-Z'],
    model: 'WM25L-Z',
    vendor: 'Smartwings',
    description: 'Roller shade',
    fromZigbee: [fz.cover_position_tilt, fz.battery],
    toZigbee: [backwards_cover_state, tz.cover_position_tilt],
    meta: {battery: {dontDividePercentage: true}, coverInverted: false},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
        //await reporting.batteryPercentageRemaining(endpoint);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
    exposes: [e.cover_position(), e.battery()],
};

module.exports = definition;
```

mtnbiker717, 2023-03-13:
> I am guessing the firmware version is fileVersion = 2. Is that the version you are seeing @mtnbiker717 and @BradleyFord

I've tried to get the Dev Console to output what file version and I can't seem to get it to pull up...

Koenkk, 2023-03-13:
Thanks, integrated the fix. Also fixed the missing power source.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-13 by Koenkk

BradleyFord, 2023-03-13:
Those changes work for me including the power source