Koenkk/zigbee2mqtt #15543: _TZ3000_vjhcenzo stopped working after 1.28.4

felipejgribeiro, 2022-12-14:
### What happened?

After I upgraded Z2M to version 1.28.4 all my _TZ3000_vjhcenzo devices stopped working when I sent any command to them. They just works when you physically push the button.
Downgrading Z2M to the previous version make all work again.

### What did you expect to happen?

When you send a command or create an automation via software the light should turn on.

### How to reproduce it (minimal and precise)

Every time I try to turn on a light through HA or even Z2M this error is shown: Device 'XYZ' has no endpoint 'left' (or center or right. It depends on the button).

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff zigbee 3.0 usb dongle plus

semerim, 2022-12-14:
Same here, had to rollback to 1.28.2-1

Koenkk, 2022-12-14:
Duplicate of #15386 (TLDR; will be fixed in next version)