Koenkk/zigbee2mqtt #15553: [New device support]: Sandy Beach TRV (Tuya0601, _TZE200_xby0s3ta)

Woodpeckercz, 2022-12-14:
### Link

https://www.amazon.co.uk/Sandy-Beach-Thermostat-Connecting-Additional/dp/B093B4SCBT/ref=asc_df_B093B4SCBT/?tag=googshopuk-21&linkCode=df0&hvadid=499398303979&hvpos=&hvnetw=g&hvrand=7886526820560141816&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1007850&hvtargid=pla-1348771650200&th=1

### Database entry

{"id":71,"type":"EndDevice","ieeeAddr":"0x9035eafffe2c3d2e","nwkAddr":27721,"manufId":4098,"manufName":"_TZE200_xby0s3ta","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��,+\u0019\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0011","65506":31,"65508":1,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_xby0s3ta","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":85,"powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1671050386116,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,
I need help with my new TRV.
I bought one from amazon a while back and it paired with Z2M straight away.
Now I bought a few more to install on the rest of the radiators and found that these are newer and are not currently supported in Z2M.
I managed to find some datapoints, but I'm struggling with it quite a bit since i've never done anything like this before.
So far I found these datapoints and guessed (tested) function of some of them:

2 - local temperature
3 - current temperature setpoint
4 - preset? (holiday(0), auto(1), manual(2), comfort(3), eco(4), boost(5)
7 - Child lock
13 - ?
44 - ?
102 - Min temperature
103 - Max temperature
104 - open window detection ?
105 - boost duration ?
106 - ?
107 - ?
108 - ?
109 - Valve Position
110 - ?
111 - ?
112 - week schedule?
113 - week schedule?
114 - ?
115 - ?
116 - auto child lock on/off & delay?
117 - ?
119 - running state? force valve position ?

i did get the local temperature and temperature setpoint to show up in Z2M but I'm struggling with the rest.

here's some debug log i've collected during testing.

Debug 2022-12-14 16:02:25Datapoint 102 not defined for '_TZE200_xby0s3ta' with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,5]}}
Debug 2022-12-14 16:02:25Datapoint 103 not defined for '_TZE200_xby0s3ta' with data {"dp":103,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,35]}}
Debug 2022-12-14 16:02:26Datapoint 13 not defined for '_TZE200_xby0s3ta' with data {"dp":13,"datatype":5,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-14 16:02:26Datapoint 44 not defined for '_TZE200_xby0s3ta' with data {"dp":44,"datatype":2,"data":{"type":"Buffer","data":[255,255,255,246]}}
Debug 2022-12-14 16:02:26Datapoint 104 not defined for '_TZE200_xby0s3ta' with data {"dp":104,"datatype":0,"data":{"type":"Buffer","data":[0,5,10]}}
Debug 2022-12-14 16:02:28Datapoint 105 not defined for '_TZE200_xby0s3ta' with data {"dp":105,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,44]}}
Debug 2022-12-14 16:02:28Datapoint 106 not defined for '_TZE200_xby0s3ta' with data {"dp":106,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-14 16:02:28Datapoint 107 not defined for '_TZE200_xby0s3ta' with data {"dp":107,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,35]}}
Debug 2022-12-14 16:02:28Datapoint 108 not defined for '_TZE200_xby0s3ta' with data {"dp":108,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,15]}}
Debug 2022-12-14 16:02:29Datapoint 109 not defined for '_TZE200_xby0s3ta' with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
Debug 2022-12-14 16:02:29Datapoint 110 not defined for '_TZE200_xby0s3ta' with data {"dp":110,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-14 16:02:29Datapoint 111 not defined for '_TZE200_xby0s3ta' with data {"dp":111,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-14 16:02:29Datapoint 112 not defined for '_TZE200_xby0s3ta' with data {"dp":112,"datatype":0,"data":{"type":"Buffer","data":[6,0,20,8,0,15,11,30,15,12,30,15,17,30,20,22,0,15]}}
Debug 2022-12-14 16:02:30Datapoint 113 not defined for '_TZE200_xby0s3ta' with data {"dp":113,"datatype":0,"data":{"type":"Buffer","data":[6,0,20,8,0,15,11,30,15,12,30,15,17,30,20,22,0,15]}}
Debug 2022-12-14 16:02:30Datapoint 114 not defined for '_TZE200_xby0s3ta' with data {"dp":114,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,15]}}
Debug 2022-12-14 16:02:30Datapoint 115 not defined for '_TZE200_xby0s3ta' with data {"dp":115,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-14 16:02:30Datapoint 116 not defined for '_TZE200_xby0s3ta' with data {"dp":116,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-14 16:02:30Datapoint 117 not defined for '_TZE200_xby0s3ta' with data {"dp":117,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,1]}}
Debug 2022-12-14 16:02:31Datapoint 119 not defined for '_TZE200_xby0s3ta' with data {"dp":119,"datatype":0,"data":{"type":"Buffer","data":[2,0,0]}}
Debug 2022-12-14 16:02:31Datapoint 119 not defined for '_TZE200_xby0s3ta' with data {"dp":119,"datatype":0,"data":{"type":"Buffer","data":[2,0,0]}}
Debug 2022-12-14 16:06:06Datapoint 109 not defined for '_TZE200_xby0s3ta' with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Debug 2022-12-14 16:06:06Datapoint 119 not defined for '_TZE200_xby0s3ta' with data {"dp":119,"datatype":0,"data":{"type":"Buffer","data":[1,0,1]}}
Debug 2022-12-14 16:18:10Datapoint 4 not defined for '_TZE200_xby0s3ta' with data {"dp":4,"datatype":4,"data":{"type":"Buffer","data":[2]}}
Debug 2022-12-14 16:19:10Datapoint 4 not defined for '_TZE200_xby0s3ta' with data {"dp":4,"datatype":4,"data":{"type":"Buffer","data":[1]}}
Debug 2022-12-14 16:19:49Datapoint 4 not defined for '_TZE200_xby0s3ta' with data {"dp":4,"datatype":4,"data":{"type":"Buffer","data":[5]}}
[... 68 lines cut ...]
	e.position(), e.child_lock(),
	exposes.binary('heating', ea.STATE, 'ON', 'OFF').withDescription('Device valve is open or closed (heating or not)'), // not working
	exposes.climate()
                .withSystemMode(['off', 'heat'], ea.STATE_SET) // not working
                .withPreset(['auto', 'manual', 'comfort', 'eco', 'boost']) // partialy working
                .withLocalTemperatureCalibration(-5, 5, 0.1, ea.STATE_SET) // not working
                .withLocalTemperature(ea.STATE) // working
                .withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET), // working
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [

	[2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
	[3, 'local_temperature', tuya.valueConverter.divideBy10],
	[4, 'preset', tuya.valueConverterBasic.lookup({'holiday': tuya.enum(0),'auto': tuya.enum(1), 'manual': tuya.enum(2), 'comfort': tuya.enum(3), 'eco': tuya.enum(4), 'boost': tuya.enum(5)})],
	[7, 'child_lock', tuya.valueConverter.lockUnlock],	
	[109, 'position', tuya.valueConverter.raw],
	[119, 'heating', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

Woodpeckercz, 2022-12-15:
update on my converter:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_xby0s3ta',
        },
    ],

    model: 'TS0601_thermostat',
    vendor: 'TuYa',
    description: 'Radiator valve with thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.child_lock(), e.window_detection(),
        exposes.binary('window_open', ea.STATE).withDescription('Window open?'),
        e.battery_low(), e.valve_detection(), e.position(),
        exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE).withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET,
            'Mode of this device, in the `heat` mode the TS0601 will remain continuously heating, i.e. it does not regulate ' +
            'to the desired temperature. If you want TRV to properly regulate the temperature you need to use mode `auto` ' +
            'instead setting the desired temperature.')
            .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET)
            .withPreset(['schedule', 'manual', 'boost', 'comfort', 'eco', 'away'])
            .withRunningState(['idle', 'heat'], ea.STATE),
        e.auto_lock(), e.away_mode(), e.away_preset_days(), e.boost_time(), e.comfort_temperature(), e.eco_temperature(), e.force(),
        e.max_temperature(), e.min_temperature(), e.away_preset_temperature(),
        exposes.composite('programming_mode').withDescription('Schedule MODE ⏱ - In this mode, ' +
            'the device executes a preset week programming temperature time and temperature.')
            .withFeature(e.week())
            .withFeature(exposes.text('workdays_schedule', ea.STATE_SET))
            .withFeature(exposes.text('holidays_schedule', ea.STATE_SET))
            ],
    meta: {
        // All datapoints go in here
    tuyaDatapoints: [

        [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
        [3, 'local_temperature', tuya.valueConverter.divideBy10],
        [4, 'preset', tuya.valueConverterBasic.lookup({'away': tuya.enum(0),'schedule': tuya.enum(1), 'manual': tuya.enum(2), 'comfort': tuya.enum(3), 'eco': tuya.enum(4), 'boost': tuya.enum(5)})],
        [7, 'child_lock', tuya.valueConverter.lockUnlock],
        [44, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
        [102, 'minTemp', tuya.valueConverter.raw],
        [103, 'maxTemp', tuya.valueConverter.raw],
        [114, 'away_temperature', tuya.valueConverter.divideBy10],
        [109, 'position', tuya.valueConverter.raw],
        [119, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],
        ],
    },
};

module.exports = definition;
`

Woodpeckercz, 2022-12-15:
I'm afte figuring out that the TRV has the same datapoint mapping as:
modelID: 'TS0601', manufacturerName: '_TZE200_z1tyspqw'}
so it just needs to be added to the fingerprint and it should work.

Woodpeckercz, 2022-12-19:
Here's what I was able to identify so far:

`{"dpValues":[{"data":{"data":[0,0,0,255],"type":"Buffer"},"datatype":2,"dp":2}],"seq":28683} - Curent temperature setpoint
{"dpValues":[{"data":{"data":[0,0,0,245],"type":"Buffer"},"datatype":2,"dp":3}],"seq":31243} - local temperature
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":26123} - away
{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":4}],"seq":17163} - auto
{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":4}],"seq":9483} - manual
{"dpValues":[{"data":{"data":[3],"type":"Buffer"},"datatype":4,"dp":4}],"seq":11531} - comfort
{"dpValues":[{"data":{"data":[4],"type":"Buffer"},"datatype":4,"dp":4}],"seq":10763} - eco
{"dpValues":[{"data":{"data":[5],"type":"Buffer"},"datatype":4,"dp":4}],"seq":13835} - boost
{"dpValues":[{"data":{"data":[6],"type":"Buffer"},"datatype":4,"dp":4}],"seq":21003} - complex
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":7}],"seq":8715} - child lock
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":5,"dp":13}],"seq":61708}
{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":44}],"seq":59403} - local temperature calibration
{"dpValues":[{"data":{"data":[0,0,0,3],"type":"Buffer"},"datatype":2,"dp":102}],"seq":19724} - Minimum temperature limit
{"dpValues":[{"data":{"data":[0,0,0,35],"type":"Buffer"},"datatype":2,"dp":103}],"seq":34572} - Maximum temperature limit
{"dpValues":[{"data":{"data":[1,9,10],"type":"Buffer"},"datatype":0,"dp":104}],"seq":9740} - open window on/off, temperature, time
{"dpValues":[{"data":{"data":[0,0,2,88],"type":"Buffer"},"datatype":2,"dp":105}],"seq":38924} - boost time in seconds
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":106}],"seq":40716} - ?? (boost time remeining??)
{"dpValues":[{"data":{"data":[0,0,0,30],"type":"Buffer"},"datatype":2,"dp":107}],"seq":58635} - comfort temperature
{"dpValues":[{"data":{"data":[0,0,0,12],"type":"Buffer"},"datatype":2,"dp":108}],"seq":55819} - eco temperature
{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":109}],"seq":47884} - position
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":110}],"seq":63756}
{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":111}],"seq":34827} - program week type 5+3
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":111}],"seq":34059} - program week type 6+1
{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":111}],"seq":32267} - program week type 7
{"dpValues":[{"data":{"data":[0,0,20,8,0,15,11,30,15,12,30,15,17,30,20,22,0,15],"type":"Buffer"},"datatype":0,"dp":112}],"seq":32779} - week data
{"dpValues":[{"data":{"data":[6,0,20,8,0,15,11,30,15,12,30,15,17,30,20,22,0,15],"type":"Buffer"},"datatype":0,"dp":113}],"seq":33035} - week data
{"dpValues":[{"data":{"data":[0,0,0,15],"type":"Buffer"},"datatype":2,"dp":114}],"seq":53259} - away temperature
{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":117}],"seq":53515} - away duration in days
{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":116}],"seq":3852} - auto lock on/off
{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":117}],"seq":13}
{"dpValues":[{"data":{"data":[1,0,1],"type":"Buffer"},"datatype":0,"dp":119}],"seq":8707}`

Woodpeckercz, 2022-12-19:
Now I just need some help implementing it into the coverter

lucapolesel, 2022-12-19:
I was about to comment on this issue. Just got mine and was testing it.
I've noticed that the manu commented on amazon sying that it is just a HY367 so I tried to add it to the fingeprint of the current converter and it seems to work just fine.
Also the features exposed seem to match to what you have found.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/89341b10d02a30c2fbe7a26d1315fb8af26040f4/devices/tuya.js#L1686

Woodpeckercz, 2022-12-20:
> I was about to comment on this issue. Just got mine and was testing it. I've noticed that the manu commented on amazon saying that it is just a HY367 so I tried to add it to the fingerprint of the current converter and it seems to work just fine. Also the exposed features seem to match with what you have found.
> 
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/89341b10d02a30c2fbe7a26d1315fb8af26040f4/devices/tuya.js#L1686

Is the position showing correctly?

closed 2022-12-20 by Koenkk