Koenkk/zigbee2mqtt #15554: Lidl HG06106C Exposed power_on_behavior but UNSUPPORTED_ATTRIBUTE'

shirou93, 2022-12-14:
### What happened?

Lidl HG06106C Exposed power_on_behavior but when try change UNSUPPORTED_ATTRIBUTE'.

Propably it's dummy expose.

### What did you expect to happen?

Working exposed fetures or remove from GUI.

### How to reproduce it (minimal and precise)

Try change power_on_behavior on Lidl HG06106C exposed webgui panel.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff 3 USB

### Debug log

Publish 'set' 'power_on_behavior' to 'Światło Wejście' failed: 'Error: Write 0x842e14fffe1b95c6/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

shirou93, 2022-12-14:
Same issue on https://www.zigbee2mqtt.io/devices/AC25697.html#ledvance-ac25697

Koenkk, 2022-12-15:
Seems it is not supported, disabled it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-15 by Koenkk

serlinGi, 2023-01-03:
hello,
I have same problem with Lidl led strip HG06104A (Livarno Lux smart LED light strip 2.5m TS0505A)  and Lidl light bulb (Livarno Lux E14 candle CCT TS0502A)

I've update  at 1.29.0 ; my Coordinator is Sonoff-P
zStack3x0, revision 20220219

I ask you if i'is possible to disable at the next update, thanks

[image]
[image]

Koenkk, 2023-01-04:
@serlinGi done! 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

serlinGi, 2023-01-04:
> 

perfect, thx 💪

serlinGi, 2023-01-04:
> @serlinGi done! Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

perfect, thx 💪

bruvv, 2023-01-14:
Same for:
https://www.zigbee2mqtt.io/devices/14158704L.html#lidl-14158704l
and
https://www.zigbee2mqtt.io/devices/14156506L.html#lidl-14156506l

Koenkk, 2023-01-14:
@shirou93 can you check if the power on behavior works with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0505A', manufacturerName: '_TZ3000_dbou1ap4'}],
    model: 'HG06106C',
    vendor: 'Lidl',
    description: 'Livarno Lux E27 bulb RGB',
    toZigbee: [...extend.light_onoff_brightness_colortemp_color({disableColorTempStartup: true, disablePowerOnBehavior: true}).toZigbee, tuya.tz.power_on_behavior],
    fromZigbee: [...extend.light_onoff_brightness_colortemp_color({disableColorTempStartup: true, disablePowerOnBehavior: true}).fromZigbee, tuya.fz.power_on_behavior],
    exposes: [...extend.light_onoff_brightness_colortemp_color({disableColorTempStartup: true, disablePowerOnBehavior: true}).exposes, e.power_on_behavior()],
    meta: {applyRedFix: true, enhancedHue: false},
    configure: async (device, coordinatorEndpoint, logger) => {
        device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 29});
    },
};

module.exports = definition;

```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

shirou93, 2023-01-14:
I check tomorrow and come back with logs.

shirou93, 2023-01-15:
@Koenkk  [image]

After changes sadly on logs ^

Koenkk, 2023-01-15:
I see, disabled it for all lidl lights.