Koenkk/zigbee2mqtt #15556: Smart plug (Aqara EU smart plug SP-EUC01) reports power as a "diagnostic" value instead of it being a "sensor" value in Home Assistant.

ShadowRep, 2022-12-15:
### What happened?

I started to use Aqara EU smart plugs (SP-EUC01).
After I paired them, I found that they do not report the momentanous power draw as a "sensor" value in HA.

My shelly 1pm does that and I expect it because it is a core sensor value of a device that (also) measures power.

### What did you expect to happen?

The power draw should be exposed as a "sensor" value instead of being reported as a "diagnostic" value.

### How to reproduce it (minimal and precise)

Connect a Aqara EU smart plug (SP-EUC01) to zigbee2mqtt in Home Assistant.
Look at the device info.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

ShadowRep, 2022-12-15:
I am sorry, but as far as I understand it the linked PR only says that new entity categories are added.
**_That is fine._**
But I don't think that power should be reported as a "diagnostic" entity and it is not stated in the linked PR, either.

The PR states:

> An entity that provides information about the device itself (e.g., battery capacity, signal, tampering), is considered diagnostic information

But power in this case is **_not_** an information about the device itself. It is a feature of the device to measure the power draw **_of the connected device_**. Therefore it Is not "diagnostic".

Since the shelly devices seem to do it differently I still think this is a misclassification of the entity type.
That, or the shelly integration isn't doing it the right way.

If I am wrong please tell me why.

Koenkk, 2022-12-17:
@frenck in https://github.com/Koenkk/zigbee2mqtt/pull/9293/files#diff-c63411dead8da381e5b5e1577175477e6536b9c8d4ad997cd701a3f10d46d233R549 you added `power` (= instantaneous demand in Watt) as a diagnostic sensor. Is this this the correct? I'm not sure about this, also given the description you gave in the PR:

> An entity that provides information about the device itself (e.g., battery capacity, signal, tampering), is considered diagnostic information; thus get diagnostic is set at the entity category.

frenck, 2022-12-19:
Entity categories, like `diagnostic` or `config` are indicating "non-primary" entities.

For example, in the case of a wall plug: Its main purpose would be: To turn the device plugged into it: On/off. Therefore, the switch would be a primary entity (and thus is not categorized and has no `entity_category`).

Things like power, energy usage, voltage, power factors, child lock switches, status indicator lights, power on behavior, etc, are all considered secondary (and thus get an `entity_category` assigned).

The reason for this all is: Non-primary entities are not by default exposed to Google Assistant, Amazon Alexa, HomeKit (and other external), and they don't show up on auto-generated dashboards. It is here to keep UIs and apps clean from sensors and controls people generally don't use in their daily controls.

That said, if someone wants to add/expose a non-primary control to a voice assistant or add those specific cases to a dashboard: They can do that. The `entity_category` just controls its default behavior to keep things neat and clean out of the box.

TL;DR: Looking at this issue, I would state it is not valid, and the current working is correct as intended.

Koenkk, 2022-12-19:
Thanks for the explanation @frenck !

closed 2022-12-19 by Koenkk