Koenkk/zigbee2mqtt #15560: [New device support]: ALCHEMY_VIII - AL8RGB13W-AP

cloudbr34k84, 2022-12-15:
### Link

https://www.amazon.com.au/Downlight-Automation-Compatible-SmartThings-Lightify/dp/B07KWXFDQC

### Database entry

IEEE: 0x00124b001a53b1db Nwk: 0xBAA6 Device Type: Router LQI: Unknown RSSI: Unknown Last Seen: 2022-01-07T08:41:17 Power Source: Powered

### Comments

New device

### Supported color modes

Colour Temp, RGBWW

cloudbr34k84, 2023-01-02:
This would be great to get this device supported  as we already support another one from the same manufacturer?

cloudbr34k84, 2023-01-05:
@Koenkk hey any chance of getting this device supported??

Koenkk, 2023-01-06:
Can you provide the database.db entry of this device and the color temp range? https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

cloudbr34k84, 2023-01-06:
@Koenkk sorry still new to this, but where do i find the database.db for this device?? is that a HA database or Z2m?
when i try and get the colour range i get this error
Publish 'set' 'read' to '0x00124b001a53b1db' failed: 'Error: Read 0x00124b001a53b1db/11 lightingColorCtrl(["colorTempPhysicalMin","colorTempPhysicalMax"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

[image]

cloudbr34k84, 2023-01-06:
@Koenkk ok i think i know what you mean and i cant find the devices in the database even after restarting.
[image]

Koenkk, 2023-01-08:
Can you click on the device and provide a screenshot of the about page?

cloudbr34k84, 2023-01-08:
@Koenkk here you go
[image]

Koenkk, 2023-01-09:
Thanks, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-09 by Koenkk

cloudbr34k84, 2023-01-09:
@Koenkk Legend! appreciate it