Koenkk/zigbee2mqtt #15563: z2m logs errors when receiving null kWh from new device

bruberg, 2022-12-15:
### What happened?

A newly installed smartplug device reports null kWh, most likely because it hasn't used enough energy yet. Zigbee2mqtt logs this as errors:

```Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'energy')}```

The device's state:
```
{
    "current": 0.18,
    "energy": null,
    "indicator_mode": "off/on",
    "linkquality": 65,
    "power": 41,
    "power_outage_memory": "on",
    "state": "ON",
    "voltage": 239
}
```

I realize this is a passing error that will go away after a while, so it's not an important issue.

### What did you expect to happen?

I expect that this should not create error entries in the log.

### How to reproduce it (minimal and precise)

Install some kind of smartplug and don't let anything draw power through it.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Zigbee 3.0 USB Dongle plus

danieledwardgeorgehitchcock, 2022-12-15:
what device is this?

bruberg, 2022-12-15:
It's a [ConnectE power socket](https://en.connecte.no/stikkontakt-semi-flush), which registers as a [TuYa TS0121_plug](https://www.zigbee2mqtt.io/devices/TS0121_plug.html#tuya-ts0121_plug).

After having let it run long enough to let it calculate energy usage so it actually has a value, it still logs errors.

```
info 2022-12-15 11:58:24 MQTT publish: topic 'zigbee2mqtt/kjokken_stikkontakt_1', payload '{"current":0,"energy":0.01,"indicator_mode":"off/on","linkquality":40,"power":0,"power_outage_memory":"on","state":"OFF","voltage":235}'
error 2022-12-15 11:58:25 Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'energy')}
debug 2022-12-15 11:58:25 TypeError: Cannot read properties of undefined (reading 'energy') at Object.convert (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:647:24) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:106:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18) at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70) at ZStackAdapter.emit (node:events:513:28)
```

This is a bit weird, I have a few more of these and none of them have complained like this before.

bruberg, 2022-12-15:
Reviewing logs, it seems this is not a new error - I've only noticed when installing a new device. Looks like more of them are causing this error.

Koenkk, 2022-12-15:
This error comes from a Datek HSE2905E, can you provide the debug log a few lines before the error occurs?

bruberg, 2022-12-15:
Aha, yes, I have one of those too. Thanks for identifying it, and sorry about mistakenly suspecting the socket.

Will these log entries suffice?
```
Dec 15 21:20:00 z2mqtt Zigbee2MQTT: Received Zigbee message from 'ams_hovedhus', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":9024}' from endpoint 1 with groupID 0
Dec 15 21:20:00 z2mqtt Zigbee2MQTT: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'energy')}
```

Koenkk, 2022-12-17:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-17 by Koenkk

vahaldor, 2023-05-27:
I get the error as well. Using TUYA plugs and NOUS that reports as TUYA TS021 as well. How can I narrow down which device actually reports this error?

Zigbee2MQTT:error 2023-05-27 14:34:53: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}