Koenkk/zigbee2mqtt #15564: Sonoff Zigbee Dongle Plus CC2652P : zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'

ScheneF, 2022-12-15:
### What happened?

Hey there,

I'm running Home Assistant OS on a RaspberryPi 3.

[image]

```
Home Assistant 2022.12.6
Supervisor 2022.11.2
Operating System 9.4
Frontend 20221213.0 - latest
```

Zigbee2MQTT is not starting properly. 
According to the log, it stucks at `Starting zigbee-herdsman (0.14.76)`

---

My goal is to use my [Bosch thermostat 2](https://www.zigbee2mqtt.io/devices/BTH-RA.html) via the dongle and Zigbee2MQTT. For this I bought one from Amazon (unfortunately the Dongle-E and not the -P version). This one was not able to finish the interview with the thermostat.

After figuring out that the dongle is not recommended and experimental, I bought the Dongle-P. 

[image]

I adjusted the config and tried to start the Add-On. It always gets stuck at `Starting zigbee-herdsman (0.14.76)`.

I also tried the Edge-Version of Zigbee2MQTT (without success).
I also uninstalled Zigbee2MQTT, restarted the Home Assistant instance and reinstalled the Add-On. 
I also tried to first install the Add-On and then plug the dongle in to the Raspberry Pi.
I did not install ZHA or use the dongle in any other context than Zigbee2MQTT.

I have no further idea and I'm looking for help 😿

You can find the full log below.

I'd appreciate any help 😄 
Thank you very much :

### What did you expect to happen?

The Add-On should start normally and work.

### How to reproduce it (minimal and precise)

Install the Add-On (Zigbee2MQTT) via HACS (imported repo) and configure it as follows:

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: /dev/ttyUSB0
zigbee_herdsman_debug: true
```

--- 

Use a SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102(N) as adapter.
Try to start the Add-On.

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

cannot find out due to not working Add-On

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102(N)

### Debug log

```
[12:54:41] INFO: Preparing to start...
[12:54:43] INFO: Socat not enabled
[12:54:44] INFO: Zigbee Herdsman debug logging enabled
[12:54:50] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-12-15 12:55:03: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-12-15 12:55:03: Logging to console and directory: '/config/zigbee2mqtt/log/2022-12-15.12-55-03' filename: log.txt
Zigbee2MQTT:debug 2022-12-15 12:55:03: Removing old log directory '/config/zigbee2mqtt/log/2022-12-13.18-30-22'
Zigbee2MQTT:info  2022-12-15 12:55:03: Starting Zigbee2MQTT version 1.28.4 (commit #unknown)
Zigbee2MQTT:info  2022-12-15 12:55:03: Starting zigbee-herdsman (0.14.76)
Zigbee2MQTT:debug 2022-12-15 12:55:03: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyUSB0"}}'
2022-12-15T11:55:05.049Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-12-15T11:55:05.056Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-12-15T11:55:05.065Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-12-15T11:55:05.104Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2022-12-15T11:55:05.105Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-12-15T11:55:05.106Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-12-15T11:55:05.112Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
```

danieledwardgeorgehitchcock, 2022-12-15:
I'm not sure the device is correct in your config. Mine is `/dev/ttya` - worth a check!

ScheneF, 2022-12-15:
Hey daniel :) 

thanks for the input!

Here it is: 
[image]

Should be correct, right?  🫣

ScheneF, 2022-12-16:
I did a full new install of Home Assistant and now it works. (unfortunately my Bosch Thermostat 2 is "not supported" 😢) but Z2M runs.

closed 2022-12-16 by ScheneF

HeeroML, 2023-02-05:
> I did a full new install of Home Assistant and now it works. (unfortunately my Bosch Thermostat 2 is "not supported" 😢) but Z2M runs.

Just to add for you: it's supported. You need to go to system settings in z2mqtt and add a install code. The code is the extract of the code on the battery cover, the qr one. Than it will connect after some mins.

callarya, 2023-02-11:
Hi HeeroML

I have the same issue as it mentioned above, May I ask you to add some more detail to your solution, I cannot follow the steps.