Koenkk/zigbee2mqtt #15568: DLKZMK11LM device power source unknown

Quitsoon, 2022-12-15:
### What happened?

Hi ! This device is powered from main, but report an unknown power source

[image]

### What did you expect to happen?

It should report being powered by mains.

### How to reproduce it (minimal and precise)

Add the device.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20221102

### Adapter

  Zigbee 3.0 USB Dongle plus P (CC2652P)

Koenkk, 2022-12-15:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-15 by Koenkk

Quitsoon, 2022-12-15:
Thank you very much !