Koenkk/zigbee2mqtt #15569: [Feature request]: 1ch switch module TS110E / _TZ3210_zxbtub8r, no switch_type option

ComicBit, 2022-12-15:
### Is your feature request related to a problem? Please describe

I have 1ch, 2ch, switch modules bought from one shop. 2 channel modules support switch_type. The 1ch module unfortunately supports only switching on/off, there is no switch_type option. The problem is, that I have standard switches and the standard configuration implies that the switch is a momentary switch.

### Describe the solution you'd like

Just implementing the already present logic for the 2ch switch for the 1ch, because it's very critical form the usage

### Describe alternatives you've considered

I should change the kind of physical switches in the wall

### Additional context

Also some functionalities are not responding, eg: power_on_behavior results not working

Koenkk, 2022-12-15:
Could you check if the issue is fixed with the following external converter:  https://gist.github.com/Koenkk/864f435469f043ceb77383e0e0262613

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Koenkk, 2022-12-17:
Whooops, updated https://gist.github.com/Koenkk/864f435469f043ceb77383e0e0262613

ComicBit, 2022-12-17:
The fix works! You also added 2 other functions which are min_brightness and max_brightness, which I can set but they don't seem to change the way I can interact with the light limiting the scale at all. Is this something possible to fix you think? In any case thank you for the amazing job!

Koenkk, 2022-12-18:
It seems for the min/max brightness uses a different api. To figure this out you need to sniff the traffic when setting it via the TuYa gateway (if this device supports it at all). I've integrated the changes (without min/max brightness)

ComicBit, 2022-12-20:
I would gladly test this but I don't have a tuya hub, I will reopen the ticket in case I will have any updates in future

closed 2022-12-20 by ComicBit

doneAlexandro, 2023-01-11:
Hello! I note that "min/max_brightness" from your converter https://gist.github.com/Koenkk/864f435469f043ceb77383e0e0262613 works as it should be for a hardware switch, which is what it is for.
However, using this converter is difficult when controlled from an external switch and MQTT: "State" in the zigbee2MQTT GUI gives two switch state parameters of the same name: "state" and "state_1". One is for remote control, the other is for local. Thus, it is impossible to turn off the device remotely and turn it on locally, and vice versa. And that's the problem.
It would not be so difficult, but when the device is turned off remotely, the "brightness" parameter is reset to 0, and this is what prevents the device from turning on locally.
I apologize for the incorrect terminology in places, I'm just starting to learn zigbee2MQTT, Home Assistant and everything else.