Koenkk/zigbee2mqtt #15570: Hive SLR2b - Error on Home assistant core with system_mode: emergency_heating

dwardu89, 2022-12-15:
### What happened?

When ever i push a message to zigbee2mqtt to enable emergency_heating on my  Hive SLR2b.
Home assistant reports this error:

```
Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:621 
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22)) 
First occurred: 15:24:01 (167 occurrences) 
Last logged: 17:28:42

Invalid modes mode: emergency_heating
```

### What did you expect to happen?

I am expecting that there might be something missing in the herdsman converter, most likely ensuring that the `modes` is set, or needs to be overridden.

Its an error that shows up every time I use my on demand heating setup to enable emergency heating.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB

sjorge, 2022-12-17:
IIRC that mode is not exposed to HA deliberately, as it requires some extra stuff to be set in one go.
All Hive SLR* have section on the device page describing what you need to send.

https://www.zigbee2mqtt.io/devices/SLR2b.html#how-to-start-edit-native-boost-heat-endpoint

capandcamera, 2023-01-18:
I'm receiving the same error, my script sends the correct message (and boost is set on the device) but this error shows.

`Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:621
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 14:01:15 (8 occurrences)
Last logged: 14:06:51

Invalid modes mode: emergency_heating`

closed 2023-02-27 by github-actions[bot]

capandcamera, 2023-02-28:
Can this be re-opened as this is still an issue:

```
Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:744
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 08:43:57 (11 occurrences)
Last logged: 09:04:23

Invalid modes mode: emergency_heating