Koenkk/zigbee2mqtt #15574: Xiaomi SP-EUC01 voltage_calibration is not working

stickpin, 2022-12-15:
### What happened?

I did downgrade to the firmware v32 to avoid toggling issue (https://github.com/Koenkk/zigbee2mqtt/issues/13903).
With v41 the voltage was reporting correctly (229V) with v32 voltage being around 18V higher than expected (247V).
I've tried to set -18 in Settings (specific) -> voltage_calibration, I've got the notification that the value is successfully set, but it doesn't affect the voltage on the next reports.

Does anyone know how to fix it?

### Zigbee2MQTT version

 1.28.4

### Adapter firmware version

20221102

### Adapter

Slaesh's CC2652RB

stickpin, 2022-12-16:
@Koenkk sorry for bothering, any idea what could be the problem?
Thanks a lot in advance.

Koenkk, 2022-12-17:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-17 by Koenkk

stickpin, 2022-12-17:
@Koenkk oh, I was not aware that I was not implemented.
Thanks a lot! :)