Koenkk/zigbee2mqtt #15583: Senic EnOcean PTM 215Z paired, but no action triggered

pawelwaniek, 2022-12-16:
### What happened?

I was able to pair the Senic (Friends of Hue) Switch which is based on EnOcean PTM 215Z according to the  [documentation](https://www.zigbee2mqtt.io/devices/PTM_215Z.html).

No button press trigger an action. The actions state is "N/A"/empty and never changes.

[image]
[image]
[image]

### What did you expect to happen?

In the best case an action triggered or at least a info in the debug log

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus TI CC2652P + CP2102(N) 

### Debug log

debug 2022-12-16 10:35:47: Loaded state from file /app/data/state.json
info  2022-12-16 10:35:47: Logging to console and directory: '/app/data/log/2022-12-16.10-35-47' filename: log.txt
info  2022-12-16 10:35:47: Starting Zigbee2MQTT version 1.28.4 (commit #52e545f)
info  2022-12-16 10:35:47: Starting zigbee-herdsman (0.14.76)
debug 2022-12-16 10:35:47: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
info  2022-12-16 10:35:48: zigbee-herdsman started (resumed)
info  2022-12-16 10:35:48: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
debug 2022-12-16 10:35:48: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b002a1aa428","panID":6754}
info  2022-12-16 10:35:48: Currently 2 devices are joined:
info  2022-12-16 10:35:48: Office - Ceiling 1 - Light (0x0017880106a0fd4e): 9290022169 - Philips Hue white ambiance E27 with Bluetooth (Router)
info  2022-12-16 10:35:48: Office - Switch (0x000000000171a490): PTM 215Z - EnOcean Pushbutton transmitter module (GreenPower)
warn  2022-12-16 10:35:48: `permit_join` set to  `true` in configuration.yaml.
warn  2022-12-16 10:35:48: Allowing new devices to join.
warn  2022-12-16 10:35:48: Set `permit_join` to `false` once you joined all devices.
info  2022-12-16 10:35:48: Zigbee: allowing new devices to join.
info  2022-12-16 10:35:48: Connecting to MQTT server at mqtt://***.***.*.*
debug 2022-12-16 10:35:48: Using MQTT login with username: ******
info  2022-12-16 10:35:48: Connected to MQTT server
info  2022-12-16 10:35:48: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
info  2022-12-16 10:35:48: MQTT publish: topic 'zigbee2mqtt/office_lights/availability', payload '{"state":"online"}'
info  2022-12-16 10:35:48: Started frontend on port 0.0.0.0:8080
info  2022-12-16 10:35:48: MQTT publish: topic 'homeassistant/light/0x0017880106a0fd4e/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/Office - Ceiling 1 - Light/set","device":{"identifiers":["zigbee2mqtt_0x0017880106a0fd4e"],"manufacturer":"Philips","model":"Hue white ambiance E27 with Bluetooth (9290022169)","name":"Office - Ceiling 1 - Light","sw_version":"1.101.2"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"max_mireds":500,"min_mireds":150,"name":"Office - Ceiling 1 - Light","schema":"json","state_topic":"zigbee2mqtt/Office - Ceiling 1 - Light","supported_color_modes":["color_temp"],"unique_id":"0x0017880106a0fd4e_light_zigbee2mqtt"}'
info  2022-12-16 10:35:48: MQTT publish: topic 'homeassistant/select/0x0017880106a0fd4e/effect/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Office - Ceiling 1 - Light/set/effect","device":{"identifiers":["zigbee2mqtt_0x0017880106a0fd4e"],"manufacturer":"Philips","model":"Hue white ambiance E27 with Bluetooth (9290022169)","name":"Office - Ceiling 1 - Light","sw_version":"1.101.2"},"enabled_by_default":false,"icon":"mdi:palette","name":"Office - Ceiling 1 - Light effect","options":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"unique_id":"0x0017880106a0fd4e_effect_zigbee2mqtt"}'
info  2022-12-16 10:35:48: MQTT publish: topic 'homeassistant/sensor/0x0017880106a0fd4e/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880106a0fd4e"],"manufacturer":"Philips","model":"Hue white ambiance E27 with Bluetooth (9290022169)","name":"Office - Ceiling 1 - Light","sw_version":"1.101.2"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","name":"Office - Ceiling 1 - Light power on behavior","state_topic":"zigbee2mqtt/Office - Ceiling 1 - Light","unique_id":"0x0017880106a0fd4e_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
info  2022-12-16 10:35:48: MQTT publish: topic 'homeassistant/select/0x0017880106a0fd4e/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Office - Ceiling 1 - Light/set/power_on_behavior","device":{"identifiers":["zigbee2mqtt_0x0017880106a0fd4e"],"manufacturer":"Philips","model":"Hue white ambiance E27 with Bluetooth (9290022169)","name":"Office - Ceiling 1 - Light","sw_version":"1.101.2"},"entity_category":"config","icon":"mdi:power-settings","name":"Office - Ceiling 1 - Light power on behavior","options":["off","on","toggle","previous"],"state_topic":"zigbee2mqtt/Office - Ceiling 1 - Light","unique_id":"0x0017880106a0fd4e_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
info  2022-12-16 10:35:48: MQTT publish: topic 'homeassistant/sensor/0x0017880106a0fd4e/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880106a0fd4e"],"manufacturer":"Philips","model":"Hue white ambiance E27 with Bluetooth (9290022169)","name":"Office - Ceiling 1 - Light","sw_version":"1.101.2"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Office - Ceiling 1 - Light linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/Office - Ceiling 1 - Light","unique_id":"0x0017880106a0fd4e_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2022-12-16 10:35:48: MQTT publish: topic 'homeassistant/sensor/0x0017880106a0fd4e/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880106a0fd4e"],"manufacturer":"Philips","model":"Hue white ambiance E27 with Bluetooth (9290022169)","name":"Office - Ceiling 1 - Light","sw_version":"1.101.2"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"Office - Ceiling 1 - Light update state","state_topic":"zigbee2mqtt/Office - Ceiling 1 - Light","unique_id":"0x0017880106a0fd4e_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
info  2022-12-16 10:35:48: MQTT publish: topic 'homeassistant/binary_sensor/0x0017880106a0fd4e/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880106a0fd4e"],"manufacturer":"Philips","model":"Hue white ambiance E27 with Bluetooth (9290022169)","name":"Office - Ceiling 1 - Light","sw_version":"1.101.2"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","name":"Office - Ceiling 1 - Light update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/Office - Ceiling 1 - Light","unique_id":"0x0017880106a0fd4e_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
info  2022-12-16 10:35:48: MQTT publish: topic 'homeassistant/update/0x0017880106a0fd4e/update/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/bridge/request/device/ota_update/update","device":{"id [... cut ...] 0x000000000171a490"],"manufacturer":"EnOcean","model":"Pushbutton transmitter module (PTM 215Z)","name":"Office - Switch"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Office - Switch linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/Office - Switch","unique_id":"0x000000000171a490_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

### STARTED PRESSING BUTTINS HERE - NO LOGS AFTER ####

pawelwaniek, 2022-12-20:
I tried to pair it with a new channel and this requires a factory reset of the switch as stated in the [documentation](https://www.zigbee2mqtt.io/devices/PTM_215Z.html). After factory reset, everything works like expected

Koenkk, 2022-12-20:
Great, assuming this can be closed now.

closed 2022-12-20 by Koenkk