Koenkk/zigbee2mqtt #15585: Bosch Thermostat II (BTH-RA) sets specific temperature at random

dflvunoooooo, 2022-12-16:
### What happened?

This thermostats switch to either 17 or 21 degree on there own. I set them at a specific temp and after a few hours they change that temp without my intervention or without any automation! I now needed an automation to counter this behavior, so the room wouldn't heat up to 21 °C. This happens with the normal HA addon and the edge version.

[image]
Here is the graph for the last 24 hours. I indicated the random changes. They are only pulses because my automation sets the temp down if there is no one in the room at the time. 

### What did you expect to happen?

I expect the thermostat to keep the temperature I set for ever.

### How to reproduce it (minimal and precise)

Set any temp, wait some time.

### Zigbee2MQTT version

1.28.4-dev commit: 3796084 and 1.28.4-1

### Adapter firmware version

20211217

### Adapter

zStack3x0 (Sonoff Zigbee USB Dongle Plus)

### Debug log

Only had the normal log. will post it here, as soon as I get it.

programmdesign, 2022-12-16:
Hi. I had this issue too. It is not a zigbee2mqtt issue though. It is a bug in the firmware of the device. There were two updates  recently:

- https://www.bosch-smarthome.com/at/de/service/aktuelle-meldungen/release-infos/system-solutions/2022-10-27.html
- https://www.bosch-smarthome.com/at/de/service/aktuelle-meldungen/release-infos/system-solutions/2022-11-14.html

Unfortunately, you need a Bosch Controller, as Bosch does not support OTA updates (see discussion `[here](https://community.bosch-smarthome.com/t5/feedback-und-wünsche/frei-verfügbare-firmware-update-files/m-p/22903/emcs_t/S2h8ZW1haWx8dG9waWNfc3Vic2NyaXB0aW9ufExCUElDT1pPR1RETFpSfDIyOTAzfFNVQlNDUklQVElPTlN8aEs#M2705)`)

I bought a controller and upgraded the firmware. I'm currently checking, if the issues is solved. Unless Bosch does not enable OTA updates, I'll probably not go with Bosch devices. You have to remove the device from zigbee2mqtt first, then rest and pair each device with the Bosch Controller, wait for the upgrade, remove the device again from the Bosch controller, add it back to zigbee2mqtt. Not really cool. I was also considering to keep the Bosch controller only for the Thermostat, but the Bosch SCH core components does not allow you to control the thermostat in Home Assistant, nor did I want another unnecessary controller. Than I rather go with Homematic, as I can at least run their controller within Home Assistant. As nice as this device feels and looks, w/o OTA updates it is a bit troublesome. Beyond that, Bosch seems only to offer Matter upgrades in the future for the controller, not for the device.

FYI - I'm also testing the Aqara E1, which works great. It is a bit louder (40db) than both the Homematic (about 30db) and the Bosch (about 35 db) but does the job well and will also get a Matter upgrade. 

With all the 3 devices, you got one general issues: if you use the directly, you loosing the "magic" and algorithms that the companies provide via their controllers (e.g., using schedules, etc.)

dflvunoooooo, 2022-12-16:
Ah thank you very much for your fast and detailed answer! This is a crucial information. Then I can stop looking in the z2m logs for the source of the issue. And thanks for the detailed procedure. I will try this as soon as I can lay my hands on a Bosch controller.

OTA updates is something I didn't think of when buying this thermostats. But I will in the future. I ruled out a lot of other thermostats because of the position of the display. I see no sense in a display at the front, but this is probably my taste. I used Eurotronics before, but hat a lot of issues with two of these, so I switched to Bosch. I will not keep the Bosch controller, would only use it for this updates. 

I don't need the "magic" the controllers or apps offer. I have Home Assistant for my own magic :D

Would you be so kind as to report to me, if the update fixed this issue?

programmdesign, 2022-12-16:
@dflvunoooooo I think, you can re-open this issue. The jump to 21 degrees just happened again on my device. 

[image]

:(

programmdesign, 2022-12-16:
It seems, this behaviour happens randomly. However, I found on case that I can reproduce:

- Switch from "Auto" to "Heat" mode on the Lovelace card and then back to "Auto". After 1-2 secs, the target temp jumps to 21 degrees celcius.

dflvunoooooo, 2022-12-17:
Ah, damn! Did you have a look in the z2m log? 

Yes, for me it happened randomly, but also if the thermostat switched from off to auto.

programmdesign, 2022-12-17:
Now 17 degrees are my new 21 ;) Why ever ... 
The odd thing is the following:

**Initial state**
- system_mode: Auto
- occupied_heating_setpoint: 15

**Events**
 1. Click on "Heat" on Lovelace card and wait till the card displays the "heat" status  
 2. Click on "Auto" on Lovelace card and wait till the card displays the "auto" status  

**Published messages**

So, I have two click events, but four messages in the logs.

- I can't explain the first message, as I pressed "Heat", but at least it is just unnecessary, but didn't mess with the occupied_heating_setpoint

```
Info 2022-12-17 12:24:28MQTT publish: topic 'zigbee2mqtt/Thermostat - OG - Büro', payload '{"battery":100,"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"normal","linkquality":14,"local_temperature":15.6,"local_temperature_calibration":-0.6,"occupied_heating_setpoint":15,"system_mode":"auto","window_open":"OFF"}
```

- After, I get the following two messages, which are actually what I expected to see:

`Info 2022-12-17 12:24:33MQTT publish: topic 'zigbee2mqtt/Thermostat - OG - Büro', payload '{"battery":100,"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"normal","linkquality":14,"local_temperature":15.6,"local_temperature_calibration":-0.6,"occupied_heating_setpoint":15,"system_mode":"heat","window_open":"OFF"}'`

`Info 2022-12-17 12:24:35MQTT publish: topic 'zigbee2mqtt/Thermostat - OG - Büro', payload '{"battery":100,"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"normal","linkquality":14,"local_temperature":15.6,"local_temperature_calibration":-0.6,"occupied_heating_setpoint":15,"system_mode":"auto","window_open":"OFF"}'`

Until now, the heating set point still has the correct value. 

- Last but not least, there is a 4th message, which finally set the heating set point to 17 degrees. 

`Info 2022-12-17 12:24:35MQTT publish: topic 'zigbee2mqtt/Thermostat - OG - Büro', payload '{"battery":100,"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"normal","linkquality":14,"local_temperature":15.6,"local_temperature_calibration":-0.6,"occupied_heating_setpoint":17,"system_mode":"auto","window_open":"OFF"}'`

This final message seems to be the reason for the issue, however I'm not aware what triggers it. I made sure I turned off all my automations, to also make sure those are not causing the problem.

**Full log**

```
Debug 2022-12-17 12:24:28Received Zigbee message from 'Thermostat - OG - Büro', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":1560}' from endpoint 1 with groupID 0
Info 2022-12-17 12:24:28MQTT publish: topic 'zigbee2mqtt/Thermostat - OG - Büro', payload '{"battery":100,"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"normal","linkquality":14,"local_temperature":15.6,"local_temperature_calibration":-0.6,"occupied_heating_setpoint":15,"system_mode":"auto","window_open":"OFF"}'
Debug 2022-12-17 12:24:31Received MQTT message on 'zigbee2mqtt/Thermostat - OG - Büro/set/system_mode' with data 'heat'
Debug 2022-12-17 12:24:31Publishing 'set' 'system_mode' to 'Thermostat - OG - Büro'
Info 2022-12-17 12:24:33MQTT publish: topic 'zigbee2mqtt/Thermostat - OG - Büro', payload '{"battery":100,"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"normal","linkquality":14,"local_temperature":15.6,"local_temperature_calibration":-0.6,"occupied_heating_setpoint":15,"system_mode":"heat","window_open":"OFF"}'
Debug 2022-12-17 12:24:34Received MQTT message on 'zigbee2mqtt/Thermostat - OG - Büro/set/system_mode' with data 'auto'
Debug 2022-12-17 12:24:34Publishing 'set' 'system_mode' to 'Thermostat - OG - Büro'
Info 2022-12-17 12:24:35MQTT publish: topic 'zigbee2mqtt/Thermostat - OG - Büro', payload '{"battery":100,"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"normal","linkquality":14,"local_temperature":15.6,"local_temperature_calibration":-0.6,"occupied_heating_setpoint":15,"system_mode":"auto","window_open":"OFF"}'
Debug 2022-12-17 12:24:35Received Zigbee message from 'Thermostat - OG - Büro', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1700}' from endpoint 1 with groupID
Info 2022-12-17 12:24:35MQTT publish: topic 'zigbee2mqtt/Thermostat - OG - Büro', payload '{"battery":100,"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"normal","linkquality":14,"local_temperature":15.6,"local_temperature_calibration":-0.6,"occupied_heating_setpoint":17,"system_mode":"auto","window_open":"OFF"}'
```

programmdesign, 2022-12-17:
PS: Could it be that the other, random, behaviour is triggered as the device is unintentionally paired with another device? I had such behaviour with buttons. I had an IKEA switch, which I added to Z2M. When I pressed it, it turn off my Aqara power socket, even though I had just added the device and didn't pair anything. Maybe, this could be the cause here too if the Thermostat gets a temperature value from another device.

dflvunoooooo, 2022-12-17:
I have the exact same four messages. Very interesting. Is there a way to see where the messages came from? Is it the thermostat sending data to z2m or the other way around? 

In my case, I doubt it is paired to anything else, there isn't much to pair to. And I don't see how this would trigger the temp change.

programmdesign, 2022-12-17:
Well, a temperature sensor could just send a value to the thermostat when paired. However, it is unlikely it would do it always at the same temperature. I just had the randomness again. 21 degrees are back. As one can see in the screenshot, temperature changed at 18:45, but nothing is in the log for that time:

[image]

programmdesign, 2022-12-18:
@Koenkk, any hint for us how we can debug / investigate this issue further? 🙏

purposestesting, 2022-12-19:
This problem seems to be very wide spreaded at the moment.
https://github.com/KartoffelToby/better_thermostat/issues/791

programmdesign, 2022-12-19:
@purposestesting yes, it looks like. Although I am not sure it is related to better thermostat.

aberridg, 2022-12-20:
> This problem seems to be very wide spreaded at the moment. [KartoffelToby/better_thermostat#791](https://github.com/KartoffelToby/better_thermostat/issues/791)

Thanks - this is the issue I opened. My issue is slightly different though - the temperature doesn't randomly jump to 21 degrees. It randomly rises to the max or falls to the minimum. I have to have an automation to keep it stable. The problem does not occur with any of the other thermostats in my system, that do not use better thermostat.

programmdesign, 2022-12-20:
Interesting. I only have this particular issue with the Bosch BTH-RA. I can confirm though, that also the Bosch BTH-RA can jump up (mostly 17 or 21 degree) or down (8 degrees). The jump to 8 degree could be explained by an open window mode, however, it seems to happen also when this window wasn't opened. 

I have no temperature issues with the [Aqara E1, only with it's preset_mode](https://github.com/Koenkk/zigbee2mqtt/issues/15641). Also the Homematic EVO seems to work ok, except that it sometimes switches to "Idle" mode, for reasons I haven't yet understood. 

If you have it for all thermostats, that would hint towards and issue with Z2M and/or HASS itself. I'm not in the code though. Happy to help debug this, but that topic needs someone who has knowledge of the related code base.

programmdesign, 2022-12-22:
I'm sure, how to proceed. Overall, is seems to me, that thermostats are not working well with HASS and/or there is a bug. None of which I tested, works as it should. The best experience I got so far integration-wise is the Aqara E1, as it supports externals sensors out of the box. I now decided to return my BOSCH Smart Home devices, as it also seems, that the current Generation II devices will only by Matter compatible via their new controller. Which means, you a) have to buy a new Bosch II controller to get full thread and matter support and b) the devices themselves not support thread/matter natively (https://community.bosch-smarthome.com/t5/smart-home-news/neue-pressemeldung-zu-bosch-smart-home-und-matter/bc-p/23917/emcs_t/S2h8ZW1haWx8bWVudGlvbl9zdWJzY3JpcHRpb258TEJaNlRYMzZGMUJONUl8MjM5MTd8QVRfTUVOVElPTlN8aEs#M412) 

Happy to help if I can, but will not have a Bosch device any longer.

dflvunoooooo, 2022-12-26:
I now have one of my four thermostats switch to auto state and 17 °C once I set it to off. Very annoying!

dflvunoooooo, 2022-12-29:
It seems this is already known and not possible to change at the moment. I found a comment at a pull request [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5150#issuecomment-1353179576) and a typo corrected at the same pr [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5150#issuecomment-1354338312).

closed 2022-12-29 by dflvunoooooo

dflvunoooooo, 2022-12-30:
Setting the thermostat to 'heat' rather than 'auto' seems to solve this issue. I updated them with a Bosch Controller and reconnected them afterwards. Then I have kept two thermostats in heat and two in auto mode. Only those in auto mode switched back to 21 or 17 ° C.

programmdesign, 2023-01-03:
@dflvunoooooo does your device heat continuously if it is in 'heat' mode or does it reduce the temperature if the target temperature is reached?

dflvunoooooo, 2023-01-03:
@programmdesign No, it does not heat continuously,  Bosch seems to be different. As stated by zBBJake in the pull request: «"heat" should better be called "manuell"». The "auto" mode seems to follow the heating plan which can be configured via the Bosch app. And "heat" simply keeps the chosen temperature.

czernobill, 2024-04-28:
> Setting the thermostat to 'heat' rather than 'auto' seems to solve this issue. I updated them with a Bosch Controller and reconnected them afterwards. Then I have kept two thermostats in heat and two in auto mode. Only those in auto mode switched back to 21 or 17 ° C.

I have the same problem. However, I cannot use the heat mode. Each time I select it, it changes back to auto after a few seconds.

ManniTom, 2025-06-02:
> I have the same problem. However, I cannot use the heat mode. Each time I select it, it changes back to auto after a few seconds. <
it is not sufficient to set the state to heat.
you have to call the service.
eg from AppDeamon goes like this: 
            self.call_service("climate/set_hvac_mode", entity_id="climate.thermostat_55", hvac_mode="off")