Koenkk/zigbee2mqtt #15588: Entity 'device' is unknown, sending from openHAB

Secretman76, 2022-12-16:
### What happened?

Hi
I'm using Z2M on my Synology in Docker. was a bit difficult to get it to work because I wanted to put the volume on /data and it was always somehow not quite coherent.

But I did it and actually everything works so far.
use the _TZE200_bvu2wnxz and in openHAB I can read out all the values. also in the frontend the data are so far available and controllable.

but what I do not get at all is, the Stuerung in openHAB under the Thing/Channels, value change of the slider etc...
have at least once the slider managed, but the sends a MQTT traffic out, which then arrives with an error in the log.

manually changed value
`Debug 2022-12-16 14:53:54Received MQTT message on 'zigbee2mqtt/Radiator/current_heating_setpoint/set' with data '29'.
`
value changed via openHAB slider
`Error 2022-12-16 14:53:54Entity 'Radiator/current_heating_setpoint' is unknown
`
and I can't set the "Child lock" via openHAB either. 

the configuration.yaml looks like this. 
```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.xxxxxxxx
serial:
  port: /dev/ttyUSB0
zigbee_herdsman_debug: true
experimental:
  output: attribute
advanced:
  ikea_ota_use_test_url: true
  legacy_api: false
  log_level: debug
  homeassistant_legacy_entity_attributes: false
  legacy_availability_payload: false
frontend:
  port: 8038
  host: 0.0.0.0
  auth_token: null
  url: http://192.xxxxxxx
device_options:
  legacy: false
devices:
  '0x84baxxxxxxx':
    friendly_name: Radiator_Test
```

the herdsman folder was refreshed with the Dats from your Github. 

maybe you have a solution.

Greetings Tim

### Zigbee2MQTT version

 1.28.4 commit: 52e545f

### Adapter firmware version

 20220219

### Adapter

zStack3x0

### Debug log

Info 2022-12-16 14:53:27Network scan finished
Info 2022-12-16 14:53:27MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{"routes":false,"type":"raw","value":{"links":[{"depth":1,"linkquality":170,"lqi":170,"relationship":1,"routes":[],"source":{"ieeeAddr":"0xa4c1381e56f8bd47","networkAddress":39992},"sourceIeeeAddr":"0xa4c1381e56f8bd47","sourceNwkAddr":39992,"target":{"ieeeAddr":"0x00124b0029e8d736","networkAddress":0},"targetIeeeAddr":"0x00124b0029e8d736"},{"depth":1,"linkquality":135,"lqi":135,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x84ba20fffe97c022","networkAddress":13412},"sourceIeeeAddr":"0x84ba20fffe97c022","sourceNwkAddr":13412,"target":{"ieeeAddr":"0x00124b0029e8d736","networkAddress":0},"targetIeeeAddr":"0x00124b0029e8d736"}],"nodes":[{"definition":null,"failed":[],"friendlyName":"Coordinator","ieeeAddr":"0x00124b0029e8d736","lastSeen":null,"networkAddress":0,"type":"Coordinator"},{"definition":{"description":"Thermostatic radiator valve","model":"TS0601_thermostat_3","supports":"lock (state), battery_low, climate (current_heating_setpoint, local_temperature, system_mode, running_state, local_temperature_calibration), scale_protection, frost_protection, error, working_day, schedule, schedule_monday, schedule_tuesday, schedule_wednesday, schedule_thursday, schedule_friday, schedule_saturday, schedule_sunday, linkquality","vendor":"TuYa"},"friendlyName":"Radiator_Test","ieeeAddr":"0x84ba20fffe97c022","lastSeen":1671198770371,"manufacturerName":"_TZE200_bvu2wnxz","modelID":"TS0601","networkAddress":13412,"type":"EndDevice"}]}},"status":"ok","transaction":"t7s8t-4"}'
Debug 2022-12-16 14:53:54Received MQTT message on 'zigbee2mqtt/Radiator/current_heating_setpoint/set' with data '29'
Error 2022-12-16 14:53:54Entity 'Radiator/current_heating_setpoint' is unknown

lux73, 2022-12-16:
in your z2m Config your Zigbee Device is named '**Radiator_Test**'

but in your Debug Log you will send a message to a Device named '**Radiator**'

the Names should match at all - otherwise it would not work ^^

please show your 'Things' & 'Items' from the desired TRV

Secretman76, 2022-12-16:
Hi lux

uff, there was clearly the problem behind the screen :-D

I redid the connection and removed all elements of the adapter and cleaned up the .db, too.
Connected radiator and named it Zigbee_Radiator. thx. for the tipp 👍 

after that...

in openHab <Thing> created:
Generic MQTT Thing called <Zigbee_Radiator>

Channel added to Ting:
<Zigbee_Radiator_curent_heating_setpoint> created and connected with 
MQTT State Topic <zigbee2mqtt/Zigbee_Radiator/current_heating_setpoint>
to establish the connection.

```
UID: mqtt:topic:48d68627ee:Zigbee_Radiator
label: Zigbee_Radiator
thingTypeUID: mqtt:topic
configuration: {}
bridgeUID: mqtt:broker:48d68627ee
location: Zentrale
channels:
  - id: Zigbee_Radiator_curent_heating_setpoint
    channelTypeUID: mqtt:number
    label: Zigbee_Radiator_curent_heating_setpoint
    description: ""
    configuration:
      commandTopic: zigbee2mqtt/Zigbee_Radiator/current_heating_setpoint
      min: 5
      stateTopic: zigbee2mqtt/Zigbee_Radiator/current_heating_setpoint
      max: 35

```
The radiator is connected and Z2M also captures the channels and the corresponding values. I also have them on the slider in openHAB.
in MQTT-Explorer I see it, too.

```
Zigbee_Radiator
curent_heating_setpoint = 35
system_mode = heat
running_state = heat
current_heating_setpoint = 25
local_temperature = 22
```

in openHAB I tried it with the following in MQTT Topics: 

1.
MQTT Command Topic: zigbee2mqtt/Zigbee_Radiator/current_heating_setpoint/set
therefor I have the entity-Problem
```
Debug	Received MQTT message on 	'zigbee2mqtt/  Zigbee_Radiator/current_heating_setpoint/set' with data '19'
Error	Entity 					 'Zigbee_Radiator/current_heating_setpoint' is unknown

```
2.
MQTT Topics: zigbee2mqtt/Zigbee_Radiator/current_heating_setpoint
nothing in Z2M appears but in MQTT-Explorer. I see it setting under <curent_heating_setpoint> but no transmitting to Radiator

3.
stateTopic: zigbee2mqtt/Zigbee_Radiator/current_heating_setpointxxx (right, a "wrong" name)
sends commands witch i see in the log.

`Debug Received MQTT message on 'zigbee2mqtt/  Zigbee_Radiator/current_heating_setpoint/set' with data '19'`

so, I don´t know how it will work.  

greets Tim

lux73, 2022-12-18:
hm, i'm using OpenHAB exclusively via Text Config's: xxx.things / xxx.items / xxx.rules

i think i can't help with your WebIf integrated/configured Things - it differs a lot from my textual Config's...  sorry

but if you think it helps i can Post some snippets from my Text Files

lux73, 2022-12-18:
as a hint - your CommandTopic seems not correct!

this is my:

`commandTopic="zigbee2mqtt/thermostat-wz/set/occupied_heating_setpoint"`

you have to need to add 'set' inside

Secretman76, 2022-12-18:
Hi lux73

that was it... thank you soooo much...
was searching so long and dont find it. maybe not read it so clear...

mqtt/device/set/channel and not mqtt/device/channel/set 
ok, that ist a have to know :-D

now it works as designed :-) 

thank you!!!

greets Tim

lux73, 2022-12-18:
you're welcome - i'm always happy to help

i'm using OpenHAB for over than 3 Years - especially the beginning it was very difficult to handle ;-)

have a nice Day!

Secretman76, 2022-12-18:
yes, 
but is nice to solve the issues and when it run than it is realy smart :-)

have a nice day, too.
greets from DE :-)
Tim

closed 2022-12-18 by Secretman76