Koenkk/zigbee2mqtt #15594: [Feature request]:

milkpirate, 2022-12-16:
### Is your feature request related to a problem? Please describe

No

### Describe the solution you'd like

I once found a `configuration.yaml` with all options and description, my version looks like so:

```yaml
# Required: Home Assistant integration (MQTT discovery) (default: false)
homeassistant: false

# Required: allow new devices to join.
# WARNING: Disable this after all devices have been paired! (default: false)
permit_join: false

frontend:
  # Optional, default 8080
  port: 8881
  # Optional, default 0.0.0.0
  host: 0.0.0.0
  # Optional, enables authentication, disabled by default
#  auth_token: your-secret-token

devices: devices.yaml
# groups: groups.yaml

# Required: MQTT settings
mqtt:
  # Required: MQTT base topic for Zigbee2MQTT MQTT messages
  base_topic: zigbee2mqtt
  # Required: MQTT server URL (use mqtts:// for SSL/TLS connection)
  server: 'mqtts://mosquitto:8884'
  # Optional: absolute path to SSL/TLS certificate of CA used to sign server and client certificates (default: nothing)
  ca: /usr/local/share/ca-certificates/root_ca.crt
#  # Optional: absolute paths to SSL/TLS key and certificate for client-authentication (default: nothing)
#  key: '/etc/ssl/server.pem'
#  cert: '/etc/ssl/server.crt'
#  # Optional: MQTT server authentication user (default: nothing)
  user: "{{ mosquitto.users.zigbee2mqtt.name }}"
#  # Optional: MQTT server authentication password (default: nothing)
  password: "{{ mosquitto.users.zigbee2mqtt.pass }}"
#  # Optional: MQTT client ID (default: nothing)
#  client_id: 'zigbee2mqtt'
#  # Optional: disable self-signed SSL certificates (default: true)
#  reject_unauthorized: true
#  # Optional: Include device information to mqtt messages (default: false)
  include_device_information: true
#  # Optional: MQTT keepalive in seconds (default: 60)
#  keepalive: 60
#  # Optional: MQTT protocol version (default: 4), set this to 5 if you
#  # use the 'retention' device specific configuration
  version: 5

# Required: serial settings
serial:
  # Required: location of the adapter (e.g. CC2531).
  # To autodetect the port, set 'port: null'.
  port: /dev/ttyZigBee
#  # Optional: disable LED of the adapter if supported (default: false)
#  disable_led: false
#  # Optional: adapter type, not needed unless you are experiencing problems (default: shown below, options: zstack, deconz)
#  adapter: null

# Optional: Block devices from the network (by ieeeAddr) (default: empty)
# Previously called `ban` (which is deprecated)
#blocklist:
#  - '0x000b57fffec6a5b2'

# Optional: Allow only certain devices to join the network (by ieeeAddr)
[... 104 lines cut ...]
#        coordinator: '#e04e5d'
#        router: '#4ea3e0'
#      font:
#        coordinator: '#ffffff'
#        router: '#ffffff'
#        enddevice: '#000000'
#      line:
#        active: '#009900'
#        inactive: '#994444'

## Optional: see 'Device specific configuration' below
#device_options: {}
## Optional, see 'External converters configuration' below
#external_converters: []
```

I updated to 1.28.4 lately and was wondering if there are other/new options. I tired to find a similar version of the full blown config above but I only found partials or minimal configs. Could you supply one?

### Describe alternatives you've considered

N/A

### Additional context

N/A

ericvb, 2023-01-14:
I wish I had found this feature request several hours earlier.
It explains much better the options needed to use SSL/TLS when connecting to the MQTT broker.
The current documentation on it assumes too much user knowledge about working with certificates.

Give for example a working configuration for the well-known LetsEncrypt certificate. For example where the user can find the root certificate for LetsEncrypt, etc.

Explain the meaning of each SSL configuration line: **ca, key and cert**.
This last one is still not completely clear to me, I've configured  the 3 options, but I see above, that only one, the ca, is configured?
**ca:** the letsencrypt root certificate
**key:** my private key?
**cert:** the created cert by letsencrypt for my domain name

So please adapt also the documentation about the MQTT section.
Me and other users will be grateful :-)

milkpirate, 2023-01-14:
# My setup:
The server (mosquitto) I am connecting to uses self signed SSL certificates (that I created with openssl) to offer SSL encryption of all the traffic, like any other web page (google or so) for example. Mosquitto uses its key/cert to do the encryption. The server cert was signed with another cert, the root CA.

To give the root CA above is totally optional and you could establish an encrypted connection to a sever with a self signed certificate BUT it is not considered secure since man in the middle attacks or server spoofing is possible. In the real world the solution to this are so called chains of trust where there are trusted certificate authorities (e.g. Amazon or LetsEncrypt) that sign sub certs for web pages. So that the clients can verify that they are not redirected to a fishy site. Since in my local network I have my own DNS setup and servers I cant rely/use the public infrastructure. So I need to tell every client who wants to connect to my mosquitto server to trust ist by installing my root CA that has signed the mosquittos cert. And the connection can be encrypted and verified.

# Client Certs
An alternative solution is to create for every client a dedicated key/cert combination to authenticate against the server and encrypt the traffic. Then you wouldnt need to set up any users/passwords and would just install the client cert/key to every client who wants to connect to the server.

Since I rely on SSL with user/passwords I did not need key/cert options above. I just wrote down what came to the top of my head and there might even be inaccuracies so take everything with a grain of salt, I am not an expert myself and this is just how I understood everything. As always there are (dis)advantages to each solution of encrypting and authenticating. If you want to know more you have to check the web... or ChatGPT :D its quite good at explaining.

tim-vk, 2023-01-30:
Well, the configuration is checked through a json schema. This is all neatly kept in: [settings.schema.json](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/util/settings.schema.json). You might discuss if this is actually documentation or not (since it is not super user friendly, and not yaml but json) but it does contain anything you need.

In the case of @ericvb it has the following "documented":

```
        "ca": {
          "type": "string",
          "title": "Certificate authority",
          "requiresRestart": true,
          "description": "Absolute path to SSL/TLS certificate of CA used to sign server and client certificates",
          "examples": ["/etc/ssl/mqtt-ca.crt"]
        },
        "key": {
          "type": "string",
          "title": "SSL/TLS key",
          "requiresRestart": true,
          "description": "Absolute path to SSL/TLS key for client-authentication",
          "examples": ["/etc/ssl/mqtt-client.key"]
        },
        "cert": {
          "type": "string",
          "title": "SSL/TLS certificate",
          "description": "Absolute path to SSL/TLS certificate for client-authentication",
          "requiresRestart": true,
          "examples": ["/etc/ssl/mqtt-client.crt"]
        ]
```

closed 2023-03-09 by github-actions[bot]