Koenkk/zigbee2mqtt #15595: Bosch Smart Thermostat 2 parring problem

norbertbxl, 2022-12-16:
### What happened?

Hello
Can someone help me pair this thermostat with z2m,
I think I'm setting the pairing mode wrong, but I can't find any information on how to do it anywhere,
thank you

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

0x26720700

### Adapter

conbee2

danieledwardgeorgehitchcock, 2022-12-16:
You may need to use an install code to pair a bosch device

danieledwardgeorgehitchcock, 2022-12-16:
There may be a label with a QR code on your device - the install code should be printed on that. Use it with [this MQTT topic](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-install-code-add)

norbertbxl, 2022-12-16:
> There may be a label with a QR code on your device - the install code should be printed on that. Use it with [this MQTT topic](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-install-code-add)

A
[image]

norbertbxl, 2022-12-17:
I found this information , my conbee coordinator could be the problem?

https://github.com/Koenkk/zigbee2mqtt/pull/13780

itkama, 2022-12-28:
Correct, it currently does not work with a Conbee.

closed 2023-02-04 by github-actions[bot]