Koenkk/zigbee2mqtt #15596: Wrong product in Z2M

Liionboy, 2022-12-16:
### What happened?

Hello

Anyone know why this https://zigbee.blakadder.com/Tongou_TO-Q-SY1-JZT.html on z2m show like this https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3 ?

CIrcuit braker is 25A model.

Thanks

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

20220219

### Adapter

zStack3x0

Liionboy, 2022-12-17:
After uograde firmware via zigbee gateway from tuya now it this https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_ is there any config to change with wat is exactly?

Koenkk, 2022-12-17:
Can you provide the data/databse.db entry of this device?

Liionboy, 2022-12-17:
Hello,
This is in database:

{"id":28,"type":"Router","ieeeAddr":"0x847127fffe50a76c","nwkAddr":7219,"manufId":4098,"manufName":"_TZ3000_qeuvnohg","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,1026,57344,57345],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2674,"53251":15918,"53252":12462,"53253":3089,"powerOnBehavior":1,"switchType":0}},"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_qeuvnohg","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"rmsCurrent":560,"activePower":32,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":224}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,1]}},"msTemperatureMeasurement":{"attributes":{"measuredValue":-630}},"genOta":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c95407","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c95407","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c95407","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1671271740605,"defaultSendRequestWhen":"immediate"}

Liionboy, 2022-12-17:
an a picture how is now after upgrade via Gateway:

[image]

Koenkk, 2022-12-18:
This should already be fixed in the dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Liionboy, 2022-12-18:
And when plan to release on master?

Koenkk, 2022-12-18:
Next release is on 1 January.

Liionboy, 2022-12-18:
Thanks, i will wait until 1 January

closed 2022-12-18 by Liionboy