Koenkk/zigbee2mqtt #15597: suddenly stopped working

bobsdavenport, 2022-12-16:
### What happened?

cant open device even after removing adding usb repeatedly

device esists:
pi@raspberrypi4g:~ $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Dec 16 21:00 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0008BDE499-if00 -> ../../ttyACM0

device test ok:
pi@raspberrypi4g:~ $ test -w /dev/ttyACM0 && echo success || echo failure
success

but  

since yesterday never starts even after reboots and so on

> zigbee2mqtt@1.28.0 start
> node index.js

Zigbee2MQTT:info  2022-12-16 21:13:08: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-12-16.21-13-08' filename: log.txt
Zigbee2MQTT:info  2022-12-16 21:13:08: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)
Zigbee2MQTT:info  2022-12-16 21:13:08: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:error 2022-12-16 21:13:49: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-12-16 21:13:49: Failed to start zigbee
Zigbee2MQTT:error 2022-12-16 21:13:49: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-12-16 21:13:49: Exiting...
Zigbee2MQTT:error 2022-12-16 21:13:49: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:109:5)

### What did you expect to happen?

i expected it to start

### How to reproduce it (minimal and precise)

npm start

### Zigbee2MQTT version

latest

### Adapter firmware version

cant even start to read this

### Adapter

usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0008BDE499-if00

### Debug log

adaptor working perfectly for months, then yesterday, poof!

even adter reboots, disconnecting and reconnecting

and testing its there

i suspect the software is missing something,, as its broken before numerous times and needed reinstalls and so forth.

 but never. this sudden cessation of basics such as connecting to usb adapter.

Koenkk, 2022-12-17:
The CC2531 is known to crash (and therefore not recommended anymore). Reflashing should fix it, if this occurs more consider upgrading to a more powerful adapter.

closed 2022-12-17 by Koenkk

bobsdavenport, 2022-12-17:
no probs, actually had a sonoff dongle Plus ready & waiting, just didn't fancy all the reconfiguration.

but all done now

interestingly the same CC2531 does actually connect on anoter pi i have as a dev/test unit.

but as CC2531 has reliability issues, i suspect it better left as history.
just strange it worked for ages then all of sudden stopped.
maybe the firmware is corrupted or some other component on board has aged:)