Koenkk/zigbee2mqtt #15599: Legrand 064873 064872 Home & away switch leave the zigbee network

Loic691, 2022-12-16:
### What happened?

This device is pairing sucessfully but it leave automatically network after 10 or 20 minutes if no use.
After when pushing button the device is joining network automaticaly.
My USB dongle is Sonoff Dongle E with EFR32MG21

### What did you expect to happen?

Expect that the device keep in the network and we can see it in the list of the device in the frontend

### How to reproduce it (minimal and precise)

Just pairing the device after reseting it.
Interview is OK
Test the button and then don't touch it for 10, 20 or 30 minutes

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Sonoff ZBDongle-E (EFR32 silab chip)

Loic691, 2022-12-16:
@kirovilya For information and maybe linked with others legrand issues

kirovilya, 2022-12-18:
I am now testing on Legrand 067774 and Elelabs shield with firmware 6.10.3 - I see that after a while a message about the device leaving comes up:
```
zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 6d90012300010024db60b48a000074040004
zigbee-herdsman:adapter:ezsp:ezsp <== 0x23: {"_id_":35,"_cls_":"childJoinHandler","_isRequest_":false,"index":1,"joining":0,"childId":56100,"childEui64":{"type":"Buffer","data":[0,4,116,0,0,138,180,96]},"childType":4}
zigbee-herdsman:adapter:ezsp:debg Device left network request received: 56100 00047400008ab460
zigbee-herdsman:controller:log Device leave '0x00047400008ab460'
zigbee-herdsman:controller:log Removing device from database '0x00047400008ab460'
Device '0x00047400008ab460' left the network

```
Moreover, in the traffic sniff, there is nothing of the kind - apparently the stick itself decided to disconnect the sensor from the network.

Then I tried the stick with firmware 7.1.1 and the sensor lasted an hour or more - did not go out of the network.
I don’t know how to react to this ... according to my observations, it works on firmware 7.1.1, although it doesn’t work for you with this firmware.

Loic691, 2022-12-18:
Yes it's strange.
If you want more test it's possible but i can't sniff at this time.

Loic691, 2022-12-18:
In fact it seams Legrand 067774 and home/away 064873 have not  exactly the same leaving network behaviour.

In my prod environment 067774 leave network randomly only on push button events (or I do not saw others events)  whereas 064873 leave network 100% 10 minutes after approximatively after last push button events...

Loic691, 2023-01-02:
Hi all
Problem is similar with 1.29.0
After 15 days of using this module, some times the device don't pair aigain in the network so my button is unusable after habe leaved the network.

MichaelMure, 2023-01-12:
Did you happen to have bought the home/away switch as part of a legrand zigbee starting pack? I was under the impression that their pairing is somehow hardcoded with the provided gateway.

Loic691, 2023-01-13:
> Did you happen to have bought the home/away switch as part of a legrand zigbee starting pack? I was under the impression that their pairing is somehow hardcoded with the provided gateway.

I bought the home/away switch on second hand.
I had no difficulty to paired it with Z2M.
But my problem is the device leave the network after 10 minutes randomly.
When I pressed the buttons, sometimes I have to press several times... I suppose it's time necessary to joined again the network, configure, interview,...
It'as very strange. My opinion is not so good with legrand wireless switch...

kirovilya, 2023-02-11:
@Loic691 please check on the latest dev/edge version of z2m, I made some changes for ezsp adapter

closed 2023-03-22 by github-actions[bot]