Koenkk/zigbee2mqtt #15604: [New device support]: Aqara S3 Thermostat

masafra86, 2022-12-17:
### Link

https://www.aqara.com/Smart-Thermostat-S3_overview

### Database entry

{"id":29,"type":"Router","ieeeAddr":"0x54ef4410005e6e16","nwkAddr":33299,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.airrtc.pcacn2","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[1029,64704,3,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"dateCode":"20220225","modelId":"lumi.airrtc.pcacn2"}},"aqaraOpple":{"attributes":{"238":1248015,"247":{"type":"Buffer","data":[3,40,37,5,33,54,0,9,33,0,6,10,33,0,0,13,35,15,11,19,0,14,35,0,0,0,0,200,32,3,201,39,252,254,63,8,158,7,190,10,154,32,0]},"308":{"type":"Buffer","data":[148,52,105,229,130,242]},"527":{"type":"Buffer","data":[196,233,90,49,205,244,115,204,96,82,248,69,55,253,155,40]},"528":1,"529":24,"532":1,"533":0,"534":15,"535":3,"536":1,"545":1,"546":55,"551":1,"552":"911(68)","554":5,"572":1,"586":0,"590":3,"591":"0x0abe079e083ffefc","592":"0x0000000000000000","593":1,"594":0,"596":1,"597":80,"598":0,"599":352323840,"600":0,"601":1,"602":1,"605":1,"606":{"type":"Buffer","data":[]}}},"msRelativeHumidity":{"attributes":{"measuredValue":6100}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":769,"inClusterList":[64704],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":769,"inClusterList":[64704],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":15,"stackVersion":27,"hwVersion":1,"dateCode":"20220225","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1671271532341,"defaultSendRequestWhen":"immediate"}

### Comments

Sorry I made a mistake copying the database in the previous issue: the right device is [lumi.airrtc.pcacn2] as stated in the external converter I tried to create, and the comment in the previous issue was: 

I'm following all the steps to add this thermostat, but I'm not able to get past the pairing step, after pairing it, creating the .js file and adding it to the external converters entry in the configuration, I still get the same error.
I'm trying to modify another file about a similar device i found, that is this one:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
xiaomi: {manufacturerCode: 0x115f, disableDefaultResponse: true},
};

const tzLocal = {
ZNCJMB14LM: {
key: ['theme'
],
convertSet: async (entity, key, value, meta) => {
if (key === 'theme') {
const lookup = {'classic': 0, 'concise': 1};
await entity.write('aqaraOpple', {0x0215: {value: lookup[value], type: 0x20}}, manufacturerOptions.xiaomi);
return {state: {theme: value}};
} else {
throw new Error(Not supported: '${key}');
}
},
},
};

const fzLocal = {
ZNCJMB14LM: {
cluster: 'aqaraOpple',
type: ['attributeReport', 'readResponse'],
convert: (model, msg, publish, options, meta) => {
const result = {};
if (msg.data.hasOwnProperty(0x0215)) {
const lookup = {0: 'classic', 1: 'concise'};
result.theme = lookup[msg.data[0x0215]];
}
return result;
},
},
};

const definition = {
zigbeeModel: ['lumi.switch.n4acn4'],
model: 'ZNCJMB14LM',
vendor: 'Xiaomi',
description: 'Aqara S1 Smart Touch Panel',
fromZigbee: [fz.on_off, fzLocal.ZNCJMB14LM],
toZigbee: [tz.on_off, tzLocal.ZNCJMB14LM],
meta: {multiEndpoint: true},
endpoint: (device) => {
return {'l1': 1, 'l2': 2, 'l3': 3};
},
exposes: [
e.switch()
.withEndpoint('l1')
.setAccess('state', ea.STATE_SET),
e.switch()
.withEndpoint('l2')
.setAccess('state', ea.STATE_SET),
e.switch()
.withEndpoint('l3')
.setAccess('state', ea.STATE_SET),
exposes.enum('theme', ea.SET, ['classic', 'concise'])
.withDescription('Display theme'),
],
configure: async (device, coordinatorEndpoint, logger) => {
await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
await reporting.onOff(device.getEndpoint(1));
},
ota: ota.zigbeeOTA,
};

module.exports = definition;

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.airrtc.pcacn2'],
    model: 'SLR1',
    vendor: 'Aqara',
    description: 'Heating thermostat',
    fromZigbee: [fz.thermostat_att_report, fz.thermostat_weekly_schedule_rsp],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_system_mode, tz.thermostat_running_state,
        tz.thermostat_occupied_heating_setpoint, tz.thermostat_control_sequence_of_operation, tz.thermostat_weekly_schedule,
        tz.thermostat_clear_weekly_schedule, tz.thermostat_temperature_setpoint_hold, tz.thermostat_temperature_setpoint_hold_duration],
    exposes: [exposes.climate().withSetpoint('occupied_heating_setpoint', 7, 30, 1).withLocalTemperature()
        .withSystemMode(['off', 'auto', 'heat']).withRunningState(['idle', 'heat']).withPiHeatingDemand()],
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(5);
        const binds = ['genBasic', 'genIdentify', 'genAlarms', 'genTime', 'hvacThermostat'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(endpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
        await reporting.thermostatRunningState(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatTemperatureSetpointHold(endpoint);
        await reporting.thermostatTemperatureSetpointHoldDuration(endpoint);
    },
};

module.exports = definition;
```

masafra86, 2022-12-19:
About temperature, this should be the log lines I get when temperature changes:
Received Zigbee message from 'Aqara S3 Thermostat', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,38,5,33,54,0,9,33,0,8,10,33,211,176,13,35,15,11,19,0,14,35,0,0,0,0,200,32,3,201,39,252,254,63,24,52,8,234,11,154,32,0],"type":"Buffer"}}' from endpoint 1 with groupID null

but I'm not able to understand how to use them

masafra86, 2022-12-19:
I managed to understand the meaning of the string 

Received Zigbee message from 'Aqara S3 Thermostat', type 'attributeReport', cluster 'aqaraOpple', data '{"586":0,"591":"0x09f60a5a183ffefc"}' from endpoint 1 with groupID null

Nut I'm not able to split it into pieces of informations, due to my lack of knowledge in Javascript.
"586" is the system mode, that can be 'off' or 'heat'
"591" has many different information: 
09f6 is the temperature set on the thermostat (must be divided for 100)
0a5a is the local temperature sensed by the internal sensor (must be divided for 100)
1 is the state of the switch, that is on, but can be also 0, off
83ffefc is a mystery
can anyone translate it for me in javascript? I need fromZigbee and toZigbee

masafra86, 2023-01-10:
The following is the work I tried to do on the S3 thermostat, and needs to be implemented. I still don't know how to set and receive the temperatures and the heat/off state

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const herdsman = require('zigbee-herdsman');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {

    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},

};

const fzLocal = {
    
    ZNWKQ11LM: {
        cluster: 'aqaraOpple',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty(0x0215)) {
                const lookup = {0: 'classic', 1: 'concise'};
                result.theme = lookup[msg.data[0x0215]];
            }
            if (msg.data.hasOwnProperty(0x0258)) {
                const lookup = {0: 'classic', 1: 'normal', 2: 'desert'};
                result.wallpaper = lookup[msg.data[0x0258]];
            }    
            // this part is not working, I tried but I don't know how to
            if (msg.data.hasOwnProperty(0x024f)) {
                const data = (msg.data.hasOwnProperty(0x024f));
                const systemMode = msg.data[0x024a] === 0 ? 'off' : 'heat';

                // Extract the temperature set on the thermostat (09f6) and convert it to a number
                const temperatureSet = parseInt(msg.data[0x024f].substr(2, 4), 16) / 100;

                // Extract the local temperature sensed by the internal sensor (0a5a) and convert it to a number
                const localTemperature = parseInt(msg.data[0x024f].substr(6, 4), 16) / 100;

                // Extract the state of the switch (1) and convert it to a boolean
                const switchState = msg.data[0x024f][10] === '1';

                // const localTemperature = parseInt(data.slice(6, 10), 16)/100;
                // const switchState = parseInt(data.slice(10, 11), 16);
                
                // const setpoint = parseInt(data.slice(2, 6), 16)/100;
                result.temperatureSet;
                result.localTemperature;
                result.switchState;
            } // the following is working properly
            if (msg.data.hasOwnProperty(0x024e)) {
                const lookup = {1: 'double tube air cond', 2: 'four tubes air cond', 3: 'floor heating',4: 'fresh air fan', 5: 'double tube air cond and floor heat', 6: 'floor heating and fresh air fan'};
                result.type_external_device = lookup[msg.data[0x024e]];
            }

            if (msg.data.hasOwnProperty(0x0214)) {
                const lookup = {1: 'pointer clock', 2: 'world photos', 3: 'time of the day'};
                result.screen_saver_style = lookup[msg.data[0x0214]];
            }
            if (msg.data.hasOwnProperty(0x0213)) {
                result.standby_enabled = msg.data[0x0213] & 1 ? true : false;
            }
            if (msg.data.hasOwnProperty(0x0254)) {
                const lookup = {1: 'mute', 0: 'high'};
                result.beep_volume = lookup[msg.data[0x0254]];
            }
[... 188 lines cut ...]
        
        exposes.numeric('volume', ea.STATE_SET).withValueMin(1).withValueMax(100).withUnit('%')
            .withDescription('Volume (will not persist if Mute is enabled)'),
        exposes.numeric('lcd_brightness', ea.STATE_SET).withValueMin(1).withValueMax(100).withUnit('%')
            .withDescription('LCD brightness (will not persist if auto-brightness is enabled)'),
        exposes.enum('language', ea.STATE_SET, ['chinese', 'english']).withDescription('Interface language'),
        exposes.enum('screen_saver_style', ea.STATE_SET, ['Pointer clock', 'World photos', 'Time of the day']),
        exposes.numeric('standby_time', ea.STATE_SET).withValueMin(0).withValueMax(300).withUnit('min')
            .withDescription('Display standby time'),
        exposes.enum('font_size', ea.STATE_SET, ['small', 'medium', 'large']).withDescription('Display font size'),
        exposes.binary('lcd_auto_brightness_enabled', ea.STATE_SET, true, false).withDescription('Enable LCD auto brightness'),
        exposes.binary('night_mode', ea.STATE_SET, true, false).withDescription('Enable night mode'),
        exposes.binary('do_not_disturb', ea.STATE_SET, true, false).withDescription('Enable Do not disturb'),
        
        exposes.binary('screen_saver_enabled', ea.STATE_SET, true, false).withDescription('Enable screen saver'),
        exposes.numeric('standby_lcd_brightness', ea.STATE_SET).withValueMin(1).withValueMax(100).withUnit('%')
            .withDescription('Standby LCD brightness'),

        
    ],
    meta: {disableDefaultResponse: true},
    
};

module.exports = definition;

masafra86, 2023-02-11:
I've tried to look for other devices that may work in a similar way, and I didn't find anything. Maybe the most similar is the Eigenstone Aqara S2 thermostat. I found an open issue here https://zigbee2mqtt.discourse.group/t/support-eigenstone-thermostat-s2-ktwkq03es/1099/2 but nothing more

masafra86, 2023-02-25:
Finally I managed to sniff the traffic directly from the Aqara G3 Hub and I'm now able to compile some more things on the thermostat. Now I can set the temperature on the thermostat and turn it on and off (that is basically what it is needed for), but I still can't get the local temperature. This is the converter I tinkered :

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const herdsman = require('zigbee-herdsman');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};
const fzLocal = {
    ZNWKQ11LM: {
        cluster: 'aqaraOpple',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty(0x0215)) {
                const lookup = {0: 'classic', 1: 'concise'};
                result.theme = lookup[msg.data[0x0215]];
            }
            if (msg.data.hasOwnProperty(0x0258)) {
                const lookup = {0: 'classic', 1: 'normal', 2: 'desert'};
                result.wallpaper = lookup[msg.data[0x0258]];
            }    
            if (msg.data.hasOwnProperty(0x024f)) {
                const data = (msg.data.hasOwnProperty(0x024f));
                // Extract the temperature set on the thermostat and convert it to a number
                const occupied_heating_setpoint = parseInt(data.substr(2, 4), 16) / 100;
                // Extract the local temperature sensed by the internal sensor and convert it to a number
                const local_temperature = parseInt(data.substr(6, 4), 16) / 100;
                // Extract the state of the switch (1) and convert it to a boolean
                const switchState = data[10] === '1';
                const running_state = switchState === 0 ? 'idle' : 'heat';
                result.current_heating_setpoint = occupied_heating_setpoint;
                result.local_temperature = local_temperature;
                result.running_state = running_state;
            }
            if (msg.data.hasOwnProperty(0x024e)) {
                const lookup = {1: 'double tube air cond', 2: 'four tubes air cond', 3: 'floor heating',4: 'fresh air fan', 5: 'double tube air cond and floor heat', 6: 'floor heating and fresh air fan'};
                result.type_external_device = lookup[msg.data[0x024e]];
            }
            if (msg.data.hasOwnProperty(0x0214)) {
                const lookup = {1: 'pointer clock', 2: 'world photos', 3: 'time of the day'};
                result.screen_saver_style = lookup[msg.data[0x0214]];
            }
            if (msg.data.hasOwnProperty(0x0213)) {
                result.standby_enabled = msg.data[0x0213] & 1 ? true : false;
            }
            if (msg.data.hasOwnProperty(0x0254)) {
                const lookup = {1: 'mute', 0: 'high'};
                result.beep_volume = lookup[msg.data[0x0254]];
            }
            if (msg.data.hasOwnProperty(0x024A)) {
                const lookup = {0: 'idle', 1: 'heat'};
                result.system_mode = lookup[msg.data[0x024A]];
            }
            if (msg.data.hasOwnProperty(0x0255)) {
                result.volume = msg.data[0x0255];
            }
            if (msg.data.hasOwnProperty(0x217)) {
                const lookup = {3: 'small', 4: 'medium', 5: 'large'};
                result.font_size = lookup[msg.data[0x217]];
            }
            if (msg.data.hasOwnProperty(0x210)) {
                const lookup = {0: 'chinese', 1: 'english'};
                result.language = lookup[msg.data[0x210]];
            }
[... 125 lines cut ...]
        exposes.climate().withSetpoint('current_heating_setpoint', 16, 32, 0.5, ea.STATE_SET).withLocalTemperature(ea.STATE),
        exposes.enum('type_external_device', ea.STATE_SET, ['double tube air cond', 'four tubes air cond', 'floor heating', 'fresh air fan', 'double tube air cond and floor heat', 'floor heating and fresh air fan']).withDescription('Type of switch'),
        exposes.enum('wallpaper', ea.STATE_SET, ['classic', 'normal', 'desert']).withDescription('Wallpaper'),
        exposes.enum('beep_volume', ea.STATE_SET, ['mute', 'high']).withDescription('Beep volume'),
        exposes.enum('system_mode', ea.STATE_SET, ['on', 'off']).withDescription('System mode'),
        exposes.enum('running_state', ea.STATE, ['idle', 'heating']).withDescription('Accensione'),
        exposes.numeric('volume', ea.STATE_SET).withValueMin(1).withValueMax(100).withUnit('%')
            .withDescription('Volume (will not persist if Mute is enabled)'),
        exposes.numeric('lcd_brightness', ea.STATE_SET).withValueMin(1).withValueMax(100).withUnit('%')
            .withDescription('LCD brightness (will not persist if auto-brightness is enabled)'),
        exposes.enum('language', ea.STATE_SET, ['chinese', 'english']).withDescription('Interface language'),
        exposes.enum('screen_saver_style', ea.STATE_SET, ['Pointer clock', 'World photos', 'Time of the day']),
        exposes.numeric('standby_time', ea.STATE_SET).withValueMin(0).withValueMax(300).withUnit('min')
[... cut ...]
        exposes.binary('night_mode', ea.STATE_SET, true, false).withDescription('Enable night mode'),
        exposes.binary('do_not_disturb', ea.STATE_SET, true, false).withDescription('Enable Do not disturb'),
        exposes.binary('screen_saver_enabled', ea.STATE_SET, true, false).withDescription('Enable screen saver'),
        exposes.numeric('standby_lcd_brightness', ea.STATE_SET).withValueMin(1).withValueMax(100).withUnit('%')
            .withDescription('Standby LCD brightness'),
    ],
    meta: {disableDefaultResponse: true},
};
module.exports = definition;

masafra86, 2023-02-25:
I don't know how to solve the problem I face when I change the temperature on the climate. The device correctly changes the temperature, but on Z2MQTT I get an error "Exception while calling fromZigbee converter: data.substr is not a function}".
I think this can be related to the fact that I'm not able to see the local temperature the thermostat

closed 2023-04-05 by github-actions[bot]

teras, 2024-10-20:
So, this device is not supported? I was looking at it and liked it, but a device without support of  zigbee2mqtt is an irrelevant device for me :P

What' I find curious is that I saw it supports zigbee 3.0, and I thought all v3 devices were supposed to be compatible

Chengxuyuan0, 2024-12-12:
Looking forward to your follow-up, I'm also trying to integrate this device!

semyonchetvertnyh, 2025-01-09:
@masafra86 could you please send your latest version of the converter?

masafra86, 2025-01-10:
Well, this is my latest, buy i face up after a while
Next year, when ill have time again, ill do it again

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const herdsman = require('zigbee-herdsman');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {

    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},

};

const fzLocal = {
    
    ZNWKQ11LM: {
        cluster: 'aqaraOpple',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty(0x0215)) {
                const lookup = {0: 'classic', 1: 'concise'};
                result.theme = lookup[msg.data[0x0215]];
            }
            if (msg.data.hasOwnProperty(0x0258)) {
                const lookup = {0: 'classic', 1: 'normal', 2: 'desert'};
                result.wallpaper = lookup[msg.data[0x0258]];
            }    
            if (msg.data.hasOwnProperty(0x024f)) {
                const data = (msg.data.hasOwnProperty(0x024f));
                const systemMode = msg.data[0x024a] === 0 ? 'off' : 'heat';

                // Extract the temperature set on the thermostat (09f6) and convert it to a number
                const temperatureSet = parseInt(msg.data[0x024f].substr(2, 4), 16) / 100;

                // Extract the local temperature sensed by the internal sensor (0a5a) and convert it to a number
                const localTemperature = parseInt(msg.data[0x024f].substr(6, 4), 16) / 100;

                // Extract the state of the switch (1) and convert it to a boolean
                const switchState = msg.data[0x024f][10] === '1';

                // const localTemperature = parseInt(data.slice(6, 10), 16)/100;
                // const switchState = parseInt(data.slice(10, 11), 16);
                
                // const setpoint = parseInt(data.slice(2, 6), 16)/100;
                result.temperatureSet;
                result.localTemperature;
                result.switchState;
            }
            if (msg.data.hasOwnProperty(0x024e)) {
                const lookup = {1: 'double tube air cond', 2: 'four tubes air cond', 3: 'floor heating',4: 'fresh air fan', 5: 'double tube air cond and floor heat', 6: 'floor heating and fresh air fan'};
                result.type_external_device = lookup[msg.data[0x024e]];
            }

            if (msg.data.hasOwnProperty(0x0214)) {
                const lookup = {1: 'pointer clock', 2: 'world photos', 3: 'time of the day'};
                result.screen_saver_style = lookup[msg.data[0x0214]];
            }
            if (msg.data.hasOwnProperty(0x0213)) {
                result.standby_enabled = msg.data[0x0213] & 1 ? true : false;
            }
            if (msg.data.hasOwnProperty(0x0254)) {
                const lookup = {1: 'mute', 0: 'high'};
                result.beep_volume = lookup[msg.data[0x0254]];
            }
[... 179 lines cut ...]
        
        exposes.numeric('volume', ea.STATE_SET).withValueMin(1).withValueMax(100).withUnit('%')
            .withDescription('Volume (will not persist if Mute is enabled)'),
        exposes.numeric('lcd_brightness', ea.STATE_SET).withValueMin(1).withValueMax(100).withUnit('%')
            .withDescription('LCD brightness (will not persist if auto-brightness is enabled)'),
        exposes.enum('language', ea.STATE_SET, ['chinese', 'english']).withDescription('Interface language'),
        exposes.enum('screen_saver_style', ea.STATE_SET, ['Pointer clock', 'World photos', 'Time of the day']),
        exposes.numeric('standby_time', ea.STATE_SET).withValueMin(0).withValueMax(300).withUnit('min')
            .withDescription('Display standby time'),
        exposes.enum('font_size', ea.STATE_SET, ['small', 'medium', 'large']).withDescription('Display font size'),
        exposes.binary('lcd_auto_brightness_enabled', ea.STATE_SET, true, false).withDescription('Enable LCD auto brightness'),
        exposes.binary('night_mode', ea.STATE_SET, true, false).withDescription('Enable night mode'),
        exposes.binary('do_not_disturb', ea.STATE_SET, true, false).withDescription('Enable Do not disturb'),
        exposes.enum('homepage', ea.STATE_SET, ['scene', 'feel', 'thermostat', 'switch']).withDescription('Default display homepage'),
        exposes.binary('screen_saver_enabled', ea.STATE_SET, true, false).withDescription('Enable screen saver'),
        exposes.numeric('standby_lcd_brightness', ea.STATE_SET).withValueMin(1).withValueMax(100).withUnit('%')
            .withDescription('Standby LCD brightness'),

        
    ],
    meta: {disableDefaultResponse: true},
    
};

module.exports = definition;

semyonchetvertnyh, 2025-01-10:
@masafra86 thank you! could you please just tell what kind of features are working correctly and what's left to fix?

masafra86, 2025-01-11:
Well, as far as I remember, all the selections were properly working (theme, language, system mode…), but unfortunately, not the main one. Even if I was able to understand the string with the temperature selection, I wasn’t able to make it readable and convert it. The attribute to work on is 0x024f that is composed of:
 0x (hex number for the selected temperature)(hex number for the senses temperature)(1 or 0 for thermostat on or off)(7 digits for the operating mode)
The operating modes I understood should be (look at the last 7 digits in the examples):
Uint64: 731843958899998460 (0x0a280834013ffefc) Double tube air conditionin

Uint64: 731843959017438972 (0x0a280834083ffefc) floor heating system

Uint64: 731843958950330108 (0x0a280834043ffefc) fresh air fan

Uint64: 731843958899998460 (0x0a280834013ffefc) double tube and floor

Uint64: 731843958950330108 (0x0a280834043ffefc) floor heating and fresh air

How I was able to sniff these values? With a signer sniffer and wireshark, you may find guides online, but the important thing was that I joined the zigbee network of my Aqara Hub G3 on channel 25 with 8c:8c:e1:12:4c:03:41:80:bd:b7:3a:5b:36:5b:89:7e  Aqara Network transport. 

anyway, as I already said, I gave up with working on the thermostat temperature so I ended up paring with the Aqara hub via the Aqara app (in Chinese language because in English is not recognized), and integrate the hub in home assistant with HomeKit. At least I can work with the thermostat and it gives me the humidity (and I must admit it’s incredibly stable)