Koenkk/zigbee2mqtt #15605: [New device support]: TS0601 (_TZE200_yjjdcqsq)

aleskr, 2022-12-17:
### Link

https://de.aliexpress.com/item/1005004838591092.html?spm=a2g0o.productlist.main.23.252b4bb62YTPUI&algo_pvid=6481110b-ecc8-4ba6-9e4f-caf2270b7e93&algo_exp_id=6481110b-ecc8-4ba6-9e4f-caf2270b7e93-11&pdp_ext_f=%7B%22sku_id%22%3A%2212000030689821897%22%7D&pdp_npi=2%40dis%21EUR%2110.0%216.49%21%21%21%21%21%402145288516712737733456139d0748%2112000030689821897%21sea&curPageLogUid=iLNgz0ABkzMXem.3.41e163684uIg4A&gatewayAdapt=glo2deu

### Database entry

{"id":3,"type":"EndDevice","ieeeAddr":"0xa4c138391f751cb7","nwkAddr":52575,"manufId":4417,"manufName":"_TZE200_yjjdcqsq","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"[`0+\u0019[`0+\u0007\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1671271965036,"defaultSendRequestWhen":"immediate"}

### Comments

Can't see any information from this sensor "unsupported"

SmileyMan, 2022-12-21:
https://github.com/Koenkk/zigbee2mqtt/issues/14819

rtorrente, 2022-12-30:
Another issue that addresses this sensor : https://github.com/Koenkk/zigbee2mqtt/issues/15729

Koenkk, 2022-12-31:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/15729

closed 2022-12-31 by Koenkk

lubeat11, 2026-11-03:
In this version, hvac_modes from Home Assistant are mapped directly to the thermostat, which now exposes only off and heat.
The state switch used for enabling/disabling the thermostat is no longer necessary, as power control is now handled through hvac_modes.
The switch has been kept for backward compatibility with existing Home Assistant integrations.

[bht9000_nuovo.js](https://github.com/user-attachments/files/34322011/bht9000_nuovo.js)