Koenkk/zigbee2mqtt #15609: [Feature request]: Disable updates to be exposed to home assistant

yabbah, 2022-12-17:
### Is your feature request related to a problem? Please describe

It had been great if I was able to disable the updates for entities in home assistant. At the moment I got 90+ devices for updates. I dont want to update them either as I dont face any issue. 

### Describe the solution you'd like

No OTA updates will be shown up in Settings -> Updates in Home Assistant.

### Describe alternatives you've considered

Skip update in home assistant. But while update HA they will be back

### Additional context

No

Bascht74, 2023-01-03:
Can't you disable the update entity of the device completly? Just search for "update." within the entities...

yabbah, 2023-01-04:
> Can't you disable the update entity of the device completly? Just search for "update." within the entities...

Sure I can, but with 140+ devices it would be nice with a "Not expose updates to home assistant" in Z2M settings as I see it.

closed 2023-02-11 by github-actions[bot]