Koenkk/zigbee2mqtt #15619: Scene creation error INSUFFICIENT_SPACE

timota, 2022-12-18:
### What happened?

I'm trying to create scene for `Aqara roller shade companion E1` with `scene_store` scene service and got an error 

```
Publish 'set' 'scene_store' to '0x54ef441000443447' failed: 'Error: Scene add not succesfull ('INSUFFICIENT_SPACE')'
```

Device details:
```
Device type
EndDevice
Zigbee Model
lumi.curtain.acn002
Zigbee Manufacturer
LUMI
Description
Aqara roller shade companion E1
Support status
Supported

IEEE Address0x54ef441000443447
Network address
0x47A8
Firmware build date
11-24-2021
```

### What did you expect to happen?

Scene will be created and store on device

### How to reproduce it (minimal and precise)

1. Pair device 
2. Set desired cover position
3. Goto to device scene 
4. Press Store

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

SMLIGHT SLZB-06 Zigbee ethernet USB POE WiFi LAN adapter

### Debug log

```
2022-12-18 00:51:18Saving state to file /config/zigbee2mqtt/state.json
Debug 2022-12-18 00:51:18Received MQTT message on 'zigbee2mqtt/0x54ef441000443447/set' with data '{"scene_store":{"ID":0,"name":"Scene 0"}}'
Debug 2022-12-18 00:51:18Publishing 'set' 'scene_store' to '0x54ef441000443447'
Info 2022-12-18 00:51:19MQTT publish: topic 'zigbee2mqtt/0x54ef441000443447', payload '{"battery":47,"charging_status":false,"device":{"applicationVersion":27,"dateCode":"11-24-2021","friendlyName":"0x54ef441000443447","hardwareVersion":1,"ieeeAddr":"0x54ef441000443447","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":18344,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":16,"last_seen":"2022-12-18T00:51:19+00:00","linkquality":72,"motor_state":"pause","position":6,"power_outage_count":0,"running":false,"state":"OFF","update":{"installed_version":"unknown","state":null},"update_available":null}'
Error 2022-12-18 00:51:19Publish 'set' 'scene_store' to '0x54ef441000443447' failed: 'Error: Scene add not succesfull ('INSUFFICIENT_SPACE')'
Debug 2022-12-18 00:51:19Error: Scene add not succesfull ('INSUFFICIENT_SPACE') at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:5581:23) at runMicrotasks (<anonymous>) at processTicksAndRejections (node:internal/process/task_queues:96:5) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
```

timota, 2022-12-18:
Does it mean that device does not support scenes ?

Koenkk, 2022-12-18:
It means the device does not have any memory left to store the scene.

timota, 2022-12-18:
Thank you for a quick response.

timota, 2022-12-18:
Sorry to reopening issue. But, still can not added scenes. 
Some background info:
- no scene on device at all. 
- unable to create scenes via `scene_store` service
- able to create scene via `scene_add` service with content
```json
{
    "scene_add": {
        "ID": 3,
        "name": "Chill scene",
        "position": 6,
    }
}
```  
scene appear in GUI, but when im trying to recall a scene - nothing happened. 

What im doing wrong ? Or `Aqara roller shade companion E1` doesnt support scenes at all?

Koenkk, 2022-12-19:
`position` is not supported for scenes.

timota, 2022-12-20:
probably another stupid question - not supported by ZB or by Z2M ?

Koenkk, 2022-12-20:
I've added support for it, can you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

timota, 2022-12-21:
@Koenkk Thank you.
I have tried with latest edge (1.28.4-dev commit: [58445a2])  

Unfortunately it doesnt work. 
1. scene_store doesnt work - same error `INSUFFICIENT_SPACE` (no scene at all on device)
2. scene_add with content above - raised an error

```
 2022-12-21 09:57:42Received MQTT message on 'zigbee2mqtt/f1-r1-cover-01/set' with data '{ "scene_add": { "ID": 3, "name": "Chill scene", "position": 6 } }'
Debug 2022-12-21 09:57:42Publishing 'set' 'scene_add' to 'f1-r1-cover-01'
Info 2022-12-21 09:57:43MQTT publish: topic 'zigbee2mqtt/f1-r1-cover-01', payload '{"battery":48,"charging_status":false,"color_mode":"xy","device":{"applicationVersion":27,"dateCode":"10-09-2022","friendlyName":"f1-r1-cover-01","hardwareVersion":1,"ieeeAddr":"0x54ef441000443447","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":18344,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":17,"last_seen":"2022-12-21T09:57:43+00:00","linkquality":182,"motor_state":"pause","position":5,"power_outage_count":1,"running":false,"state":"OFF","update":{"installed_version":"unknown","state":"idle"},"update_available":null}'
Error 2022-12-21 09:57:43Publish 'set' 'scene_add' to 'f1-r1-cover-01' failed: 'TypeError: Command 0x54ef441000443447/1 genScenes.add({"groupid":0,"sceneid":3,"scenename":"","transtime":0,"extensionfieldsets":[{"clstId":258,"len":1,"extField":[6]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot read properties of undefined (reading '0'))'
Debug 2022-12-21 09:57:43TypeError: Command 0x54ef441000443447/1 genScenes.add({"groupid":0,"sceneid":3,"scenename":"","transtime":0,"extensionfieldsets":[{"clstId":258,"len":1,"extField":[6]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot read properties of undefined (reading '0')) at /app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:259:74 at Array.forEach (<anonymous>) at BuffaloZcl.writeExtensionFieldSets (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:258:28) at BuffaloZcl.write (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:597:25) at ZclFrame.writePayloadCluster (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:158:21) at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:78:18) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:389:22) at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:357:25) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42) at /app/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18
```

Koenkk, 2022-12-21:
Pushed a fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

timota, 2022-12-21:
@Koenkk thank you for a fix. It works(i can call `scene_add` with success), but position set incorrectly. 

1. `scene_add` with config:
```
{
    "scene_add": {
        "ID": 3,
        "name": "Chill scene",
        "position": 6
    }
}
```

2. recall scene. logs below
```
Debug 2022-12-21 23:12:10Received MQTT message on 'zigbee2mqtt/f1-r1-cover-01/set' with data '{"scene_recall":3}'
Debug 2022-12-21 23:12:10Publishing 'set' 'scene_recall' to 'f1-r1-cover-01'
Info 2022-12-21 23:12:11MQTT publish: topic 'zigbee2mqtt/f1-r1-cover-01', payload '{"battery":47,"charging_status":false,"color_mode":"xy","device":{"applicationVersion":27,"dateCode":"10-09-2022","friendlyName":"f1-r1-cover-01","hardwareVersion":1,"ieeeAddr":"0x54ef441000443447","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":18344,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":18,"last_seen":"2022-12-21T23:12:11+00:00","linkquality":185,"motor_state":"pause","position":0,"power_outage_count":1,"running":false,"state":"OFF","update":{"installed_version":"unknown","state":"idle"},"update_available":null}'
Debug 2022-12-21 23:12:11Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x54ef441000443447!
Info 2022-12-21 23:12:11Successfully recalled scene
Info 2022-12-21 23:12:11MQTT publish: topic 'zigbee2mqtt/f1-r1-cover-01', payload '{"battery":47,"charging_status":false,"color_mode":"xy","device":{"applicationVersion":27,"dateCode":"10-09-2022","friendlyName":"f1-r1-cover-01","hardwareVersion":1,"ieeeAddr":"0x54ef441000443447","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":18344,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":18,"last_seen":"2022-12-21T23:12:11+00:00","linkquality":185,"motor_state":"pause","position":0,"power_outage_count":1,"running":false,"state":"OFF","update":{"installed_version":"unknown","state":"idle"},"update_available":null}'
Debug 2022-12-21 23:12:12Received Zigbee message from 'f1-r1-cover-01', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":55027712,"presentValue":1}' from endpoint 1 with groupID 0
Info 2022-12-21 23:12:12MQTT publish: topic 'zigbee2mqtt/f1-r1-cover-01', payload '{"battery":47,"charging_status":false,"color_mode":"xy","device":{"applicationVersion":27,"dateCode":"10-09-2022","friendlyName":"f1-r1-cover-01","hardwareVersion":1,"ieeeAddr":"0x54ef441000443447","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":18344,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":18,"last_seen":"2022-12-21T23:12:12+00:00","linkquality":185,"motor_state":"rising","position":0,"power_outage_count":1,"running":true,"state":"OFF","update":{"installed_version":"unknown","state":"idle"},"update_available":null}'
```

3. position set to `95` instead of `6` (it can be 5, as when you set position to 5, it jumps to 6 - not a big deal)

To make it work as desired - position should be reverted, set `95` instead of `5`. Can you lineup actual position with scene behaviour, please.

Thanks

Koenkk, 2022-12-22:
Done!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

timota, 2022-12-23:
@Koenkk Thank you. 

Going to test it more deeply during holidays.  I will close this issue, in case of unexpected error - i will reopen. 

One more time thank you and have a Merry Xmas and Happy New Year.

closed 2022-12-23 by timota

Koenkk, 2022-12-23:
Great, and you too!

Based on the comment from @sjorge  in https://github.com/Koenkk/zigbee-herdsman-converters/commit/9f0136a83025e23a1a708ffa80590bb573ddfffa I did a small improvement. It should not affect the functionality in your case but can you check that the position is not inverted again?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)