Koenkk/zigbee2mqtt #15626: [New device support]: TS0202 - _TZ3000_6ygjfyll - Human Motion Sensor

nikitah4x, 2022-12-18:
### Link

https://www.aliexpress.com/item/1005003187399208.html?spm=a2g0o.order_list.order_list_main.23.2350586as8BW9c&gatewayAdapt=glo2isr

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c13802c1184273","nwkAddr":32786,"manufId":4417,"manufName":"_TZ3000_6ygjfyll","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3000_6ygjfyll","powerSource":3,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b0026b73f72","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1671378157472,"defaultSendRequestWhen":"immediate"}

### Comments

Hey,
I tried to add it by myself, but it didn't work, and I spend a lot of time trying to do it.
Please help me with it.
I edited the external converter and added the fingerprint for this device but still it doesn't work.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_dr6sduka'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_ef5xlc9q'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_vwqnz1sn'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_2b8f6cio'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_nss8amz9'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_71kfvvma'},
        {modelID: 'TS0202', manufacturerName: '_TZE200_bq5c8xfe'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_qjqgmqxr'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_mmtwjmaq'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_zwvaj5wy'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_bsvqrxru'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_tv3wxhcz'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_hqbdru35'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_otvn3lne'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_tiwq83wk'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_ykwcwxmz'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_hgu1dlak'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_6ygjfyll'},
        {modelID: 'WHD02', manufacturerName: '_TZ3000_hktqahrq'}],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    whiteLabel: [{vendor: 'Mercator Ikuü', model: 'SMA02P'},
        {vendor: 'TuYa', model: 'TY-ZPR06'},
        {vendor: 'Tesla Smart', model: 'TS0202'}],
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery(), e.battery_voltage()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        try {
            await reporting.batteryPercentageRemaining(endpoint);
        } catch (error) {/* Fails for some https://github.com/Koenkk/zigbee2mqtt/issues/13708*/}
    },
};

module.exports = definition;
```

rapejim, 2022-12-19:
@nikitah4x this master commit have your device:
https://github.com/Koenkk/zigbee-herdsman-converters/commit/ae34a0fc0d50289582c126a4991bb6aadb44b7ee

nikitah4x, 2022-12-19:
Hey,
@rapejim I fixed it by moving to the Zigbee2MQTT Edge addon instead of the regular one.

rapejim, 2022-12-19:
> Hey, @rapejim I fixed it by moving to the Zigbee2MQTT Edge addon instead of the regular one.

Then, @nikitah4x, the issue can be closed, right?

nikitah4x, 2022-12-19:
@rapejim Yes.

closed 2022-12-19 by Koenkk