Koenkk/zigbee2mqtt #15629: HomeAssistant report "Add-on is not running. Please start it first" but... Zigbee2MQTT is running!

rcxrbt, 2022-12-18:
### What happened?

Trying to open  the Zigbee2MQTT GUI, HA  report "Add-on is not running. Please start it first" but... Zigbee2MQTT is running!

[image]

Zigbee2MQTT is running also if I disable "Start on boot" option...

[image]

### What did you expect to happen?

Normal  Zigbee2MQTT  start up.

### Zigbee2MQTT version

1.28.4-1 

### Adapter firmware version

N/A at this moment

### Adapter

N/A at this moment

rcxrbt, 2022-12-18:
On GUI: Zigbee2MQTT version 1.28.4-1
On log: Zigbee2MQTT version 1.28.4 

Is it the same version?

rcxrbt, 2022-12-18:
Configuration:

```
data_path: /share/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/dev/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto
  user: xxxxxxx
  pass: xxxxxxx
serial:
  port: /dev/ttyACM0
  disable_led: true
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
advanced:
  log_level: info
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
zigbee_herdsman_debug: true
```

jasonbaptist, 2022-12-21:
This has also happened to me after the .12 HA update - Log updating with pull request from devices, but HA thinks its not running.

rcxrbt, 2022-12-21:
> This has also happened to me after the .12 HA update - Log updating with pull request from devices, but HA thinks its not running.
How did you solve it?

rcxrbt, 2022-12-21:
> This has also happened to me after the .12 HA update - Log updating with pull request from devices, but HA thinks its not running.

how did you solve it?

Catt00, 2022-12-28:
Good to see that i am not alone with this issue.

rcxrbt, 2022-12-28:
I tried to restore previous backups (up tp 4 month old)... but today, updating to "Home Assistant 2022.12.8" seems all worked fine!

closed 2023-02-04 by github-actions[bot]

Stupco, 2023-04-02:
I've just started having this problem, how did everyone fix it?

Catt00, 2023-04-02:
> I've just started having this problem, how did everyone fix it?|
> 
> Running: HA Version 2023.3.4 Z2M Version: 1.30.3-1

Test to reboot the host completely. Have been working for me.

Stupco, 2023-04-03:
> Test to reboot the host completely. Have been working for me.

Hard reboot of hardware did seem to work, with a bit of teething issues. Thanks :)