Koenkk/zigbee2mqtt #15635: No converter available for 'transition' (45) on TS0601 (WZ5_rgbw)

pankrusheff, 2022-12-19:
### What happened?

I use the [adaptive lighting](https://github.com/basnijholt/adaptive-lighting) add-on. This add-on controls the lighting and in my case, an error occurs in interaction with the [rgb controller](https://www.zigbee2mqtt.io/devices/WZ5_rgbw.html#skydance-wz5_rgbw). At the same time, lighting changes occur.

### What did you expect to happen?

Interaction does not generate errors

### How to reproduce it (minimal and precise)

Install the default adaptive lighting add-on.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

 20220219

### Adapter

zStack3x0

### Debug log

```
2022-12-19 09:02:08Received MQTT message on 'zigbee2mqtt/RGB. Mixer/set' with data '{"state":"ON","color":{"h":29.581,"s":84.314},"transition":45.0,"brightness":254}'
Debug 2022-12-19 09:02:08Publishing 'set' 'brightness' to 'RGB. Mixer'
Debug 2022-12-19 09:02:08Publishing 'set' 'state' to 'RGB. Mixer'
Error 2022-12-19 09:02:08No converter available for 'transition' (45)
Info 2022-12-19 09:02:08MQTT publish: topic 'zigbee2mqtt/RGB. Mixer', payload '{"brightness":254,"color":{"h":360,"hex":"#008018","hue":360,"s":43,"saturation":43},"color_mode":"hs","color_temp":370,"last_seen":"2022-12-19T06:01:19.566Z","linkquality":61,"state":"ON","white_brightness":199}'
Debug 2022-12-19 09:02:08Received Zigbee message from 'RGB. Mixer', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":2}],"seq":13056}' from endpoint 1 with groupID 0
Info 2022-12-19 09:02:08MQTT publish: topic 'zigbee2mqtt/RGB. Mixer', payload '{"brightness":254,"color":{"h":360,"hex":"#008018","hue":360,"s":43,"saturation":43},"color_mode":"hs","color_temp":370,"last_seen":"2022-12-19T06:02:08.748Z","linkquality":61,"state":"ON","white_brightness":199}'
Debug 2022-12-19 09:02:08Received Zigbee message from 'RGB. Mixer', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[48,49,54,56,48,49,97,101,48,51,101,52],"type":"Buffer"},"datatype":3,"dp":5}],"seq":13312}' from endpoint 1 with groupID 0
Info 2022-12-19 09:02:08MQTT publish: topic 'zigbee2mqtt/RGB. Mixer', payload '{"brightness":254,"color":{"h":360,"hex":"#008018","hue":360,"s":43,"saturation":43},"color_mode":"hs","color_temp":370,"last_seen":"2022-12-19T06:02:08.817Z","linkquality":65,"state":"ON","white_brightness":199}'
```

Koenkk, 2022-12-19:
These devices use a custom TuYa api which does not support transitions (at least as far as we know)

pankrusheff, 2022-12-20:
thx for info and great thx for Ur work.

closed 2022-12-20 by pankrusheff