Koenkk/zigbee2mqtt #15637: [Feature request]: Add support for Aqara Cube T1 Pro (Xiami CTP-R01)

dabrahim, 2022-12-19:
### Is your feature request related to a problem? Please describe

I am unable to integrate the new Aqara Cube T1 Pro (model : CTP-R01) with zigbeetomqtt in Home Assistant. The current integration only supports the older version of the cube (model: MFKZQ01LM). The pairing is successful however the device is listed as unsupported.

### Describe the solution you'd like

I'd like the device to be integrated to the xiaomi.js integrations file. 

### Describe alternatives you've considered

I've tried creating a custom integration from the already existing one (MFKZQ01LM model) however and couldn't figure out how to update it properly. For example the current implementation uses 'fz.MFKZQ01LM_action_multistate' in it's fromZigbee which I don't know the equivalent for the model CTP-R01.

### Additional context

Model name : CTP-R01

closed 2022-12-19 by dabrahim

yschen5812, 2022-12-29:
Any reason why this feature request was closed? I'm interested in the integration

jeroen-zzx, 2023-01-17:
Here is some logging I got and could relate to what I did. My converter is called CTP-R01 file.

Received when rolling over to a different side: 
To side 3:
  "message": "No converter available for 'CTP-R01' with cluster 'genMultistateInput' and type 'attributeReport' and data '{\"presentValue\":1026}'"
  "message": "No converter available for 'CTP-R01' with cluster 'genMultistateInput' and type 'attributeReport' and data '{\"presentValue\":1026}'"
  "message": "No converter available for 'CTP-R01' with cluster 'genMultistateInput' and type 'attributeReport' and data '{\"presentValue\":1026}'"
To side 1:
  "message": "No converter available for 'CTP-R01' with cluster 'genMultistateInput' and type 'attributeReport' and data '{\"presentValue\":1024}'"}
  "message": "No converter available for 'CTP-R01' with cluster 'genMultistateInput' and type 'attributeReport' and data '{\"presentValue\":1024}'"
  "message": "No converter available for 'CTP-R01' with cluster 'genMultistateInput' and type 'attributeReport' and data '{\"presentValue\":1024}'"
To side 6:
  "message": "No converter available for 'CTP-R01' with cluster 'genMultistateInput' and type 'attributeReport' and data '{\"presentValue\":1029}'"}
  
Received message when placing on a side (0=1, 1=2, etc.)
  "message": "No converter available for 'CTP-R01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{\"329\":5}'"}
  "message": "No converter available for 'CTP-R01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{\"329\":0}'"}
  "message": "No converter available for 'CTP-R01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{\"329\":4}'"}

Received message when turning right:
  "message": "No converter available for 'CTP-R01' with cluster 'genAnalogInput' and type 'attributeReport' and data '{\"267\":500,\"329\":5,\"presentValue\":105.20000457763672}'"
  "message": "Received Zigbee message from 'zigbeeCube01', type 'attributeReport', cluster 'aqaraOpple', data '{\"329\":5}' from endpoint 2 with groupID 0"
  "message": "No converter available for 'CTP-R01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{\"329\":5}'"

Received message when turning left:
  "message": "No converter available for 'CTP-R01' with cluster 'genAnalogInput' and type 'attributeReport' and data '{\"267\":500,\"329\":5,\"presentValue\":-53.55000686645508}'"
  "message": "Received Zigbee message from 'zigbeeCube01', type 'attributeReport', cluster 'aqaraOpple', data '{\"329\":5}' from endpoint 2 with groupID 0"
  "message": "No converter available for 'CTP-R01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{\"329\":5}'"

Received when shaking (looks the same as in MFKZQ01LM_action_multistate; 0=shake):
  "message": "No converter available for 'CTP-R01' with cluster 'genMultistateInput' and type 'attributeReport' and data '{\"presentValue\":0}'"
  "message": "Received Zigbee message from 'zigbeeCube01', type 'attributeReport', cluster 'genMultistateInput', data '{\"presentValue\":0}' from endpoint 2 with groupID 0"

jeroen-zzx, 2023-01-18:
Nevermind, already requested in:
https://github.com/Koenkk/zigbee2mqtt/issues/15652