Koenkk/zigbee2mqtt #15641: Aqara E1 thermostat switches to preset mode "manual" if target temperature is changed

programmdesign, 2022-12-19:
### What happened?

I'm currently testing the Aqara E1 thermostat, which is set to `operation: heat`, `preset_mode: auto`. 
[image]

If I change the target temperature using the the Lovelace card, the `preset_mode` switches automatically to `manual`.

[image]

I do get the same behaviour, if I set the target temperature directly in the settings of the device on z2m. 

I'm not 100% sure, but it feels this behaviour suddenly appeared. However, I didn't add any automation nor was there a firmware update. Does anyone have the same behaviour?

### What did you expect to happen?

The preset_mode should not change once the target temperature is adjusted.

### How to reproduce it (minimal and precise)

- Set the Thermostat to  `operation: heat`, `preset_mode: auto`. 
- Change the target temperature

### Zigbee2MQTT version

 1.28.4

### Adapter firmware version

20220219

### Adapter

Slash 

### Debug log

Debug 2022-12-19 11:44:02Publishing 'set' 'occupied_heating_setpoint' to 'Thermostat - DG - Schlafzimmer'
Info 2022-12-19 11:44:03MQTT publish: topic 'zigbee2mqtt/Thermostat - DG - Schlafzimmer', payload '{"away_preset_temperature":"12.0","battery":100,"calibrated":true,"child_lock":"UNLOCK","linkquality":116,"local_temperature":17.2,"occupied_heating_setpoint":12.5,"preset":"auto","sensor":"external","system_mode":"heat","update":{"installed_version":"0.0.0_0025","state":null},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":null,"window_detection":"OFF","window_open":false}'
Debug 2022-12-19 11:44:03Received Zigbee message from 'Thermostat - DG - Schlafzimmer', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1250}' from endpoint 1 with groupID 0
Info 2022-12-19 11:44:03MQTT publish: topic 'zigbee2mqtt/Thermostat - DG - Schlafzimmer', payload '{"away_preset_temperature":"12.0","battery":100,"calibrated":true,"child_lock":"UNLOCK","linkquality":116,"local_temperature":17.2,"occupied_heating_setpoint":12.5,"preset":"auto","sensor":"external","system_mode":"heat","update":{"installed_version":"0.0.0_0025","state":null},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":null,"window_detection":"OFF","window_open":false}'
Debug 2022-12-19 11:44:04Received Zigbee message from 'Thermostat - DG - Schlafzimmer', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1250}' from endpoint 1 with groupID 0
Info 2022-12-19 11:44:04MQTT publish: topic 'zigbee2mqtt/Thermostat - DG - Schlafzimmer', payload '{"away_preset_temperature":"12.0","battery":100,"calibrated":true,"child_lock":"UNLOCK","linkquality":120,"local_temperature":17.2,"occupied_heating_setpoint":12.5,"preset":"auto","sensor":"external","system_mode":"heat","update":{"installed_version":"0.0.0_0025","state":null},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":null,"window_detection":"OFF","window_open":false}'
Info 2022-12-19 11:44:04MQTT publish: topic 'zigbee2mqtt/Strom - OG - Büro', payload '{"consumer_connected":true,"consumption":24.76,"current":0.1677,"device_temperature":23,"energy":24.76,"illuminance":0,"illuminance_lux":0,"linkquality":105,"power":43.83,"power_outage_count":15,"power_outage_memory":true,"state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":false,"voltage":224}'
Debug 2022-12-19 11:44:04Received Zigbee message from 'Thermostat - DG - Schlafzimmer', type 'attributeReport', cluster 'aqaraOpple', data '{"626":0}' from endpoint 1 with groupID 0
Info 2022-12-19 11:44:04MQTT publish: topic 'zigbee2mqtt/Thermostat - DG - Schlafzimmer', payload '{"away_preset_temperature":"12.0","battery":100,"calibrated":true,"child_lock":"UNLOCK","linkquality":116,"local_temperature":17.2,"occupied_heating_setpoint":12.5,"preset":"manual","sensor":"external","system_mode":"heat","update":{"installed_version":"0.0.0_0025","state":null},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":null,"window_detection":"OFF","window_open":false}'

closed 2023-01-26 by github-actions[bot]

tbrasser, 2024-01-05:
I feel this is by design? Came here from your similar issue on the Bosch.

It is my understanding that:

- manual = target temperature set from external device (home assistant / Bosch app / whatever)
- auto = follow built-in scheme
- away = away target temperature

I just got the E1, my first smart TRV, Bosch is on order (and also a Nedis one), and trying to get to know them.

shrx, 2024-01-14:
Related https://github.com/Koenkk/zigbee2mqtt/issues/19070