Koenkk/zigbee2mqtt #15643: [Feature request]: Reset energy total for smartplugs like TS011F_plug_1 (trough button or switch)

malosaa, 2022-12-19:
### Is your feature request related to a problem? Please describe

I know there is a way to use the dev command.
I would rather want to see a switch/button added to Z2m so we manual can reset with an automation in homeassistant.
That would be awsome.

Regards 

### Describe the solution you'd like

Adding a button / switch where we can use an automation in homeassistant to reset it in one go, without going to the dev console.
Or just press the switch and it resets

### Describe alternatives you've considered

nope

### Additional context

thats it

Koenkk, 2022-12-19:
You can send this by publishing to `zigbee2mqtt/DEVICE_NAME/set` payload `{"reset":""}`

malosaa, 2022-12-22:
Thanks koen, i gonna try this 

> {"reset":""}

Thanks koen, but when i do that it says, no converter available for reset.

malosaa, 2022-12-23:
Thanks,

I'm getting this error now 

` 2022-12-23 14:44:59Error: Cannot read properties of undefined (reading 'includes')`

Koenkk, 2022-12-23:
Is this with z2m 1.28.4?

malosaa, 2022-12-23:
yes this one 

1.28.4-1

Koenkk, 2022-12-23:
Found the issue, updated https://github.com/Koenkk/zigbee2mqtt/issues/15643#issuecomment-1363025281

malosaa, 2022-12-24:
Perfect, it works. 
Thankyou !

Koenkk, 2022-12-25:
Thanks, added it for all devices!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-25 by Koenkk

lalik77, 2024-04-30:
Hello guys , I have Home Assistant , mqtt and zigbe2mqtt are running  in docker. The both methods
 i've tried doesn't work for me.
 - First method
[image]
[image]
- Second method from this [article](https://smarthomescene.com/guides/how-to-reset-smart-plugs-energy-total-in-zigbee2mqtt-and-zha/)

malosaa, 2024-06-06:
> Hello guys , I have Home Assistant , mqtt and zigbe2mqtt are running in docker. The both methods i've tried doesn't work for me.
> 
> * First method
>   ![2024-04-30_06-13-39](https://private-user-images.githubusercontent.com/45859153/326655929-52137191-bfc7-4966-8233-e9d1a1ed2857.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc2NzMxMzYsIm5iZiI6MTcxNzY3MjgzNiwicGF0aCI6Ii80NTg1OTE1My8zMjY2NTU5MjktNTIxMzcxOTEtYmZjNy00OTY2LTgyMzMtZTlkMWExZWQyODU3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA2VDExMjAzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMzNTcxZjc3OWEwODAxOWVjZWU2MDk0ODA5ZjNmN2E4YTkzNjZiMDE1OTRjZDI2Y2FkNjEyMzRjMGRmYmIxODAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.y2KiQWdaCE8lG-quQlBXyBpjcQXfv37l2lS61C1vDR0)
>   ![2024-04-30_06-17-30](https://private-user-images.githubusercontent.com/45859153/326655933-277065ea-711e-4faf-ac42-b45820e5fc19.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc2NzMxMzYsIm5iZiI6MTcxNzY3MjgzNiwicGF0aCI6Ii80NTg1OTE1My8zMjY2NTU5MzMtMjc3MDY1ZWEtNzExZS00ZmFmLWFjNDItYjQ1ODIwZTVmYzE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA2VDExMjAzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY1NTRjNDBkZTliZWZjZTFmMDc0MGEzMzcxYTUwMGI1MmQ0NjhlOGE5NzY4NGM2MTFiZDY1MWY1MWQ0ZDEyODAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.BlupAw2kp5LschKxI9SVXeO-9PpsFI-gvwobCQwuXnE)
> * Second method from this [article](https://smarthomescene.com/guides/how-to-reset-smart-plugs-energy-total-in-zigbee2mqtt-and-zha/)

Hi, i just see your reply.

As i understand you also want to reset the total, but the methods above doesn't work?

try this action in automations, 
Induction is my induction zigbee socket
I do this once a year, i did not try yet with the latest zigbee version, but i don't think its a problem.
```
service: mqtt.publish
data:
  topic: zigbee2mqtt/Induction/set
  payload: "{\"reset\":\"\"}"
```

geotibi, 2024-07-09:
Is it any posibility to reset this to a specific value? I would like to set this to the same value as my powermeter from electrical company so I can read the exact value without makeing any calculation.
I'm using an EARU [EAKCB-T-M-Z] circuit breaker with Zigbee model TS011F and manufacturer _TZ3000_lepzuhto.

baerakafubu, 2024-11-19:
Had the same issue with resetting my Nous a4z outdoor plug. Did it like @Koenkk and @malosaa mentioned. But i have to re-pair the plug afterwards which isn't helpful at all.

My use case: charging my PHEV via standard wall plug and resetting the consumed energy after completing the charging process.

xus-droid, 2025-03-13:
The above solutions works for TS011F_plug_1, but not for TS011F_plug_3. If I try it with to publish the MQTT Topic I get the error: extra keys not allowed @ data['payload_template']
Also the solution with dev console has no impact.

I use z2mqtt 2.1.3

Log:
info 2025-03-13 14:17:38zhc:tz: Invoked '0.0' with payload '{}'

toorisrael, 2025-03-14:
> The above solutions works for TS011F_plug_1, but not for TS011F_plug_3. If I try it with to publish the MQTT Topic I get the error: extra keys not allowed @ data['payload_template'] Also the solution with dev console has no impact.
> 
> I use z2mqtt 2.1.3 TS011F_plug_3_TZ3000_okaz9tjs
> 
> Log: info 2025-03-13 14:17:38zhc:tz: Invoked '0.0' with payload '{}'

I've got exactly the same problem

walberjunior, 2025-03-31:
I'm also unable to reset, but there are no errors in the log.

Zigbee Model
TS011F
Zigbee Manufacturer
_TZ3000_0zfrhq4i

UltraBlackLinux, 2025-05-26:
hm `reset` didn't work for me. I ran it and the socket turned off, but the energy value stayed the same. Am I missing something?

I used this:
```yaml
action: mqtt.publish
metadata: {}
data:
  qos: '2'
  retain: false
  topic: zigbee2mqtt/Plug-Cart/set
  payload: '{"reset":""}'
```

tootai, 2025-05-29:
Any news? I have TS011F_plug_3, the dev console reset

> Reset energy
To reset Sum of consumed energy, use the Dev console and execute:
Endpoint: 1
Cluster: 0x00
Command: 0
Payload: (don't change this)

has no effect.

Timsche2210, 2025-06-17:
> Do you know what would be lovely? Putting an actual button on the interface that works as expected, clicking it resets the counter to zero.
> 
> [image]
> 
> There's an awful lot of 'faff' in this thread sending custom payloads etc, when we just want to get a number back to zero...

like that idea!

chilled35, 2025-06-17:
My apologies, my comment above does read as if I'm being flippant.  I didn't realise this thread is closed...

DrSamjames, 2025-08-17:
Maybe we should re-open it as I've tried a number of time and I can't get my plug to reset:

Zigbee Model: TS011F
Zigbee Manufacturer: _TZ3000_0zfrhq4i
Firmware version: 0122052017

tootai, 2025-08-17:
Le 2025-08-17 12:38, DrSamjames a écrit :
> DrSamjames left a comment (Koenkk/zigbee2mqtt#15643) [1]
> 
> Maybe we should re-open it as I've tried a number of time and I can't
> get my plug to reset:

+1

-- 
***@***.***