Koenkk/zigbee2mqtt #15649: Wrong firmware updates

purposestesting, 2022-12-19:
### What happened?

I bought a few popp 701721 (similar to danfoss ally) thermostat valves and integrated them to home assistant. They allow ota updates which i did. They all had initial the firmware "01.12.0008 01.12". Now a few trv´s have version "01.18.0008 01.18" with firmware-date "20220928" which is the latest, some other have an unknown version "AAAABBBBCCCCDDDD" with firmware-date "20220928". I already tried to change the state.json file, change the firmware-version "AAAABBBBCCCCDDDD" to the initial 1.12 one and update again, but that did not work.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[image]
[config_entry-hassio-117a26895d5aad91aae05fdb853929e2.json.txt](https://github.com/Koe
[image]
nkk/zigbee2mqtt/files/10261041/config_entry-hassio-117a26895d5aad91aae05fdb853929e2.json.txt)

lux73, 2022-12-19:
https://github.com/Koenkk/zigbee2mqtt/issues/11049

in short: remove them from z2m & do a fresh pairing should do the Trick

purposestesting, 2022-12-19:
I updated the Sonoff  Stick to "CC1352P2_CC2652P_launchpad_coordinator_20220219" firmware. I deleted "Wohnzimmer Nord" and "Wohnzimmer Süd" and paired them again. That did not do the trick, the same different firmware versions will be displayed.

stefangehn, 2022-12-23:
You may need to restart z2m after removing the device. Apparently the firmware info even survives a device removal in z2m.

I've got 5 Danfoss Ally (same device as the Popp one) and both the firmware version as well as the firmware date was not updated after a successful OTA update (the topmost one is new and was not updated to 01.18 yet):

[image]

I guess this is rather a caching/display problem in z2m and the firmware on the device is likely to be alright. The tricky thing would be debugging this since once you've updated the firmware you cannot retry the same update easily.

twinkie0101, 2022-12-24:
I am having a similar issue but with my phillips hue bulbs.  My bulbs are on the latest firmware but says I have an update available.  Tried updating and got the following error message:

...
2022-12-23 19:16:32Update of 'Phillips Hue - Movie Room - Closet Storage' failed (Image padding contains invalid bytes)
...

But looking at the [philips hue support ](https://www.philips-hue.com/en-us/support/release-notes/lamps)page my lights have the most current firmware 1.93.11

purposestesting, 2022-12-26:
Thank you very much @stefangehn . I removed the TRV`s and then i shut down home assistant. After restart and reintegration it worked very well.

closed 2022-12-26 by purposestesting