Koenkk/zigbee2mqtt #15652: [New device support]:

dabrahim, 2022-12-19:
### Link

https://www.aqara.com/en/product/cube-t1-pro

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0x54ef44100062e05a","nwkAddr":25105,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.cagl02","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":259,"inClusterList":[0,3,6],"outClusterList":[0,3],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"20220602","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1671404743897,"defaultSendRequestWhen":"immediate"}

### Comments

I tried creating a custom converter using the one already exisiting for the older model MFKZQ01LM but I was stuck at some point, I didn't know how to properly fill the fromZigbee array in.

### External converter

```shell
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.remote.cagl02'],
    model: 'CTP-R01',
    vendor: 'Xiaomi'
    description: 'Aqara Cube T1 Pro',
    fromZigbee: [fz.xiaomi_basic],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.battery_voltage(), e.angle('action_angle'), e.device_temperature(), e.power_outage_count(false), e.cube_side('action_from_side'), e.cube_side('action_side'), e.cube_side('action_to_side'), e.cube_side('side'), e.action(['shake', 'wakeup', 'fall', 'tap', 'slide', 'flip180', 'flip90', 'rotate_left', 'rotate_right'])],
};

module.exports = definition;
```

martindell, 2022-12-27:
+1 from me - just purchased a Cube T1 Pro, not realising it's not yet supported.  I'm happy to help with figuring out how to integrate it - but creating a custom converter is beyond my (very limited) abilities

dabrahim, 2022-12-29:
> +1 from me - just purchased a Cube T1 Pro, not realising it's not yet supported.  I'm happy to help with figuring out how to integrate it - but creating a custom converter is beyond my (very limited) abilities.  It's really difficult to test too, because some cube actions seem to trigger every zigbee device on the network

From what I understand, the most important parts are the "fromZigbee" key (to process the input) and the "exposes" key (to output the entities that are displayed on HA). On the older version, they've created custom methods for the zigbee input processing. Maybe taking inspiration from there could lead us somewhere.

Koenkk, 2022-12-29:
What is logged to the debug log when triggering actions on this device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

martindell, 2022-12-29:
I couldn't get it to log anything at all when I enabled debug logging - I'm bound to be doing something wrong.  Part of the problem is that even without any external converter, the cube does weird 'stuff', messing with every other zigbee device - so there's a lot of noise to contend with.  In response to @dabrahim 's comment - I played with the converter you specified at the top of this post, but couldn't get the device to show any values in Zigbee2MQTT

[image]

JJPro, 2022-12-30:
T1 Pro is a lot different than the old cube. 
It operates in two modes: 

1. Scene Mode (default mode)
  - rotate
  - shake 
  - hold
  - side up (which side is facing up)
  - 1min inactivity 
2. Action Mode (like the classic cube)
  - slide 
  - rotate
  - tap
  - flip90, flip180
  - shake
  - 1min inactivity 

I'm new to HA, and only get to know z2m literally two weeks ago. So totally novice on coding converters. 
But I fiddled with MFKZQ01LM's code a bit, and able to get shake, rotation, hold, and side up working. 

Those satisfies most of my needs. 
I really want tap and flip to work, but that needs to change its working mode to Action Mode, and I can't figure out how. 

Here's my code, feel free to take it: 

```js
/**
  # Clusters: 

  ## Endpoint 2: 

  | Cluster            | Data                 | Description          |
  | ------------------ | -------------------- | -------------------- |
  | aqaraopple         | {329: int[0-5]}      | i side facing up     |
  | genMultistateInput | {presentValue: 0}    | action: shake        |
  | genMultistateInput | {presentValue: 4}    | action: wakeup       |
  | genMultistateInput | {presentValue: 2}    | action: fall         |
  | genMultistateInput | {presentValue: 3}    | action: rotate_right |
  | genMultistateInput | {presentValue: 1024-1029} | tap with ith side up |
  
  ## Endpoint 3: 

  | Cluster   | Data                                  | Desc                                     |
  | --------- | ------------------------------------- | ---------------------------------------- |
  | genAnalog | {267: 500, 329: 3, presentValue: -51} | 267: NA, 329: 面, presentValue: 旋转角度 |
  
   
 */

/**
 * 
 * # Two Modes
 * 
 * ## Scene Mode 
 *  - rotate
 *  - shake 
 *  - hold
 *  - side up 
 * 
 * ## Action Mode 
 *  - slide 
 *  - rotate
 *  - tap twice
 *  - flip90, flip180
 *  - shake 
 *  - trigger after one-min inactivity
 * 
 * 
 * - fromZigbee
 *  - analog ( rotation)
 *  - multistate: 
 *      - action -- presentValue: 
 *          [x] rotate_left
 *          [x] rotate_right
 *          [x] shake 0
 *          [x] hold 2
 *          [?] slide
 *          [?] flip90
 *          [?] flip180
 * 
 * - exposes
 */

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const { isLegacyEnabled } = require('zigbee-herdsman-converters/lib/utils');
const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');
const e = exposes.presets;
const ea = exposes.access;

const action_multistate = {
  cluster: 'genMultistateInput',
  type: ['attributeReport', 'readResponse'],
  options: [exposes.options.legacy()],
  convert: (model, msg, publish, options, meta) => {
    const value = msg.data['presentValue'];
    let result = null;

    if (value === 0) result = { action: 'shake' };
    else if (value === 2) result = { action: 'wakeup' };
    else if (value === 4) result = { action: 'hold' };
    else if (value >= 1024) result = { action: 'side up', side: value - 1024 };

    if (result && !isLegacyEnabled(options)) {
      delete result.to_side;
      delete result.from_side;
    }

    return result ? result : null;
  },
};

const definition = {
  zigbeeModel: ['lumi.remote.cagl02'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
  model: 'CTP-R01', // Vendor model number, look on the device for a model number
  vendor: 'Lumi', // Vendor of the device (only used for documentation and startup logging)
  description: '绿米魔方T1 Pro', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
  fromZigbee: [
    // fz.xiaomi_basic,
    action_multistate,
    fz.MFKZQ01LM_action_analog,
    // fz.xiaomi_multistate_action,
    // fz.aqara_opple,
  ], // We will add this later
  toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
  meta: { battery: { voltageToPercentage: '3V_2850_3000' } },
  exposes: [
    e.battery(),
    e.angle('action_angle'),
    e.cube_side('action_from_side'),
    e.cube_side('action_side'),
    e.cube_side('action_to_side'),
    e.cube_side('side'),
    e.cube_side('side_up'),
    e.action([
      'shake',
      'wakeup',
      'hold',
      'side_up',
      // 'tap',
      // 'slide',
      // 'flip180',
      // 'flip90',
      'rotate_left',
      'rotate_right',
    ]),
  ],
};

module.exports = definition;
```

filipeaparicio91, 2023-01-01:
I tried using the external converter code provided by @JJPro and although the cube is now correctly identified by Z2M (including sensor, configuration and diagnostic entities), they all report N/A or Unknown states.
Not only that, but when I shake the cube, it toggles all of by Zigbee lights on/off, despite not having any automation set up.

JJPro, 2023-01-01:
> they all report N/A or Unknown states
Do you mean the following: 
- action 
- action_angle
- action_side
- side_up
- action_to_side
- action_from_side

It's normal you see null values, cause they only contain values while in motion, and change very fast and won't get a chance to show in z2m frontend. If you take a look at the z2m logs or node-red debug console, you'll see their values being logged. 

> Not only that, but when I shake the cube, it toggles all of by Zigbee lights on/off, despite not having any automation set up.

I don't have this issue. Could it be the gateway you use? T1 Pro requires Zigbee 3.0. I'm using SONOFF Dongle Plus-E.

filipeaparicio91, 2023-01-01:
@JJPro, thank you for the reply.

My logs are not reporting any activity from the Cube T1 Pro.

As for the dongle, I'm using the SONOFF Dongle Plus-P, which supports Zigbee 3.0 as well.

The configuration is correct, so I'm not really sure why this is happening.

sjohannsen1, 2023-01-02:
I have a similar problem. The only activity i see in my logs is from endpoint 1 cluster "genOnOff". These messages are empty but are toggling all of my devices on/off, despite having no automation set up.

JJPro, 2023-01-03:
@filipeaparicio91 @sjohannsen1 

I've updated code in my original [comment](https://github.com/Koenkk/zigbee2mqtt/issues/15652#issuecomment-1367839008). 

Last night, I reset my cube and found all my actions failing to work. Then I figured the cube is now in action mode (which works exactly like the classic cube). 

Most likely during my last experiment, I have switched its operation mode to scene mode in Aqara Home app before pairing to Z2M, and the mode was saved. I wasn't aware of the situation, and assumed that to be the default operation mode. 

Now the code has been updated, it should pull data in correctly, and work as the classic cube. I still haven't figured how to switch between modes yet. 

---

For your issue randomly toggling other devices on/off, it yet haven't occurred to me. 
I have a hypothesis, could it be like the cube is paired to a router instead of the coordinator? and that router is not fully compatible with zigbee 3.0? I'm new to HA and Z2M, just random thoughts. Then I suggest try repairing it directly to your main coordinator.

filipeaparicio91, 2023-01-03:
@JJPro,
I tried your updated code and the behavior is the same. I was able to retrieve some debug log entries from Z2M, which say this: 
```
Debug 2023-01-03 21:52:45 - Received Zigbee message from '0x54ef44100062c386', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-01-03 21:52:45 - No converter available for 'CTP-R01' with cluster 'genOnOff' and type 'commandToggle' and data '{}'
```
I keep getting these entries when I shake the cube and occasionally when I rotate it on a random side.

I continue not getting any values reported in Z2M, even in the info logs.

As for the connection, it is currently connected through a router (which supports Zigbee 3.0), but the same behavior was observed when connected directly to the coordinator.

jzee923, 2023-01-04:
> I have a similar problem. The only activity i see in my logs is from endpoint 1 cluster "genOnOff". These messages are empty but are toggling all of my devices on/off, despite having no automation set up.

I'm having this exact same issue.

I have noticed this behavior
Flip:
`Received Zigbee message from '0x54ef4410006a47c0', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0`

Rotate:
`Received Zigbee message from '0x54ef4410006a47c0', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0`

Tap:
`Received Zigbee message from '0x54ef4410006a47c0', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0`

Attempt to switch modes by pressing the reset button 5x:
`Received Zigbee message from '0x54ef4410006a47c0', type 'attributeReport', cluster 'aqaraOpple', data '{"328":0}' from endpoint 1 with groupID 0`
This last one goes back and forth from data '{"328":0}' to data '{"328":1}'

JJPro, 2023-01-04:
Hey, I updated my code again in [original comment](https://github.com/Koenkk/zigbee2mqtt/issues/15652#issuecomment-1368512521), Now it's working on both scene mode and action mode. Pressing the LINK button 5 times to toggle between them. 

I never saw the `genOnOff`/`commandToggle` message probably because the cube needs the Aqara Home app for initial configuration, and I happen to have an Aqara hub and played with that first before pairing to Z2M and started coding my converter. I have no idea. 

I only have one cube and can't start from fresh to cover the `genOnOff` cluster personally. I don't see an option to reset the cube within Aqara Home app. and long press the LINK button and repairing doesn't give me that message either.

jzee923, 2023-01-04:
So your comment about the device being configured when your first connected it to the aqara hub got me thinking and I think I'm onto something.
I added a `configure:` section

Here's a snippet from your code with the added configure section
```
const definition = {
    zigbeeModel: ['lumi.remote.cagl02'],
    model: 'CTP-R01',
    vendor: 'Lumi',
    description: 'Aqara cube T1 Pro',
    meta: { battery: { voltageToPercentage: '3V_2850_3000' } },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff','genPowerCfg','genMultistateInput']);
    },
    fromZigbee: [aqara_opple, action_multistate, fz.MFKZQ01LM_action_analog],
```

I'm not  sure if this is complete and fully configured but I am finally starting to see the proper actions.

```
2023-01-04 01:00:02Received Zigbee message from '0x54ef4410006a47c0', type 'attributeReport', cluster 'genAnalogInput', data '{"267":500,"329":2,"presentValue":-58.40999984741211}' from endpoint 3 with groupID 0
Info 2023-01-04 01:00:02MQTT publish: topic 'zigbee2mqtt/0x54ef4410006a47c0', payload '{"action":"rotate_left","action_angle":-58.41,"action_from_side":null,"action_side":null,"action_to_side":null,"battery":null,"device_temperature":null,"linkquality":15,"operationMode":1,"operation_mode":"scene mode","power_outage_count":null,"side":null,"side_up":2,"voltage":null}'
Info 2023-01-04 01:00:02MQTT publish: topic 'zigbee2mqtt/0x54ef4410006a47c0', payload '{"action":"","action_angle":null,"action_from_side":null,"action_side":null,"action_to_side":null,"battery":null,"device_temperature":null,"linkquality":15,"operationMode":1,"operation_mode":"scene mode","power_outage_count":null,"side":null,"side_up":2,"voltage":null}'
Info 2023-01-04 01:00:02MQTT publish: topic 'zigbee2mqtt/0x54ef4410006a47c0/action', payload 'rotate_left'
```

filipeaparicio91, 2023-01-04:
@jzee923 & @JJPro

You two are getting somewhere!
Adding the configure block at first gave me a configuration error, but after switching between action and scene modes a few times, if correctly configured the cube and started reporting payloads correctly.

Commenting lines 55 & 68 was not required for my cube to start behaving correctly.

I have not tried any automation yet, but the action mode is showing some inconsistent behaviours. For example, occasionally it becomes stuck on one of the sides and does not update until a rotate or shake action happens. Using the scene mode is much more reliable, and accurately reports the side facing up every time.

Z2M logs also report rotate_left and rotate_right in both scene and action mode, but the HA action sensor only reports rotation when the cube is set to action mode.

Nevertheless, this is a massive improvement and I can finally start testing the cube without having my lights flashing all the time.

sjohannsen1, 2023-01-04:
Thank you @JJPro & @jzee923! The cube is finally reporting correctly with the addition of your code. 
Just like @filipeaparicio91 has commented, I got a configuration error that could be resolved by changing the mode a few times. 
Both modes behave consistently.
However, I noticed that the reported side was always off. After looking into the old "action_multistate"- converter, i noticed that   assignment of numbers to sides has changed. While the old cube was numbered 0-5, the new one is numbered 1-6. 

            Source: https://github.com/kirovilya/ioBroker.zigbee
                +---+
                | 3 |
            +---+---+---+
            | 5 | 1 | 2 |
            +---+---+---+
                | 4 |
                +---+
                | 6 |
                +---+
            Side 1 is with the MI logo, side 6 contains the battery door.
            
 adapted from "zigbee-herdsman-converters/converters/fromZigbee"

I added:

> `const utils = require('zigbee-herdsman-converters/lib/utils');`

and changed @JJPro converters by adding the old converter with updated sides:

 ```
const aqara_opple = {
    cluster: 'aqaraOpple',
    type: ['attributeReport', 'readResponse'],
    options: (definition) => [
      ...xiaomi.numericAttributes2Options(definition),
      exposes.enum('operation_mode', ea.STATE, ['scene mode', 'action mode']),
    ],
    convert: (model, msg, publish, options, meta) => {
      
      if (msg.data.hasOwnProperty('155') || msg.data.hasOwnProperty('328')) {
        const ops_mode = msg.data['155'] || msg.data['328'];
        meta.state.operationMode = ops_mode;
      };
      
      const operation_mode = ops_mode_lookup[meta.state.operationMode];
  
      return {
        ...xiaomi.numericAttributes2Payload(msg, meta, model, options, msg.data),
        operation_mode,
        action: 'side_up',
        side_up: msg.data['329']+1,
      };
    },
  };
  
  const action_multistate = {
    cluster: 'genMultistateInput',
    type: ['attributeReport', 'readResponse'],
    options: [exposes.options.legacy()],
    convert: (model, msg, publish, options, meta) => {
        const value = msg.data['presentValue'];
        let result;
        if (value === 0) result = { action: 'shake' };
        else if (value === 2) result = { action: 'wakeup' };
        else if (value === 4) result = { action: 'hold' };
        else if (value >= 512) result = {action: 'tap', side: value-511};
        else if (value >= 256) result = {action: 'slide', side: value-255};
        else if (value >= 128) result = {action: 'flip180', side: value-127};
        else if (value >= 64) {
            result = {
                action: 'flip90', action_from_side: Math.floor((value-64) / 8)+1, action_to_side: (value % 8)+1, action_side: (value % 8)+1,
                from_side: Math.floor((value-64) / 8)+1, to_side: (value % 8)+1, side: (value % 8)+1,
            };
        };
        else if (value >= 1024)
          result = { action: 'side_up', side_up: value - 1023 };
        if (result && !utils.isLegacyEnabled(options)) {
          delete result.to_side;
          delete result.from_side;
      };

      return result ? result : null;
      };
    };
```
Now everything works consistently and as it should for me.

filipeaparicio91, 2023-01-04:
@sjohannsen1 this works!

I had to reformat some of the else statements as Z2M was not starting, and it's now reporting the correct side every time.

Here is the full working code:
```
/**
  # Two Modes

  ## Scene Mode
    - rotate
    - shake
    - hold
    - side up
    - trigger after one-min inactivity

  ## Action Mode
    - slide
    - rotate
    - tap twice
    - flip90, flip180
    - shake
    - trigger after one-min inactivity

  # Clusters (Scene Mode): 

  ## Endpoint 2: 

  | Cluster            | Data                      | Description                   |
  | ------------------ | ------------------------- | ----------------------------- |
  | aqaraopple         | {329: 0-5}                | i side facing up              |
  | genMultistateInput | {presentValue: 0}         | action: shake                 |
  | genMultistateInput | {presentValue: 4}         | action: hold                  |
  | genMultistateInput | {presentValue: 2}         | action: wakeup                |
  | genMultistateInput | {presentValue: 1024-1029} | action: fall with ith side up |

  ## Endpoint 3: 

  | Cluster   | Data                                  | Desc                                       |
  | --------- | ------------------------------------- | ------------------------------------------ |
  | genAnalog | {267: 500, 329: 3, presentValue: -51} | 267: NA, 329: side up, presentValue: angle |

 */

  const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
  const exposes = require('zigbee-herdsman-converters/lib/exposes');
  const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');
  const utils = require('zigbee-herdsman-converters/lib/utils');
  const e = exposes.presets;
  const ea = exposes.access;
  
  const OPS_MODE = {
    ACTION: 0,
    SCENE: 1,
  };
  const ops_mode_lookup = { 0: 'action mode', 1: 'scene mode' };
  
  const aqara_opple = {
    cluster: 'aqaraOpple',
    type: ['attributeReport', 'readResponse'],
    options: (definition) => [
      ...xiaomi.numericAttributes2Options(definition),
      exposes.enum('operation_mode', ea.STATE, ['scene mode', 'action mode']),
    ],
    convert: (model, msg, publish, options, meta) => {
      
      if (msg.data.hasOwnProperty('155') || msg.data.hasOwnProperty('328')) {
        const ops_mode = msg.data['155'] || msg.data['328'];
        meta.state.operationMode = ops_mode;
      };
      
      const operation_mode = ops_mode_lookup[meta.state.operationMode];
  
      return {
        ...xiaomi.numericAttributes2Payload(msg, meta, model, options, msg.data),
        operation_mode,
        action: 'side_up',
        side_up: msg.data['329']+1,
      };
    },
  };
  
  const action_multistate = {
    cluster: 'genMultistateInput',
    type: ['attributeReport', 'readResponse'],
    options: [exposes.options.legacy()],
    convert: (model, msg, publish, options, meta) => {
        const value = msg.data['presentValue'];
        let result;
        if (value === 0) result = { action: 'shake' };
        else if (value === 2) result = { action: 'wakeup' };
        else if (value === 4) result = { action: 'hold' };
        else if (value >= 512) result = {action: 'tap', side: value-511};
        else if (value >= 256) result = {action: 'slide', side: value-255};
        else if (value >= 128) result = {action: 'flip180', side: value-127};
        else if (value >= 64) result = {action: 'flip90', action_from_side: Math.floor((value-64) / 8)+1, action_to_side: (value % 8)+1, action_side: (value % 8)+1, from_side: Math.floor((value-64) / 8)+1, to_side: (value % 8)+1, side: (value % 8)+1};
        else if (value >= 1024) result = { action: 'side_up', side_up: value - 1023 };
        if (result && !utils.isLegacyEnabled(options)) {delete result.to_side, delete result.from_side};
      return result ? result : null;
    },
  };
  
  const definition = {
    zigbeeModel: ['lumi.remote.cagl02'],
    model: 'CTP-R01',
    vendor: 'Lumi',
    description: 'Aqara cube T1 Pro',
    meta: { battery: { voltageToPercentage: '3V_2850_3000' } },
    configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await endpoint.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});
      await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff','genPowerCfg','genMultistateInput']);
    },
    fromZigbee: [aqara_opple, action_multistate, fz.MFKZQ01LM_action_analog],
    toZigbee: [],
    exposes: [
      /* Device Info */
      e.battery(),
      e.battery_voltage(),
      e.device_temperature(),
      e.power_outage_count(false),
      exposes
        .enum('operation_mode', ea.STATE, ['scene mode', 'action mode'])
        .withDescription(
          'Press LINK button 5 times to toggle between action mode and scene mode'
        ),
      /* Actions */
      e.angle('action_angle'),
      e.cube_side('action_from_side'),
      e.cube_side('action_side'),
      e.cube_side('action_to_side'),
      e.cube_side('side').withDescription('Destination side of action'),
      e.cube_side('side_up').withDescription('Upfacing side of current scene'),
      e.action([
        'shake',
        'wakeup',
        'fall',
        'tap',
        'slide',
        'flip180',
        'flip90',
        'hold',
        'side_up',
        'rotate_left',
        'rotate_right',
      ]),
    ],
  };
  
  module.exports = definition;
```

jzee923, 2023-01-04:
> As previously mentioned, I think the last missing piece is being able to change the Operation mode directly on HA or Z2M without having to open the cube.

I'm fairly certain we need to create or use an existing converter in toZigbee in order for this to work.
I tried a few things but have made no progress.
I'm not entirely sure where to go for this one.

JJPro, 2023-01-05:
@CrazyCoder no, `...` is called [spread operator](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Spread_syntax) in JS.

dabrahim, 2023-01-05:
Thank you all for your support ! I was finally able to set up the cube and trigger a few events ! The only event I'm failing to trigger so far is the "double tap" one. Aside from that everything works flawlessly on my side.

jzee923, 2023-01-06:
Been working on getting a toZigbee converter working in order to change the operating mode in the z2m ui.
[image]
I was able to get it showing up as a toggle button but it doesn't actually function

```js
const tzLocal = {
    MFCZQ12LM_scene_mode: {
        key: ['operation_mode'],
        convertSet: async (entity, key, value, meta) => {
            // Support existing syntax of a nested object just for the state field. Though it's quite silly IMO.
            const targetValue = value.hasOwnProperty('state') ? value.state : value;
            // Switches using aqaraOpple 0x0200 on the same endpoints as the onOff clusters.
            const lookupState = {"action": 0, "scene": 1};
            await entity.write('aqaraOpple', {0x0148: {value: lookupState[targetValue], type: 0x20}}, {manufacturerCode: 0x115f});
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('aqaraOpple', [0x0148], {manufacturerCode: 0x115f});
        },
    },
}
```
I'm having issues with this line here
`await entity.write('aqaraOpple', {0x0148: {value: lookupState[targetValue], type: 0x20}}, {manufacturerCode: 0x115f});`

If anyone has a sniffer they can use to get the correct values, I think we'll be one step closer to a fully functional device

I've also cleaned up the code a bit so that it better matches the samples provided in the docs. Most of this has just been a lot of trial and error so i'm not sure if everything is correct.

```js
/**
# Two Modes

## Scene Mode
    - rotate
    - shake
    - hold
    - side up
    - trigger after one-min inactivity

## Action Mode
    - slide
    - rotate
    - tap twice
    - flip90, flip180
    - shake
    - trigger after one-min inactivity

# Clusters (Scene Mode): 

## Endpoint 2: 

| Cluster            | Data                      | Description                   |
| ------------------ | ------------------------- | ----------------------------- |
| aqaraopple         | {329: 0-5}                | i side facing up              |
| genMultistateInput | {presentValue: 0}         | action: shake                 |
| genMultistateInput | {presentValue: 4}         | action: hold                  |
| genMultistateInput | {presentValue: 2}         | action: wakeup                |
| genMultistateInput | {presentValue: 1024-1029} | action: fall with ith side up |

## Endpoint 3: 

| Cluster   | Data                                  | Desc                                       |
| --------- | ------------------------------------- | ------------------------------------------ |
| genAnalog | {267: 500, 329: 3, presentValue: -51} | 267: NA, 329: side up, presentValue: angle |

*/

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
[... 126 lines cut ...]
        /* Actions */
        e.angle('action_angle'),
        e.cube_side('action_from_side'),
        e.cube_side('action_side'),
        e.cube_side('action_to_side'),
        e.cube_side('side').withDescription('Destination side of action'),
        e.cube_side('side_up').withDescription('Upfacing side of current scene'),
        e.action([
            'shake',
            'wakeup',
            'fall',
            'tap',
            'slide',
            'flip180',
            'flip90',
            'hold',
            'side_up',
            'rotate_left',
            'rotate_right',
        ]),
    ],
};

module.exports = definition;
```

JJPro, 2023-01-06:
@jzee923 

> `await entity.write('aqaraOpple', {0x0148: {value: lookupState[targetValue], type: 0x20}}, {manufacturerCode: 0x115f});`

How do you find the data points (0x0148, 0x20) to overwrite? I'm still new to z2m, there's a lot to learn. Thanks~

Ekinox09, 2023-01-06:
Hi, thanks all for your great job. I've started a thread (#15956) yesterday for this device but your job is much more advanced.
I've tested your code on a external converter and it works perfectly (plugin zigbee2MQTT on Jeedom software).
Here are my suggestion and observation:
- Suggestion: Would be great to have the sides numbered from 1 to 6 (because they are numbered this way, physically, on the cube). That would really help.
- Observation: In "action mode" my device is not reliable (on Z2M or deCONZ). Some movements are not detected, randomly, during a short period. I've noticed that a "tap" or a "shake" is always detected. But a "flip" or a "slide" are not; especially after a sleep period. On the contrary, in the "scene mode", it's much more reliable and the "side_up" is near to 100% with my device. So I will continue this way (I use the cube for managing the house modes (sleep, wakeup, leaving, coming home); unfortunately, I will have to forget "slide" and "tap" movements...

Ekinox09, 2023-01-06:
> ```js
>             //if( msg.data.hasOwnProperty(329) ){
>                 //payload.side = msg.data[329] + 1;
>             //}
> ```

I would suggest not to comment this part because (or to rework it), for me, this "side_up" information is the mode reliable on in the "scene mode".

jzee923, 2023-01-06:
That data is duplicated in aqaraOpple and the multistate. I'm not sure why, but that's the reason i commented out the aqaraopple one.

JJPro, 2023-01-06:
> That data is duplicated in aqaraOpple and the multistate. I'm not sure why, but that's the reason i commented out the aqaraopple one.

aqaraOpple is mainly for accurately detecting side_up in scene mode when there is no action. 
E.g. when you put the cube down on table super super gently. In this case, aqaraOpple's sensor value will change, but multistate won't. 
While multistate is for situation there is an action (fliping, put down with noticible movement). 

To me, cluster multistate  is carried over from classic cube, and cluster aqaraOpple is specifically created for scene mode.

jzee923, 2023-01-06:
Ok yeah that makes a lot of sense.

JJPro, 2023-01-06:
@Ekinox09 You can try my original converter code [above](https://github.com/Koenkk/zigbee2mqtt/issues/15652#issuecomment-1367839008)
Others' code are improvements upon that. 

My converter has been been working flawlessly for me, and I barely experienced missed actions. 

The only thing missing in my code is the configuration part. But since your cube is already set up and working, it doesn't need to initial configuration process any more, so should be working fine with it. 

Give it a try.

Ekinox09, 2023-01-06:
@JJPro 
- I've made my test on debug mode from the console and, when I observe the missed actions, nothing is logged. My guess is that it comes from the device itself and not the converter (which is applied, i think, after the  RAW messages we have in the debug console). But I will test it in order to be sure.
- I've just tested your file at I have the same result. I have missed actions and it comes from the device itself (nothing seems to be catched by the controller (no zcl data)). The problem can be reproduced; mainly with the flip and rotate. tap or shake are ALWAYS detected. After some tests, I miss some actions if I handle the cude "too gently" (which can happen if you look few seconds to the face you want to have on, and land the device softly).
- In the scene mode, the "side_up" event, launched after the end of any movement (even gentle), provide me 99,99% of the time the good info.
Will have to stay on this mode or to me more rough with the device ;o)
Note: I will keep your code for a while; if you need other tests, let me know. Thanks for your work.

JJPro, 2023-01-06:
@Ekinox09 

> After some tests, I miss some actions if I handle the cude "too gently" (which can happen if you look few seconds to the face you want to have on, and land the device softly).

That's normal in action mode, the classic cube behaves the same way. 
Gentle movement won't be treated as a valid action. I think that's be design, in order to avoid false positives when people are having fun fiddling with the cube in hand. 
Yeah I am this person like to hold my cube from time to time. haha. If I indeed want to trigger an action I put it down with noticeable movements; otherwise I put it down gently. 

That's for action mode. 

As for scene mode, the purpose of scene mode is to simply tell which side is up, so the side_up is always accurate, even with super subtle resting move. 

From my experiment: 
scene (side up) data is from cluster 1, used only in scene mode
action (flip, tap, slide) data is from cluster 2, used only in action mode
analog (rotate, shake) data is from cluster 3, and this is shared in both modes
That's why we got rotation and shaking in both modes. 

You can think cluster as a program. So basically, there are three programs running in cube reporting stuff to z2m, and each program is responsible for reporting different attributes.

JJPro, 2023-01-06:
BTW, I didn't integrate other's configuration code to my converter, simply because I don't understand that code. Like what it's doing, why the binding with the coordinator, ...stuff like that. 

I've been learning about zigbee protocol recently.

jzee923, 2023-01-06:
Yeah, that was stuff I was trying in order to get my cube configured properly.
Without pairing to an aqara hub like you did the cube was not behaving properly.
It feels like we've gotten about as far as we can unless someone with a sniffer can come and help us with all the proper data.
Like i said before, most of this was trial and error. The documentation is pretty lacking on what all of this does.

Ekinox09, 2023-01-06:
> Without pairing to an aqara hub like you did the cube was not behaving properly.

Hello, I don't have any hub. I'm using the cube with an USB Conbee II controller on a debian VM. I do my test "directly" with Z2M in debug mode; once everything's OK, I use the config files it with my domotic software (Jeedom with zigbee2MQTT plugin). And, without any hub, the cube behave perfectly for me (now that I've undestood that action_mode need some  "rough" movements to be triggered). I noticed that after a pairing, I have some strange behaviour; switching from one mode to another (5 clics) makes everything right. Hope it helps.

jzee923, 2023-01-06:
I was referring to [this](https://github.com/Koenkk/zigbee2mqtt/issues/15652#issuecomment-1368505241) issue that a few of us were having.
I outlined the data I was getting [here](https://github.com/Koenkk/zigbee2mqtt/issues/15652#issuecomment-1370427316) before the cube was properly configured.
I'm not sure how you got past this but I had to add a configure section in my definitions in order to get the cube to configure and start sending the correct actions.
Part of trying to get this working was binding some of the clusters.
And again, I'm not sure which of these are actually required, but it was the first step required for me to get the cube working. 
So basically, I wasn't able to use @JJPro's code. I have everything working with my config (i think) except being able to change the operation mode in the z2m gui. It seems like it should work but it keeps timing out.

Ekinox09, 2023-01-06:
Hello all,
I would suggest not to update the "action" with "side_up" in the scene mode. Using the cube in the scene mode, my final trigger is the side change (dedicated message sent after all movements  done, very reliable). During the cube movements, "action" can switch to "hold" or "shake" BUT the content of this action is systematically replaced by "side_up" once the cube is back to calm. I would be interested to let this "action" set to "shake", "hold" in order to be able to have specific action depending on the final face (shake + final face 6 = 1 scenario, etc....). I have changed the code for my own usage but this might be helpful for other (will multiplicate the actions "hold" and "shake" usage by 6). Let me know your point of view regarding this. Bye.

jzee923, 2023-01-06:
This sounds like a perfect use case for the aqaraOpple cluster 329.
It reports back which side is up regardless of your operation mode. I was worried that this would cause conflicts with the existing "side" or "side_up" which is why I had originally commented it out.
Right now in my code it updates "side"
Following your example
I pick up the cube with side 6 on top, I shake it, I place it down with side 3 facing up.

So if you look at my logs, It receives the shake action and starts with side:6 and side_up:6
Then, the aqaraOpple cluster reports in with side 3 and the final report is side:3 and side_up: 6 since we never triggered a side_up action.

Ekinox09, 2023-01-06:
Excellent !
For my need, I just commented the line "action: 'side_up'" in the "aqaraOpple" cluster. That do the trick. 
"action" property is only updated by the "action_multistate" cluster with "side_up, hold or shake" (and "rotate" with the formal analog cluster) and  "aqaraOpple" cluster only update "side_up" property with the final side up.

jzee923, 2023-01-06:
With the inclusion of the face up side value in aqaraOpple on the T1 pro cube, it makes me think all the "math" we have in the multiStateInput is redundant...

OstfrieseUnterwegs, 2023-01-10:
Does anyone else get the error

`Failed to configure 'CubeT1', attempt 1 (Error: Write 0x54ef44100062cf1e/1 aqaraOpple({"mode":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received) at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:658:23) at runNextTicks (node:internal/process/task_queues:60:5) at processTimers (node:internal/timers:504:9) at Endpoint.sendRequest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:389:28) at Object.configure (/opt/zigbee2mqtt/data/extension/externally-loaded.js:150:9) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:117:13))`

using a conbee2 USB stick
The Cube is working OK, but an error always makes me a bit nervous.

filipeaparicio91, 2023-01-10:
@OstfrieseUnterwegs, I had the same issue and was able to get the Cube configured by toggling between modes (scene and action), which can be done by clicking the reset button 5 times. Doing this a few times allowed the Cube to configure correctly.

I am using a Sonoff router though, so I'm not sure if the process is the same with a Conbee2.

smartmatic, 2023-01-11:
> @OstfrieseUnterwegs, I had the same issue and was able to get the Cube configured by toggling between modes (scene and action), which can be done by clicking the reset button 5 times. Doing this a [...]

This is possible with version 1.29.1-1 or is dev version needed?

filipeaparicio91, 2023-01-11:
@smartmatic, it should work with any version of Z2M. You just need to create an external converter for the cube and add it to your Z2M configuration file.

Here is the complete setup with the converter that I'm using:

**1. Create a new JS file under _config_ > _zigbee2mqtt_ (ex: MFCZQ12LM.js) and paste the following code:**
```
/**
  # Two Modes

  ## Scene Mode
    - rotate
    - shake
    - hold
    - side up
    - trigger after one-min inactivity

  ## Action Mode
    - slide
    - rotate
    - tap twice
    - flip90, flip180
    - shake
    - trigger after one-min inactivity

  # Clusters (Scene Mode): 

  ## Endpoint 2: 

  | Cluster            | Data                      | Description                   |
  | ------------------ | ------------------------- | ----------------------------- |
  | aqaraopple         | {329: 0-5}                | i side facing up              |
  | genMultistateInput | {presentValue: 0}         | action: shake                 |
  | genMultistateInput | {presentValue: 4}         | action: hold                  |
  | genMultistateInput | {presentValue: 2}         | action: wakeup                |
  | genMultistateInput | {presentValue: 1024-1029} | action: fall with ith side up |

  ## Endpoint 3: 

  | Cluster   | Data                                  | Desc                                       |
  | --------- | ------------------------------------- | ------------------------------------------ |
  | genAnalog | {267: 500, 329: 3, presentValue: -51} | 267: NA, 329: side up, presentValue: angle |

 */

  const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
  const exposes = require('zigbee-herdsman-converters/lib/exposes');
  const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');
  const utils = require('zigbee-herdsman-converters/lib/utils');
  const e = exposes.presets;
  const ea = exposes.access;
  
  const OPS_MODE = {
    ACTION: 0,
    SCENE: 1,
  };
  const ops_mode_lookup = { 0: 'action mode', 1: 'scene mode' };
  
  const aqara_opple = {
    cluster: 'aqaraOpple',
    type: ['attributeReport', 'readResponse'],
    options: (definition) => [
      ...xiaomi.numericAttributes2Options(definition),
      exposes.enum('operation_mode', ea.STATE, ['scene mode', 'action mode']),
    ],
    convert: (model, msg, publish, options, meta) => {
      
      if (msg.data.hasOwnProperty('155') || msg.data.hasOwnProperty('328')) {
        const ops_mode = msg.data['155'] || msg.data['328'];
        meta.state.operationMode = ops_mode;
      };
[... 71 lines cut ...]
        'hold',
        'side_up',
        'rotate_left',
        'rotate_right',
      ]),
    ],
  };
  
  module.exports = definition;
```

**2. Edit the configuration.yaml file under _config_ > _zigbee2mqtt_ and add the following lines:**

```
external_converters:
  - MFCZQ12LM.js
```

**3. Reboot Z2M and pair the Cube.**

**4. Toggle between Action and Scene mode by clicking the Cube's reset button 5 times (repeat this 3-4 times to successfully configure the cube in Z2M)**

In the future when the Cube is included natively in Z2M, you can delete the external converter and remove the code from the configuration.yaml file.

Hope this helps!

OstfrieseUnterwegs, 2023-01-11:
> This is possible with version 1.29.1-1 or is dev version needed?

I think by now I switched mode probably 20 times, deleted the cube and paired again, switched mode many times again - yet, it does not want to configure.
However, I can use the cube in ioBroker and also see the mode changing etc. So probably it is just annoying but nothing more.

Strangely when using mqttexplorer and publish to zigbee2mqtt/bridge/request/device/configure
I get a response
`{
  "data": {
    "id": "0x54ef44100062cf1e"
  },
  "status": "ok"
}`

biosniper, 2023-01-15:
> Been working on getting a toZigbee converter working in order to change the operating mode in the z2m ui. [image] I was able to get it showing up as a toggle button but it doesn't actually function [...]

Just wanted to say this is working pretty great for me as of 15th Jan 2023, so thank you very much for your code!
Using a Sonoff ZBDongle-P with Z2M via HA.

This is a bit beyond my ability to contribute to other than feedback sadly!

smartmatic, 2023-01-15:
> @smartmatic, it should work with any version of Z2M. You just need to create an external converter for the cube and add it to your Z2M configuration file.
>

Thank you for that nice guideline!

ybizeul, 2023-01-15:
Anyone managed to make « tap » work ? I don’t see anything in the logs when tapping.

jzee923, 2023-01-15:
you have to be in "action" mode and I believe it's a double tap

oskarslusitis, 2023-01-16:
Is it possible to use additional things, like "roate left" with side 3 up? I integrated Cube with code from this thread through external convertors, but couldn't find such options in UI.
Might I add i'm quite new in HA.
Thanks!

filipeaparicio91, 2023-01-16:
@oskarslusitis, if your Cube was integrated correctly, you should see a **Side Up** sensor under the device in HA.
It reports whatever side is currently facing up when an action occurs. (From experience, having the cube in **scene mode**, reports the side up for reliably).
You can then use that as an automation condition.
In my case, I have an automation set up for each one of the Cube's sides, which are triggered when an mqtt payload is received from the cube.
I have 6 automations in total, each one has a condition for **Side up** of "1, 2, 3, 4, 5 or 6".
Then I have a **Choose** action for each payload value. Ex: if trigger payload is rotate_left, do X, if trigger payload is rotate_right, do Y, if trigger payload is shake, do Z.
If you'd like, I can share an example. In fact, I don't think it would be too complicated to whip out a Blueprint for this.

ybizeul, 2023-01-16:
> you have to be in "action" mode and I believe it's a double tap

I forgot to mention that’s already the mode I’m using.

JuMi2006, 2023-01-16:
I tried the cube today and was very surprised.
After pairing and playing with the cube it switches of every Silvercrest smart plug in my house.
Not real funny. ;)

TS011F
_TZ3000_kdi2o9m6
https://www.zigbee2mqtt.io/devices/HG06337.html#lidl-hg06337

So now i am not able to figure out more information because i am not alone at home. ;)

jzee923, 2023-01-16:
@JuMi2006 a few of us were having a similar issue because the device wasn't being configured properly.
You can give my converter code above a try, it works for me.

JJPro, 2023-01-17:
I'm thinking could it be that your other devices are bound to coordinator's genOnOff cluster. And then the converter you have also binds the cube's genOnOff to the coordinator's genOnOff. 
So they now act together. 

Check the bind tab on affected devices.

JJPro, 2023-01-17:
Then do the following might help: 
1. remove `genOnOff` binding code in your converter
2. add `fz.ignore_onoff_report` to `fromZigbee` definition part

wvanbruksvoort, 2023-01-17:
> @smartmatic, it should work with any version of Z2M. You just need to create an external converter for the cube and add it to your Z2M configuration file.
>
> Here is the complete setup with the [...]

You mention you need to toggle between Action and Scene Mode for 3-4 times. How do you know when the device is successfully configured? For example if i look at the operating mode of the Cube in Z2M is see it states N/A

sladiri, 2023-01-17:
Hi, thanks my cube Pro is working as well, using the Sonoff 3.0 Usb Plus dongle.

I tried the config by @filipeaparicio91 first (https://github.com/Koenkk/zigbee2mqtt/issues/15652#issuecomment-1378753370).
Rotate and tap were both very reliable and instant (I made a video too). For tap I just picked the cube up with my fingers about a cm or two and set it down immediately/abruptly.

Then I tried to change the config from @jzee923 (https://github.com/Koenkk/zigbee2mqtt/issues/15652#issuecomment-1373029151). I did not remove the cube from the network though, I just restarted the addon. Is this a problem? I had errors after this and I had to force-remove the cube.

After pairing it did work, but tap has been very unreliable since then. There i no error, just most of the time, there is no tap event. Sometimes it registers a double-tap.

I tried restarting the Z2M and resetting the cube, didn't help apparently. As I said, initially tap worked 100% of the time, I tried using the cube for the first time too.

I am very new to this and Home Assistant, but I'll try to help debugging, if needed.

jzee923, 2023-01-17:
@sladiri according to the documentation a "tap" is triggered by picking up the cube and tapping twice.
i do see the tap triggering sometimes when i only do a single tap, however with a double tap it works very consistently for me.
regarding any kind of config change, i think it's probably best to first remove the device and re-pair it after using a new config and restarting z2m.

wvanbruksvoort, 2023-01-17:
> @oskarslusitis, if your Cube was integrated correctly, you should see a **Side Up** sensor under the device in HA.
> It reports whatever side is currently facing up when an action occurs. (From [...]

If you can share an example or share it as a blueprint would be great 😊

smartmatic, 2023-01-17:
@jzee923 Thank you for sharing your code. Its working on my side. Where i am a bit struggling is to change the mode? And what is also not detected is to tap the device in action mode.
Do you have some tipps?

jzee923, 2023-01-17:
the only way i've been able to use the ui to change modes is by physically clicking the button in the cube once. I believe this sends some kind of ping or wake command and if i immediately try to change the mode in the ui after this it works. that kind of defeats the whole point of a UI button though because if i have to open up the cube to hit the button once, i might as well just hit it 5x.

as far as tap goes, i lift up the cube and tap it against the table twice. this works very consistently for me.

[70 more replies]

closed 2023-01-31 by Koenkk