Koenkk/zigbee2mqtt #15658: Aqara TVOC air quality monitor - VOCKQJK11LM : sometime no device dataset

kris969, 2022-12-20:
### What happened?

Among devices I'm using, some of them are not reporting full zigbee dataset.
I notice, for example for the Aqara TVOC air quality monitor - `VOCKQJK11LM`, but not only that usually a complete dataset is received, but sometime the device part isn't present, it seems data available are limited to the "export ones"

My code is looking for device manufacturer name, so in this case I have an issue with his. 
 
This witness is done thru z2m state display.
complete dataset is present in log!

Have you an idea : where could cause this issue.

### What did you expect to happen?

simply have a complete dataset for each zigbee message report.

### Zigbee2MQTT version

1.28.4-dev commit: 8fc83ca

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

Zigbee2MQTT:info  2022-12-20 08:35:14: MQTT publish: topic 'zigbee2mqtt/SALON_VOC_01', payload '{"battery":100,"device":{"applicationVersion":29,"friendlyName":"SALON_VOC_01","ieeeAddr":"0x54ef4410002a5b7c","manufacturerID":4447,"manufacturerName":"LUMI","model":"VOCKQJK11LM","networkAddress":51992,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":19,"display_unit":"ppb_celsius","humidity":58.95,"last_seen":"2022-12-20T07:53:46.609Z","linkquality":255,"power_outage_count":13740,"temperature":19.53,"voc":162,"voltage":3089}'
Zigbee2MQTT:info  2022-12-20 08:48:43: MQTT publish: topic 'zigbee2mqtt/SALON_VOC_01', payload '{"battery":100,"device":{"applicationVersion":29,"friendlyName":"SALON_VOC_01","ieeeAddr":"0x54ef4410002a5b7c","manufacturerID":4447,"manufacturerName":"LUMI","model":"VOCKQJK11LM","networkAddress":51992,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":19,"display_unit":"ppb_celsius","humidity":58.4,"last_seen":"2022-12-20T08:48:43.756Z","linkquality":255,"power_outage_count":13740,"temperature":19.54,"voc":162,"voltage":3089}'

mariusmotea, 2022-12-23:
Not sure if is the same issue, but i notice on my Aqara TVOC that on last version in Bind and Reporting page the details are missing after pair. I downgrade zigbee2mqtt to version 1.28.2 and after pair the details where all back there.

kris969, 2023-01-23:
This issue isn't present anymore since the last two z2m versions.
it can be closed.

closed 2023-01-23 by Koenkk