Koenkk/zigbee2mqtt #15659: [New device support]: AVATTO Tuya ZigBee Radiator Actuator Thermostat

marcelhoogantink, 2022-12-20:
### Link

https://www.aliexpress.com/item/1005003952150617.html

### Database entry

{"id":9,"type":"EndDevice","ieeeAddr":"0xa4c1387ad36d6432","nwkAddr":2096,"manufId":4417,"manufName":"_TZE200_6rdj8dzm","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1671534450516,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to create this new device, but it is too complex for me, I got stuck on the converter, sorry.
The device looks like: https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html

marcelhoogantink, 2022-12-20:
Is already supported, but not yet available in Home Assistant

closed 2022-12-20 by marcelhoogantink

vingerha, 2022-12-21:
I just received my _TZE200_6rdj8dzm and z2m complains about the support. I re-paired a number of times without luck.
Z2M response:
```
22-12-21 11:35:29Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_6rdj8dzm'
Warning 2022-12-21 11:35:29Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2022-12-21 11:35:29MQTT publish: topic 'zigbee2mqtt/0xa4c13848c001094c', payload '{"last_seen":"2022-12-21T11:35:29+01:00","linkquality":255}'
Info 2022-12-21 11:35:30MQTT publish: topic 'zigbee2mqtt/0xa4c13848c001094c', payload '{"last_seen":"2022-12-21T11:35:30+01:00","linkquality":255}'
Warning 2022-12-21 11:35:31Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_6rdj8dzm'
Warning 2022-12-21 11:35:32Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2022-12-21 11:35:32MQTT publish: topic 'zigbee2mqtt/0xa4c13848c001094c', payload '{"last_seen":"2022-12-21T11:35:31+01:00","linkquality":255}'

marcelhoogantink, 2022-12-21:
It is not available in the current release, It will be at the next release, or in the dev-release now.
There is a converter available: https://github.com/twhittock/avatto_me167

vingerha, 2022-12-21:
Thanks for the swift response but.... 
1. I put me167.js in my data folder
2. extended configyaml as below with external_conv
```
  '0xa4c13848c001094c':
    friendly_name: '0xa4c13848c001094c'
availability: true
advanced:
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  last_seen: ISO_8601_local
  log_level: info
external_converters:
  - me167.js
```
3. removed the old pairing (force remove)
4. re-paired and still the same error. 
How can I verify if the js is (properly) loaded?
Also, I am happy to modify/test the js but other than reading code, not much of a programmer, let alone in this specific area

vingerha, 2022-12-21:
Forget it, I found the the manufacturer in the js, , replaced it by '_TZE200_6rdj8dzm'  and now it works....so maybe this needs to be added to the js? No clue how to do this myself, i.e. can one add multiple manufacturers  / devices to one js?