Koenkk/zigbee2mqtt #15660: [New device support]: Schneider Wiser Smoke Alarm CCT599001

mandrean, 2022-12-20:
### Link

https://www.se.com/ww/en/product/CCT599001/smoke-alarm-wiser-battery-10-years-lifetime-white-ip20/

### Database entry

{}

### Comments

Seems like a solid zigbee smoke alarm!

Would add support for it myself if I had a Schneider gateway, but I don't.

[image]

mandrean, 2022-12-20:
After some Googling I found this GH issue with some additional info:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6617

dennisgranasen, 2023-01-12:
No need to sniff any gateway for the basic functionality anyway. I made a PR for this just by implementing standard fZ clusters. It seems to work just fine, but more testing is of course welcome. https://github.com/Koenkk/zigbee-herdsman-converters/pull/5292

closed 2023-02-13 by mandrean

t112013, 2023-04-03:
Hello. is there any works on getting the function for triggering the alarm working ? 
That is very crutial to me if i would decide to buy these units

t112013, 2023-10-28:
Could this please be open and see if any nice sould out there can work on triggering the alarm ?