Koenkk/zigbee2mqtt #15666: Not possible to set a schedule _TZE200_e9ba97vf

Truddel, 2022-12-20:
### What happened?

Hi all,

when trying to set a schedule over the web it does not work, button is just greyed out:
[image]

Trying to send it over MQTT: 
[image]
Leads to this in the log: 
[image]

### What did you expect to happen?

Set schedule properly.

### How to reproduce it (minimal and precise)

try to set schedule via UI or MQTT.

### Zigbee2MQTT version

1.28.4-dev commit: 3796084

### Adapter firmware version

20211217

### Adapter

Sonoff Stick

vladi1234, 2022-12-21:
Hi,
As far as I know, this has never worked in the new Tuya format.

My Schedule variant in old Tuya format is much simpler and works very well. https://github.com/Koenkk/zigbee2mqtt/issues/15164#issuecomment-1326817667

trikidiki, 2023-01-02:
I was able to alter the schedule until a couple of weeks ago. Now I cannot change it via the Zigbee2MQTT interface or directly publishing from MQTT Explorer (which looks to be the same as the original poster is using.

I was struggling with switching between auto and manual which appeared to be resolved in:

Heating is completely broken right now (At least for TV02 thermostats) #15164

then with the updates that seem to have resolved that issue came this new problem of inability to change the schedule. The schedule I had set prior to this was a rough setup to get things working but needs tuning. Now my only option is to set to manual and alter the target temperature.

This has made the automatic valves redundant, it is quicker to go around the house and physically alter them.

closed 2023-02-09 by github-actions[bot]