Koenkk/zigbee2mqtt #15668: [New device support]: Afintek TRV TS0601 _TZE200_7fqkphoq

kiekerjan18, 2022-12-20:
### Link

https://afinjo.nl/afintek/614-afintek-smart-radiatorknop-knop-voor-verwarming-automatische-radiator-bediening-met-app-energiebesparend-8719925050558.html?gclid=Cj0KCQiA14WdBhD8ARIsANao07gi9PFvB94zoQunhLm_DWkIcfXQnQ5T02EGN6SEF98p7PRQmNI7KG4aAvZcEALw_wcB

### Database entry

{"definition":null,"friendly_name":"0x003c84fffec7f2ca","ieee_address":"0x003c84fffec7f2ca","status":"successful","supported":false}

### Comments

Hi, today I've paired a brand new TRV today of brand: Afintek. 

It looks identical as (_TZE200_ywdxldoj):
https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#tuya-ts0601_thermostat

I've read that it is quite easy to add it to a list of already supported TRV's of type: TS0601. 
https://github.com/Koenkk/zigbee2mqtt/issues/6873

Is it possible to add this device?

`warn  2022-12-20 16:44:19: Device '0x003c84fffec7f2ca' with Zigbee model 'TS0601' and manufacturer name '_TZE200_7fqkphoq' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-12-20 16:44:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x003c84fffec7f2ca","ieee_address":"0x003c84fffec7f2ca","status":"successful","supported":false},"type":"device_interview"}'
debug 2022-12-20 16:44:19: Received Zigbee message from 'switch hal', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 2022-12-20 16:44:19: Received Zigbee message from 'switch hal', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 2022-12-20 16:44:19: Received Zigbee message from 'switch hal', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 2022-12-20 16:44:19: Received Zigbee message from 'switch hal', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 2022-12-20 16:44:20: Received Zigbee message from '0x003c84fffec7f2ca', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
warn  2022-12-20 16:44:20: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_7fqkphoq'
warn  2022-12-20 16:44:20: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html`

Koenkk, 2022-12-20:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-20 by Koenkk

kiekerjan18, 2022-12-20:
Works like a charm! Thanks!