Koenkk/zigbee2mqtt #15669: Wall switch Moes TS0012 does not have backlight mode and power on behavior available under Configuration

relombardo, 2022-12-20:
### What happened?

Wall switch Moes TS0012 does not have backlight mode and power on behavior available under Configuration section or Exposes. These fields are available for same model with 3 gang (TS0013) and 1 gang (TS0011).

Note: The model label on each device vs the model they are identified in Z2M is as follow:

ZTS-US (1-gang) = found as ZS-EUB_1gang
ZTS-US (2-gang) = found as TS0012
ZTS-US (3-gang) = found as TS0013

In summary, ZS-EUB_1gang and TS0013 are ok, while TS0012 is not.

### What did you expect to happen?

TS0012 should have same options as TS0013, at least, and allow configuration of Backlight mode and Power on behavior.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

Koenkk, 2022-12-20:
Can you provide the data/database.db entry of this device?

osnwt, 2022-12-21:
FYI: it seems the firmwares of TS0012 and TS0013 are quite different. For instance, if you press few keys at once, TS0013 sends all, but TS0012 sends just the first one. You have to wait around 0.25-0.5s between clicks to send another one. That's why I will replace my 2-gang switches by 3-gang ones. So maybe it just does not support options you've mentioned.

But my TS0013 do not control backlight, see https://github.com/Koenkk/zigbee2mqtt/issues/15669. @Koenkk could you please have a look at it?

osnwt, 2022-12-21:
Also, if you're there, could you please explain what the **power_on_behavior_left** state of TS0013 is? It is not exposed in settings, and also there is no others for center/right keys. Is it a bug or a feature?

```
{
    "backlight_mode": "MEDIUM",
    "last_seen": "2022-12-21T13:27:23+03:00",
    "linkquality": 108,
    "power_on_behavior": "off",
    "power_on_behavior_left": "off",
    "state_center": "OFF",
    "state_left": "ON",
    "state_right": "OFF"
}
```

Koenkk, 2022-12-21:
Please provide the exact model number of the `TS0013` where the backlight mode doesn't work (preferably the `data/database.db` entry)

osnwt, 2022-12-21:
Also sorry, I mentioned wrong issue number. The correct one is: https://github.com/Koenkk/zigbee2mqtt/issues/14623

relombardo, 2022-12-21:
> Can you provide the data/database.db entry of this device?

{"id":10,"type":"EndDevice","ieeeAddr":"0xa4c1386bd2e93974","nwkAddr":10211,"manufId":4417,"manufName":"_TZ3000_18ejxno0","powerSource":"Mains (single phase)","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��5+\u0011","65506":51,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_18ejxno0","zclVersion":3,"appVersion":69,"modelId":"TS0012","powerSource":3}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07e12c","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07e12c","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1794938824},"lastSeen":1671639187936,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-12-21:
@osnwt so either the device does not support changing the backlight mode or it uses a different api. To figure this out you need to check out wether this is supported via the TuYa gateway and if it is sniff the traffic so I can reverse engineer how to set it. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

@relombardo 
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/e4c1baa313af39ccd3ac8e5cf5d48675

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

relombardo, 2022-12-21:
> @osnwt so either the device does not support changing the backlight mode or it uses a different api. To figure this out you need to check out wether this is supported via the TuYa gateway and if it [...]

Hi @Koenkk, it did not work to me. I followed your steps and tried to restart Z2M, HA and even delete and re-interview the device.

osnwt, 2022-12-21:
@Koenkk, thanks for suggestion. Since I don't have an original Tuya-compatible hub, I just ordered one, but it's a long time story. So I'll be back as soon as I have everything for traffic generation and capture.

osnwt, 2022-12-21:
BTW, checked my TS0012 via dev console by sending 0, 1 or 2 for backlight. it works as promised in the manual: 0 - always off, 1 - on for turned on switch, 2 - on for turned off switch (inverted). Power on behavior returned unsupported attribute error.

My TS0013 allowed r/w to both backlight and startup state attributes, but changing backlight mode has no effect on the device. Perhaps, it uses another DP or does not support it (I'll keep investigation for later when I have native Tuya hub).

But your converter code gave me the following error:

```
[01:58:04] INFO: Preparing to start...
[01:58:05] INFO: Socat not enabled
[01:58:06] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:20
    extend: tuya.extend.switch({backlightMode: true}),
                        ^
TypeError: Cannot read properties of undefined (reading 'switch')
    at /app/dist/util/externally-loaded.js:20:25
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```
Using Home Assistant with Z2M version 1.28.0-1 (I had issues with newer, do not update for now).

Koenkk, 2022-12-22:
@osnwt you need to be on z2m 1.28.4 for this to work

@relombardo does the backlight now show up under the exposes of the device page? (in the z2m frontend)

relombardo, 2022-12-22:
@Koenkk , it does not show up under exposes of z2m frontend.
I am adding two attachments, one for you to check if my set up is correct and the other one for you to check how I see the exposes tab.

[image]
[image]

osnwt, 2022-12-23:
Since I also have a couple of TS0012, I stepped in and tried to debug it. At first, I finally upgraded muy Z2M to the latest version. I lost some thermostat states, but they recovered after Home Assistant restart.

Next, I modified your converter to extend my TS0012 by replacing this (for my manufacturerName):

```
const definition = {
//    zigbeeModel: ['TS0012'],
     fingerprint: [
        {
            modelID: 'TS0012',
            manufacturerName: '_TZ3000_dlp6yvs8',
        },
    ],
```

After restart I have the following results:

State (note the **backlight_mode** added):
```
{
    "backlight_mode": "medium",
    "last_seen": "2022-12-23T04:00:00+03:00",
    "linkquality": 196,
    "power_on_behavior_left": "off",
    "state_left": "OFF",
    "state_right": "OFF"
}
```

Using dev console, I **was able to change** the backlight mode (it works). The model shown is now **TS0012_new** (added to be sure the converter was loaded).

But in Exposes tab I see no changes just like in the report above. Perhaps, something still is missing.

osnwt, 2022-12-23:
And a notice: for this switch the meaning of backlight mode option is different: off, normal, inverted.

Koenkk, 2022-12-23:
Found the issue, updated https://gist.github.com/Koenkk/e4c1baa313af39ccd3ac8e5cf5d48675

osnwt, 2022-12-23:
Perfect! It works!

Can you suggest if it's easy to remap description of enum for backlight behavior according to the one I mentioned earlier?

Koenkk, 2022-12-23:
Done, also fixed the enum values. Happy xmas! 🎅 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

osnwt, 2022-12-23:
Thank you for quick response and excellent support!

relombardo, 2022-12-24:
> Found the issue, updated https://gist.github.com/Koenkk/e4c1baa313af39ccd3ac8e5cf5d48675

@Koenkk , but I still do not see additional exposes and my On/Off toggles stopped working as per image below. It happened after I re-paired my device.
[image]

osnwt, 2022-12-24:
Could you please also have a look at https://github.com/Koenkk/zigbee2mqtt/issues/14965?

Faced the same problem, have few of such devices. They can be set physically with 0.5 step, and Moes says it can also be set with 0.5 step using their app (have no Tuya hub yet to verify).

So probably it's the issue with the interface layer.

osnwt, 2022-12-24:
> > Found the issue, updated https://gist.github.com/Koenkk/e4c1baa313af39ccd3ac8e5cf5d48675
> 
> @Koenkk , I still do not see additional exposes and my On/Off toggles stopped working as per image below. It happened after I re-paired my device.
> 
> [image]

Try my version: replace the manufacturerName with your own value from About tab (Zigbee Manufacturer):

```
const definition = {
//    zigbeeModel: ['TS0012'],
     fingerprint: [
        {
            modelID: 'TS0012',
            manufacturerName: '_TZ3000_dlp6yvs8',
        },
    ],
```

relombardo, 2022-12-24:
> Try my version: replace the manufacturerName with your own value from About tab (Zigbee Manufacturer):
>
> ```
> const definition = {
> //    zigbeeModel: ['TS0012'],
>      fingerprint: [ [...]

@osnwt , I added your code, changed manufacturer name accordingly, and kept all other lines from Koenkk's converter. Is that the plan?
Then I restarted my Z2M and noticed my On/Off function is working again! I don't see backlight, though. Are you using dev console to control it? If so, could you explain details on how to do it?

osnwt, 2022-12-24:
Yes, I took the latest converter, replaced zigbeeModel by fingerprint (not sure if it's requried for TS0012 in general), and after restart Z2M it worked for me in device settings (the light near switches is just a switch as X helper for this switch):

[image]

I might also click reconfigure device first, but not sure if I did that this time:

[image]

As for the dev console, IIRC it worked for me even without converter, I just added the following attribute and was able to read/write it:

[image]

relombardo, 2022-12-26:
I noticed my entry of external converters in configuration file was wrong and pointing this topic as a content of my previous lines. (Indentation issue)

@Koenkk , your latest code worked perfectly. Thank you very much. Is this feature work without converter in a future update? I am wondering if I can disable the converter once it happens and how can I be notified.

@osnwt , thank you so much for the details of your troubleshooting. Your code also worked over here after I fixed the indentation issue on my configuration file.

Koenkk, 2022-12-27:
@relombardo it will be included in the 1 January release.

ajunio3, 2023-01-20:
> Yes, I took the latest converter, replaced zigbeeModel by fingerprint (not sure if it's requried for TS0012 in general), and after restart Z2M it worked for me in device settings (the light near [...]

How can I do this to display in the exposed entities? I have a similar issue with my Avatto 3gang switch and in console I can switch on/off the backlight using the tuyaBacklightSwitch attribute

ajunio3, 2023-01-20:
here my topic with details

https://github.com/Koenkk/zigbee2mqtt/issues/15970

relombardo, 2023-01-31:
Thanks @Koenkk and @ajunio3. I removed the converter and repaired my device. Now it has the backlight function available and working. Koenkk thanks for working on the new release to fix it!

closed 2023-01-31 by relombardo