Koenkk/zigbee2mqtt #15677: After update to 1.28.4-1 smart plug recognise as Waterproof EU socket

Safs, 2022-12-21:
### What happened?

After update to 1.28.4-1 smart plug recognise as Waterproof EU socket:

**Before:** 
[image]

**After:**
[image]

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

 20201026

### Adapter

Silicon_Labs_slae.sh_cc2652rb_stick

zegl, 2022-12-21:
Hey! Could you please share the entry of the device from the `database.db` file?

Safs, 2022-12-21:
Before update:

{"id":25,"type":"Router","ieeeAddr":"0x0c4314fffe9e8a6e","nwkAddr":2761,"manufId":4098,"manufName":"_TZ3000_ew3ldmgx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_ew3ldmgx","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":238,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsCurrent":60,"activePower":4}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,36287]}},"genOnOff":{"attributes":{"onOff":1,"moesStartUpOnOff":2,"tuyaBacklightMode":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc47ca","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc47ca","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc47ca","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1798312590},"lastSeen":1671621843530,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-12-21:
There is no such device supported, are you using an external converter? (you can find it in `configuration.yaml`)

Safs, 2022-12-21:
No, device was detected and recognised automaticly. It is this plug: https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1
Why you say, not supported?

Koenkk, 2022-12-21:
Can you provide your zigbee2mqtt `configuration.yaml`?

Safs, 2022-12-22:
Sorry, this is my fault, I using  external converter for another device, but I give for it same device name as for plug. I changed to different name and everything working ok!
Thanks!

closed 2022-12-22 by Safs