Koenkk/zigbee2mqtt #15678: no_occupancy_timeout unavailable in UI

Jeoffreybauvin, 2022-12-21:
### What happened?

I don't understand how "no_occupancy_since" is supposed to be working in the UI : 

I have a RTCGQ11LM and I need to set no_occupancy_since. In the UI, I'm clicking on the "+" : 

[image]

And I have only this : 

[image]

5 secondes after, It's rolling back to screen 1. 

Am I missing something ?

After setting the right configuration in yaml, I have this : 

```
  '0x00158d000482eca2':
    friendly_name: xiaomi_motionsensor_etage_chblanche
    occupancy_timeout: 120
    no_occupancy_since: [ 10, 90, 600 ]
```

[image]

Still missing something, right ?

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20200805

### Adapter

Texas Instruments LAUNCHXL-CC1352P-2

closed 2023-01-28 by github-actions[bot]

szczepuz999, 2023-02-26:
Hi @Koenkk . I also noticed something off with those HUE montion sensors. no_occupancy_since was working for half year without issue, today I wanted to increase this time value but when I have entered specific setting page in z2m in the fields were no numbers. And when I have inputed new values, in the logs we can see such a entry

`2023-02-26 13:10:11MQTT publish: topic 'zigbee2mqtt/pirPoddasze', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"led_indication":true,"linkquality":255,"motion_sensitivity":"high","no_occupancy_since":{"undefined":30},"occupancy":false,"occupancy_timeout":0,"temperature":21.31,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'`

and this entry is coming with message when occupance became false. before it was coming only after no_occupancy time was reached (so for example 30 seconds after occupancy became false).

And then I have decided to pair new Hue montion sensor and I have exactly same experience as @Jeoffreybauvin, simply UI is self reverting to initial state and does not allow enter any valued. I have tried to close browser and open again few times and it finally worked out but still this no_occupancy_since is seen in logs as undefined... 
could you please have a look? it blows whole handling of the presence information in my automations and I guess many others.
Thanks in advance!

szczepuz999, 2023-02-26:
[image]

problem is how z2m puts data into config yml file. it puts the value with key UNDEFINED, I have maually changed to list with desired values and all works again! 🗡️