Koenkk/zigbee2mqtt #15684: [Feature request]: Add support for Local (Child) lock for ClickSmart+ Socket

jampez77, 2022-12-21:
### Is your feature request related to a problem? Please describe

I have a few of these ClickSmart+ 2 gang sockets and the locking feature isn't exposed to Z2M.
https://zigbee.blakadder.com/ClickSmart_CMA30036.html

### Describe the solution you'd like

I've seen on this Tasmota discussion that they've had some success with (de)activating the lock feature. I was hoping that this meant it would be a doable job in Z2M.

https://github.com/arendst/Tasmota/discussions/14727

### Describe alternatives you've considered

I have sniffed the traffic and gotten the same results in the Tasmota discussion. I also tried to create an external converter myself. But I have little to no idea what I'm doing and all my attempts have resulted in "No converter found" errors.

### Additional context

I think i've provided enough context but happy to provide more where i can if required.

Koenkk, 2022-12-21:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_mvn6jl7x'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_raviyuvk'}, {modelID: 'TS011F', manufacturerName: '_TYZB01_hlla45kx'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_92qd4sqa'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_zwaadvus'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_k6fvknrr'}],
    model: 'TS011F_2_gang_wall',
    vendor: 'TuYa',
    description: '2 gang wall outlet',
    extend: tuya.extend.switch({backlightMode: true, childLock: true, endpoints: ['l1', 'l2']}),
    whiteLabel: [{vendor: 'ClickSmart+', model: 'CMA30036'}],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: tuya.configureMagicPacket,
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

jampez77, 2022-12-21:
I can confirm that this works perfectly for me! Thank you so much!

Koenkk, 2022-12-21:
Great, integrated it, you can keep using the external converter until the next release.

closed 2022-12-21 by Koenkk

chukaonline, 2023-01-16:
I have a similar socket using the same converter but this socket supports power monitoring which is the reason i bought it. Can you update this converter to include monitoring?

https://a.aliexpress.com/_mPvTsoy

Database entry
{"id":31,"type":"Router","ieeeAddr":"0x84ba20fffee23572","nwkAddr":33690,"manufId":4098,"manufName":"_TZ3000_raviyuvk","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"M�0*\u0011��"\u0012ـ"\u0012\u001c�"\u0012�"\u0012u&�*\u0011\u0000\u0000\u0000\u0000\f��"*\u0012","65506":31,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_raviyuvk","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":68,"powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":230,"rmsCurrent":22,"activePower":2}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,14185]}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a4ca6b","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a4ca6b","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a4ca6b","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[4,5,6,1794,2820],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":230,"rmsCurrent":22,"activePower":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-333483973},"lastSeen":1669971830143,"defaultSendRequestWhen":"immediate"}