Koenkk/zigbee2mqtt #15699: [Feature request]: Support Candle and Fireplace effects on Hue bulbs

deviantintegral, 2022-12-22:
### Is your feature request related to a problem? Please describe

Philips hue bulbs that support colour temperature now have two "dynamic scenes": Candle and Fireplace.

After sniffing traffic to a Hue bridge, I discovered it sends a command to the `0xfc03` cluster, which is already defined as `manuSpecificPhilips2` in zigbee2mqtt. It looks like that was added to support gradient light strips at https://github.com/Koenkk/zigbee2mqtt/issues/10674.

I was able to use the custom converter against my Light Strip Plus to send the following payloads. I sniffed Candle and Fireplace, but it doesn't appear the app exposes the colour loop scene yet.

| Sniffed | MQTT value | Scene |
|--------|-------------|---------|
|0x200000 | "200000" | Stop Scene |
|0x21000101 | "21000101" | Candle |
|0x21000102 | "21000102" | Fireplace |
|0x21000103 | "21000103" | Colour loop |

I tried a few more above 0x21000103 and there was no effect, so I don't think there's additional hidden scenes.

Here's the converter I'm testing with, mostly copied from https://github.com/Koenkk/zigbee2mqtt/issues/10674#issuecomment-1159460216.

```typescript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const Buffer = require('buffer').Buffer;

const tzLocal = {
    multicolor: {
        key: ['multicolor'],
        convertSet: async (entity, key, value, meta) => {
            const payload = {data: Buffer.from(value, 'hex')};
            await entity.command('manuSpecificPhilips2', 'multiColor', payload);
        },
    },
}

const definition = {
    zigbeeModel: ['LCL001'], // <-- update this with the modelID from your data/database.db prob something similar to LCF003, 4080248P7 or 915005987201
    model: 'dynamicscenetest',
    vendor: 'Philips',
    description: 'Testing light strip',
    meta: {turnsOffAtBrightness1: true},
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    toZigbee: [tzLocal.multicolor, ...extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).toZigbee],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

I can then trigger an effect by publishing `{"multicolor": "21000100"}` to the set topic of a light.

### Describe the solution you'd like

1. I'm not sure it makes sense to continue to use the word "multicolor". These scenes work on bulbs that only support colour temperature and full RGB.
2. From an end user perspective, I think it would make the most sense if these were exposed as effects alongside "breathe" and "OK", but that may not make sense given gradient lights have custom parameters? I don't have any gradient lights to see how z2m is currently working that way.
3. I had thought I could test this out using the Dev Console tab, but I couldn't figure out how to get a properly formatted payload. The custom converter solves this with `const payload = {data: Buffer.from(value, 'hex')};`. Given Philips is likely to add additional dynamic scenes, it would be good to either somehow support this from the web UI (or an MQTT message) without a custom converter for future testing.

### Describe alternatives you've considered

None!

### Additional context

* https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6192
* https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5891#issuecomment-1073038477 (note the payloads in that comment are typo'ed)

zegl, 2022-12-22:
This is really cool, great write up!

I think that adding an `effect` enum similar to how [`gradient_scene`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/063807ec7c64af767bd95482057b0ad55ddd3361/devices/philips.js#L207-L215) is done could make sense here. It makes it really easy for users to switch between effects (simple API and UI).

Do you know if this endpoint is supported by all Hue devices with color temperature support, or is it only for gradient devices?

deviantintegral, 2022-12-22:
I don't actually have any gradient devices at all. I've seen it work with the indoor light strip, full colour A19 bulbs, and even a bulb that doesn't support colour temperature (warm only) at all for the Candle effect.

sjorge, 2022-12-23:
@Koenkk would just providing an hue specific 'effect' tz work here, we can just have that one call the original effect tz converter unless it's one of the new ones and have it handle it? Not sure how to cleanly implement this, but this indeed seems to work on coler_temp only Hue bulbs on new firmware (I have a few BT enabled ones and it seems to work on those)

Koenkk, 2022-12-23:
It would be nice if we can integrate this in effect indeed, meaning:

- create a new hue effect converter
- if the effect name is one of the hue effects send it the hue way
- if not call the original effect converter

sjorge, 2022-12-23:
I’ll have a go at this tomorrow once if i finish early with my shopping.

sjorge, 2022-12-23:
Looks like the stop_effect doesn't really do anything aside from setting it to the default colorTemperature. 🤔

deviantintegral, 2022-12-24:
Nice to see the PR. I also confirmed I see (nearly) the same attribute reports for the effects mentioned at the end of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5891#issuecomment-1073038477. I started to see if I could extend the custom converter, but it looks like `Cluster.manuSpecificPhilips2` needs to be extended with the attribute.

Koenkk, 2022-12-29:
Implemented by @sjorge in https://github.com/Koenkk/zigbee-herdsman-converters/pull/5192

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-29 by Koenkk

Kars-de-Jong, 2023-01-03:
@Koenkk I just updated to version 1.29.0 (through the Home Assistant add-on), but on my Hue white and color ambiance LightStrip plus (LCL001) I don't see the new effects in the "effect" drop down of the UI.

Koenkk, 2023-01-03:
@Kars-de-Jong enabled it for the LCL001.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

marblecoma, 2023-01-03:
@Koenkk I updated to version 1.29.0, but according to the supported devices page on zigbee2mqtt.io there is no support for the new Effects for the Hue Gradient Lightstrip (929002994901). The lightstrip does support the effects through the hue app.

marblecoma, 2023-01-05:
> This is working for me, but not if I set the effect on a group. I imagine this would be better as it would keep multiple bulbs in sync. I'm a little surprised as I expected after #13438 this would "just work". Can someone else confirm, and if so I can file a new issue.

I tested triggering both Hue Effects (Candle and Fireplace) on a zigbee group, and I can confirm that neither of the effects triggered. The effects trigger on both lights if I trigger them on each light separately.

OnlyTheH, 2023-01-29:
> @Koenkk I just updated to version 1.29.0 (through the Home Assistant add-on), but on my Hue white and color ambiance LightStrip plus (LCL001) I don't see the new effects in the "effect" drop down of the UI.

Looks like they are also missing for the Hue Go with Bluetooth (7602031P7). Currently, they can only be selected by pressing the integrated button.

Koenkk, 2023-01-30:
@Simonh98 enabled it for the 7602031P7

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

lux73, 2023-01-30:
just to be sure to understood correctly:

**all Hue white ambiance should support Candle?**

if so please add support for these:

- 8718696548738
- 8718696695203
- 8718696598283

just checked, none of them have Candle Entry under "effect" - i'm using v1.29.2

Thank you

Simonh98, 2023-02-01:
> @Simonh98 enabled it for the 7602031P7
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@Koenkk Tested and works without problems.

Mihonarium, 2024-10-26:
Hue added more effects and color customization for effects in a recent update. Is it possible to add them to Z2M?

Mihonarium, 2024-10-26:
Hue added more effects and color customization for effects in a recent update. Is it possible to add them to Z2M?

(cc @Koenkk)