Koenkk/zigbee2mqtt #15700: Eurotronic SPZB0001 local_temperature_calibration INVALID_VALUE

nicoflink, 2022-12-22:
### What happened?

Hi there,

I have a climate Eurotronic SPZB0001 and I try to calibrate the local temperature by setting the offset. Using the GUI for the HomeAssistant Integration I always get the error message “INVALID_VALUE”. So I tried to publish the message manually but the error stays the same. Can someone help me to fix this issue. Here are the logs I see within Zigbee2Mqtt:

Received MQTT message on ‘zigbee2mqtt/Büro Heizung/set’ with data ‘{ “local_temperature_calibration”: “2” }’

Debug 2022-12-11 12:17:18Publishing ‘set’ ‘local_temperature_calibration’ to ‘Büro Heizung’
Error 2022-12-11 12:17:20Publish ‘set’ ‘local_temperature_calibration’ to ‘Büro Heizung’ failed: ‘Error: Write 0x00158d0003697ba0/1 hvacThermostat({“localTemperatureCalibration”:20}, {“sendWhen”:“immediate”,“timeout”:10000,“disableResponse”:false,“disableRecovery”:false,“disableDefaultResponse”:true,“direction”:0,“srcEndpoint”:null,“reservedBits”:0,“manufacturerCode”:null,“transactionSequenceNumber”:null,“writeUndiv”:false}) failed (Status ‘INVALID_VALUE’)’

Debug 2022-12-11 12:17:20Error: Write 0x00158d0003697ba0/1 hvacThermostat({“localTemperatureCalibration”:20}, {“sendWhen”:“immediate”,“timeout”:10000,“disableResponse”:false,“disableRecovery”:false,“disableDefaultResponse”:true,“direction”:0,“srcEndpoint”:null,“reservedBits”:0,“manufacturerCode”:null,“transactionSequenceNumber”:null,“writeUndiv”:false}) failed (Status ‘INVALID_VALUE’) at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1312:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)

### What did you expect to happen?

Set calibration of local temperature to 2

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

*

### Adapter

Conbee2

lux73, 2022-12-22:
try to send a decimal Value like:

`2.0`

nicoflink, 2022-12-28:
Hi, 

thank for responding. Unfortunately, the error stays the same.

lux73, 2022-12-29:
that's strange since few weeks before i've had two of them and Offset works - but now they're replaced with Danfoss TRVs

closed 2023-02-06 by github-actions[bot]