Koenkk/zigbee2mqtt #15706: Tuya TS011F (plug 1)has update available but won´t update

henkiejan1, 2022-12-22:
### What happened?

Hi everyone,

Todat arrives 2 additional Tuya TS011F smart plugs (plug 1). Pairing without any problem. But when i checked for a new firmware version zigbee2mqtt say: ´No update availble´. But strange enough when i look under status it says:

`"update": {
        "installed_version": 77,
        "latest_version": 192,
        "state": "idle"
`

So i think the actual version is 192 and should be possible to install?

### What did you expect to happen?

Updating the firmware to the latest available one.

### How to reproduce it (minimal and precise)

1. Add a TS011F (plug 1) to zigbee2mqtt with version below 192
2. Check for updates
3. Says: ´no update available´

### Zigbee2MQTT version

1.28.4-dev

### Adapter firmware version

20221102

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (P-version)

### Debug log

`debug 2022-12-22 20:10:26Is new image available for '0xa4c13878d7336b52', current '{"fieldControl":0,"manufacturerCode":4417,"imageType":54179,"fileVersion":77}', latest meta '{"fileVersion":192,"fileSize":307682,"url":"https://images.tuyaeu.com/smart/firmware/upgrade/20220907/1662545193-oem_zg_tl8258_plug_OTA_3.0.0.bin","sha512":"01939ca4fc790432d2c233e19b2440c1e0248d2ce85c9299e0b88928cb2341de675350ac7b78187a25f06a2768f93db0a17c4ba950b60c82c072e0c0833cfcfb"}'
info 2022-12-22 20:10:26MQTT publish: topic 'zigbee2mqtt/Tuya SmartPlug 5', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":77,"dateCode":"","friendlyName":"Tuya SmartPlug 5","hardwareVersion":1,"ieeeAddr":"0xa4c13878d7336b52","manufacturerID":4417,"manufacturerName":"_TZ3000_gvn91tmx","model":"TS011F_plug_1","networkAddress":49679,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0,"indicator_mode":"off/on","last_seen":"2022-12-22T19:10:26.142Z","linkquality":171,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":77,"latest_version":192,"state":"idle"},"update_available":false,"voltage":230}'
info 2022-12-22 20:10:27MQTT publish: topic 'zigbee2mqtt/Tuya SmartPlug 6', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":77,"dateCode":"","friendlyName":"Tuya SmartPlug 6","hardwareVersion":1,"ieeeAddr":"0xa4c13837340c4c89","manufacturerID":4417,"manufacturerName":"_TZ3000_gvn91tmx","model":"TS011F_plug_1","networkAddress":6827,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0,"indicator_mode":"off/on","last_seen":"2022-12-22T19:10:27.459Z","linkquality":178,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":77,"latest_version":192,"state":"idle"},"update_available":false,"voltage":231}'
debug 2022-12-22 20:10:27Is new image available for '0xa4c13837340c4c89', current '{"fieldControl":0,"manufacturerCode":4417,"imageType":54179,"fileVersion":77}', latest meta '{"fileVersion":192,"fileSize":307682,"url":"https://images.tuyaeu.com/smart/firmware/upgrade/20220907/1662545193-oem_zg_tl8258_plug_OTA_3.0.0.bin","sha512":"01939ca4fc790432d2c233e19b2440c1e0248d2ce85c9299e0b88928cb2341de675350ac7b78187a25f06a2768f93db0a17c4ba950b60c82c072e0c0833cfcfb"}'`

Shadow505, 2023-01-05:
@Koenkk Same issue here

Koenkk, 2023-01-06:
This is due to https://github.com/Koenkk/zigbee-herdsman-converters/issues/5246, will be fixed in todays hotfix release.

adnansarajlic, 2023-01-21:
Fixed, close.

closed 2023-01-21 by Koenkk

Shadow505, 2023-01-25:
@Koenkk I just tested it again and it still does not work, still the same issue as before

Zigbee2MQTT Version
1.29.2 commit: 14021395

Koenkk, 2023-01-26:
Can you provide the debug log of your issue?

Shadow505, 2023-01-29:
@Koenkk 

```
debug 2023-01-29 19:04:34: Is new image available for 'xxxx', current '{"fieldControl":0,"manufacturerCode":4417,"imageType":54179,"fileVersion":77}', latest meta '{"fileVersion":192,"fileSize":307682,"url":"https://images.tuyaeu.com/smart/firmware/upgrade/20220907/1662545193-oem_zg_tl8258_plug_OTA_3.0.0.bin","sha512":"01939ca4fc790432d2c233e19b2440c1e0248d2ce85c9299e0b88928cb2341de675350ac7b78187a25f06a2768f93db0a17c4ba950b60c82c072e0c0833cfcfb"}'
debug 2023-01-29 19:04:34: Update available for 'xxxx': NO
```

Koenkk, 2023-01-29:
`Update available for 'xxxx': NO`, where do you see an update is available?

Shadow505, 2023-01-29:
Because it says that the currently installed file version is 77 and the one available is 192. I believe that is also exactly the issue that the OP has indicated above.

Koenkk, 2023-01-29:
@Shadow505 could you also do a check with the latest dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html). If not, give me the output of executing `cat node_modules/zigbee-herdsman-converters/package.json| grep version` in the zigbee2mqtt folder

Shadow505, 2023-01-29:
@Koenkk Yes, the dev branch is correctly reporting that there is an update available. I did not try to install it yet.

Koenkk, 2023-01-30:
@Shadow505 

> Will the device restart (and therefore turn power off) if an update is installed?

No clue, depends on the device

Shadow505, 2023-01-30:
@Koenkk  I tested it this morning, the update worked. :) The device did indeed restart and turn power off, it was therefore a good idea do do this while the PC was not in use haha. When will the fix be available in the stable branch, so that I can switch back from dev?

Koenkk, 2023-01-30:
@Shadow505 with the next release on 1 February.

Shadow505, 2023-01-31:
Awesome. Thanks for your work! :)

rbron01, 2023-04-09:
Any idea what's in this version. In the Tuya app it says it's now 3.0. But no clue what's changed. 

Any idea where to find this?

vag01, 2024-08-02:
Hello everyone,
It seems that the problem is still there.
I tried several times but the device wont update.
Here is the state the I have of it.
Any idea on how to force this update?
{
    "ac_frequency": null,
    "child_lock": "UNLOCK",
    "countdown": 0,
    "current": 0,
    "indicator_mode": "off/on",
    "linkquality": 104,
    "power": 0,
    "power_apparent": null,
    "power_outage_memory": "restore",
    "state": "OFF",
    "update": {
        "installed_version": 78,
        "latest_version": 192,
        "state": "available"
    },
    "voltage": 238,
    "energy": null,
    "update_available": null
}

Gregorius-MC, 2024-08-21:
I have the same problem when updating.
The socket can be updated manually using the instructions for installing alternative firmware. The instructions can be found on the device page here: https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html
You need to put the following in the my_index.json file:

`[
    {
        "fileVersion": 192,
        "fileSize": 307682,
        "manufacturerCode": 4417,
        "imageType": 54179,
        "sha512": "01939ca4fc790432d2c233e19b2440c1e0248d2ce85c9299e0b88928cb2341de675350ac7b78187a25f06a2768f93db0a17c4ba950b60c82c072e0c0833cfcfb",
        "url": "https://images.tuyaeu.com/smart/firmware/upgrade/20220907/1662545193-oem_zg_tl8258_plug_OTA_3.0.0.bin"
    }
]
`

And after that, you need to re-check the device update and start the process. It helped me and everything went well. I use the Sonoff E module (ember driver) and Z2M 1.39.1.