Koenkk/zigbee2mqtt #15707: Occassional massive performance degradation. HA full backup restore helps.

szymucha94, 2022-12-22:
### What happened?

This issue happened for the first time since I switched from old Xiaomi Gateway 3 running in z2m mode (via tcp, chipset EFR32MG) to sonoff CC2652P coordinator. That's about 4 months now. Back then this what happening once every two weeks, most likely because I had 30 directly connected devices while the limit of that chipset was 32.
What exactly happened was that I started to receive notifications from HA that some classes of devices fail to report in within expected time. So I tested some devices (ie. generic wireless buttons that trigger lights via zigbee plugs) and I noticed extremely high delays in responsiveness (5-20s vs typical 200ms). Other zigbee functions such as OTA checks, electrical measurements, etc also failed completely or reacted after very long delays. Both via z2m interface and HA.
I've attempted to repair single device (aqara cng detector) but this failed after many attempts. Restarting HA, reconnecting (and waiting in power off state) USB coordinator, restarting z2m, powering off all zigbee routers (with circuit breaker) didn't help, so I assume this is a problem with z2m.
**What finally helped was restoring full HA backup from yesterday that also includes z2m addon. I'm using HAOS OVA image.**
I noticed that before I restored HA backup from yesterday HA was stuck in "Wrapping up startup, not everything will be available until it is finished". Hypervisor was not overloaded, didn't not run out of RAM and the disk IO was normal. I monitored system vitals for at least 10 minutes when zigbee network was in this state, and I'm 100% sure this has nothing to do with hypervisor resources. HA VM also did not produce abnormal CPU load or RAM usage.
My network consists of:
```
By device type
End devices: 47
Router: 19
By power source
Battery: 42
Mains (single phase): 24
By model
lumi.plug.maeu01: 11
lumi.sensor_magnet.aq2: 9
lumi.curtain.acn002: 6
lumi.weather: 6
lumi.sensor_motion.aq2: 5
lumi.sensor_wleak.aq1: 5
lumi.switch.n0agl1: 4
lumi.remote.b1acn01: 4
TS011F: 4
lumi.switch.l2aeu1: 3
lumi.sen_ill.mgl01: 2
lumi.sensor_smoke: 2
lumi.switch.l1aeu1: 2
lumi.vibration.aq1: 1
lumi.airmonitor.acn01: 1
lumi.sensor_gas.acn02: 1
```
I exclude as a root cause interference from BT/WiFi as this is also monitored by HA and no drop/increased traffic/RSSI degradation was noted on the network.

### How to reproduce it (minimal and precise)

no idea how to reproduce this.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff CC2652P Plus USB adaptor (zStack3x0)

RubenKelevra, 2022-12-30:
I see the same issue on my network:

- Many timeouts in the logs.
- Sometimes devices seem to be not reachable for several minutes – so a light stays on of a group for example.
- Sometimes I hit the light button in one room and it takes 2-3 seconds before the lights respond.
- ... or I need to hit the button several times before anything happens.
- This is a particular issue for motion sensors (I use three TuYa [TS0202](https://www.zigbee2mqtt.io/devices/TS0202.html#tuya-ts0202) which have massive issues with reporting (so the status just get stuck in one state). While a Xiaomi [RTCGQ11LM](https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html#xiaomi-rtcgq11lm) and an IKEA [E1525/E1745](https://www.zigbee2mqtt.io/devices/E1525_E1745.html#ikea-e1525%252Fe1745) only see 3-4 second delays.

## What I tried:

- I switched from Zigbee light groups to HA light groups, but the issues with single stuck lights from, time to time, remained.
- I tried changing the transmit power from 5 dBm to 20 dBm in the adapter settings (not sure my adapter supports this) – no change.
- I tried to override the adapter concurrency to 2 and to 16, the issue remained (maybe a tiny bit better on 2 – but not sure on this).
- Move the Wifi router further away (from 2 meters to 4 meters) from the Zigbee Adapter - no change.
- Move the 4G Router further away from the Zigbee adapter (3 meters to 4 meters) - no change.

## Setup:

#### Wifi 2.4 GHz:

Channel 1+5

#### Zigbee channel:

25

#### Zigbee2MQTT version

1.28.4

#### Adapter firmware version

20220219

#### Adapter

Slaesh's CC2652RB stick

#### Stats

Stats

```
By device type
End devices: 28
Router: 26

By power source
Battery: 28
Mains (single phase): 26

By vendor
MLI: 17
LUMI: 11
_TZ3000_vtscrpmw: 3
_TYZB01_dl7cejts: 3
_TZE200_ckud7u2l: 3
eWeLink: 3
XIAOMI: 3
SONOFF: 3
_TZE200_8ygsuhe1: 2
IKEA of Sweden: 1
feibit: 1
_TZ3000_qdmnmddg: 1
_TZ3000_26fmupbb: 1
undefined: 1
_TYZB01_iuepbmpv: 1

By model
tint-ExtendedColor: 11
lumi.sensor_magnet.aq2: 8
TS0601: 5
tint-Remote-white: 4
TS0121: 4
TS0202: 3
SA-003-Zigbee: 3
lumi.sen_ill.mgl01: 3
01MINIZB: 3
lumi.weather: 2
ZBT-ExtendedColor: 1
TRADFRI motion sensor: 1
lumi.sensor_motion.aq2: 1
Retro Bulb Gold XXL white+ambiance: 1
ZBT-ColorTemperature: 1
FNB56-COS06FB1.7: 1
TS0207: 1
TS0203: 1
```

RubenKelevra, 2022-12-30:
@szymucha94 wrote:

> I exclude as a root cause interference from BT/WiFi as this is also monitored by HA and no drop/increased traffic/RSSI degradation was noted on the network.

Can you explain how I do this? Thanks! :)

szymucha94, 2022-12-30:
> @szymucha94 wrote:
> 
> > I exclude as a root cause interference from BT/WiFi as this is also monitored by HA and no drop/increased traffic/RSSI degradation was noted on the network.
> 
> Can you explain how I do this? Thanks! :)

If you have bt/wifi RSSIs and packet counters reported to HA just check the history graph if there is any significant spike at the time of z2m failure.
Also in z2m you have LQI for each zigbee device. Check what is the expected, "normal" value and compare it to what you have during the failure. Assuming you enabled LQI entity in HA back when if was working.

RubenKelevra, 2023-01-28:
@szymucha94 I think I found the culprit:

After upgrading to Firmware 20221102 this issue is gone.

closed 2023-03-07 by github-actions[bot]