Koenkk/zigbee2mqtt #15710: LEDVANCE Smart+ Gardenpole Mini

Roarrk, 2022-12-23:
### What happened?

The device is seemingly correctly added, interview completes, power on/off works and I can change between Blue and Green but no other color works (and it is green when selecting blue). Changing brightness works. Changing Hue turns off the device, so does changing color. 

I cannot update the firmware either, as it does not seem to be correctly identified. As far as I can see a new update is avialable on update.ledvance.com (Gardenpole Mini RGBW Z3 | 3/17/2022 | 0x1189 | 0x0040 | 01066400) but Z2m detects manufacturerCode '4489' imageType '64'.

### What did you expect to happen?

It is a White/Yellow + Color LED - it should work in all colors.

### How to reproduce it (minimal and precise)

I have tried repairing multiple times. 

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB

closed 2023-01-30 by github-actions[bot]

HJoost, 2023-05-22:
I had the same issue on mine. The solution was that the connector between the poles and the controller was not alligned correctly (which is very easy to do if you don't pay full attention!). After assembling it right it works without problems :-)

Roarrk, 2023-05-22:
Holy crap you are right. Awesome! Thank you.

Roarrk, 2023-05-22:
> I had the same issue on mine. The solution was that the connector between the poles and the controller was not alligned correctly (which is very easy to do if you don't pay full attention!). After assembling it right it works without problems :-)

I had actually given up on these! It works. Thank you.